s-nakaoka/choreonoid #20: SimpleControllerPlugin と OpenRTCPlugin について

YoheiKakiuchi, 2015-07-24:
SimpleControllerPluginとOpenRTCPluginを両方同時にシミュレーションで使うことはできるでしょうか？
JVRCのチュートリアルにあるBodyRTCと
http://jvrc.github.io/tutorials/html-ja/index.html
SimpleControllerPluginで作ったプラグインを同時に使おうとしたのですが、
シミュレーションを走らせると、どちらかだけが有効になり同時に２つ使うことができませんでした。
図のように、BodyRTCとSimpleControllerを並列についかしてあり、シミュレーションを実行すると
チェックを入れてある方が使われます。両方チェックを入れると片方しか使われません。

[image]

s-nakaoka, 2015-07-24:
今のところ同時に使えるコントローラはひとつとなっています。
必要でしたら複数コントローラを同時に使えるようにしたいと思います。
参考までにお聞きしたいのですが、今回の例では、それぞれのコントローラはロボットの異なる部位を独立して動かすというものでしょうか？それとも、コントローラ同士の連携も必要でしょうか？

YoheiKakiuchi, 2015-07-25:
> 参考までにお聞きしたいのですが、今回の例では、それぞれのコントローラはロボットの異なる部位を独立して動かすというものでしょうか？それとも、コントローラ同士の連携も必要でしょうか？

ロボットのセンサー値（カメラなど）の出力と(RTC)、ロボットの関節角度のコントロール(SimpleController)仕様としておりまして、コントローラ同士の連携は不必要かと考えております。

s-nakaoka, 2015-07-27:
了解しました。
対応したいと思いますので、しばらくお待ちください。

s-nakaoka, 2015-07-28:
対応しました。
複数のコントローラアイテムを並べておけば、デフォルでは全てが実行されます。
チェックについては、チェックをつけると、つけたコントローラ（複数可）だけの実行になります。

SimpleControllerを組み合わせて使う場合は注意が必要です。デフォルトでは全ての関節角トルクを上書きしますので、最後に(一番下に置いて）実行したSimpleControllerの出力だけが採用されることになります。
SimpleControllerの方で関節トルクを出力しない場合は、コントローラのinitialize関数等で

 setJointOutput(false);

としておくと、出力がされなくなり、他のコントローラの出力が採用されるようになります。また、

 setJointOutput(jointId, true or false);

とすることにより、関節単位で設定を切り替えることも可能です。これを使えば、例えば上半身と下半身を別々のSimpleControllerで動かすことも可能になります。

サンプルとして、TankJoystickLight.cnoid を用意しました。
これはTankJoystickサンプルの拡張版で、 JoystickControllerに加えてLightControllerも読み込まれていて、後者がライトを点滅させるというものです。
ソースは sample/SimpleController 以下にあります。

YoheiKakiuchi, 2015-08-06:
確認できました．ありがとうございます．

closed 2015-08-06 by YoheiKakiuchi