s-nakaoka/choreonoid #22: 外部のpythonインタープリタからchoreonoidに接続したい

kindsenior, 2015-08-01:
PythonPluginを用いるとchoreonoidのPythonコンソールからchoreonoidを操作できると思います.
bash等で起動したipythonの補完やショートカット等がPythonコンソールからは効きません
bash等で外部から起動したpythonからchoreonoidを操作したいのですが、可能でしょうか?

LD_LIBRARY_PATHにlib/choreonoid-1.5を
PYTHONPATHにlib/choreonoid-1.5/pythonを
追加すると
from cnoid.BodyPlugin import *
といったモジュールのインポートはできるようになったのですが、
起動してあるchoreonoidのアイテムを取得しようとして
ItemTreeView.instance()
などとすると
choreonoid上のPythonコンソールでは<cnoid.Base.ItemTreeView object at ...>のようなオブジェクトが取得できますが、
外部から起動したpythonでは取得できず、続けて
ItemTreeView.instance().checkedItems(BodyItem)
とすると'NoneType' object has no attribute 'checkeditems'となります

s-nakaoka, 2015-08-03:
その場合、操作したいChoreonoidのプロセスとPythonインタプリタのプロセスが異なるので、モジュールをインポートできたとしても、Choreonoidのプロセスを操作することにはなりません。
そのようなことを実現したい場合は、何らかのプロセス間通信によってリモートコール的なことをできるようにしておく必要があります。

この例として、CorbaPluginにはメッセージビューをCORBAオブジェクト化するサンプルが含まれています。src/CorbaPlugin/corba の MessageView.idl でMessageViewの分散オブジェクトとしての定義をしており、この実装が src/CorbaPlugin/MessageView_impl.[h/cpp] になります。これによって、CorbaPluginをビルドしておけばMessageViewを分散オブジェクトとして別プロセスから操作できます。そのサンプルとしてtest.cppというファイルが含まれていて、CMakeLists.txtのコメントアウトをはずすとbin以下にcorba-testという実行ファイルがビルドされます。Choreonoidを起動しておいてからこれを実行すると、Choreonoidのメッセージビューにこのプログラムから送ったメッセージが表示されます。

Pythonから利用したければ、omniORBのPython版を利用してIDLからスタブを生成するなり、別途Pythonラッパを書くなりすればOKです。

まあ今どきCORBAは流行らないので、この方法はあまりおすすめはできませんけどね。

話が長くなりましたが、当初の目的はipythohのような高機能なPythonシェルを利用したいということですよね。でしたら、上記のような面倒なことはせず、ipythonを組み込んだPythonConsoleViewを作るのが一番よいと思います。実はこれは当方のTODOには入れているのですが、まだしばらく手をつけられる状況にありません。実際できるかも分かりませんが、もし興味があれば試していただけるとうれしいです。

kindsenior, 2015-08-04:
丁寧な解説ありがとうございます

>まあ今どきCORBAは流行らないので、この方法はあまりおすすめはできませんけどね。
そうですね．この場合，私も
>ipythonを組み込んだPythonConsoleViewを作る」
方がいいかと思います．
必須というわけではないので直ぐにやってみるかどうかは分かりませんが，時間があるときに試してみたいと思います

closed 2015-12-22 by s-nakaoka