s-nakaoka/choreonoid #26: choreonoid が死んでしまう

YuOhara, 2015-08-21:
https://github.com/start-jsk/rtmros_choreonoid/issues/14
でissueになっていますが,choreonoidのsimulationをstartしたところでprogramが落ちてしまいます。

```
Copyright (C) 2012 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
For bug reporting instructions, please see:
<http://bugs.launchpad.net/gdb-linaro/>...
Reading symbols from /usr/bin/choreonoid...(no debugging symbols found)...done.
(gdb) r
Starting program: /usr/bin/choreonoid /home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/config/jaxon_hg.cnoid
warning: no loadable sections found in added symbol-file system-supplied DSO at 0x7ffff7ffa000
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe3e4d700 (LWP 12153)]
[New Thread 0x7fffe364c700 (LWP 12154)]
[New Thread 0x7fffe21db700 (LWP 12155)]
[New Thread 0x7fffce00b700 (LWP 12170)]
[New Thread 0x7fffcd80a700 (LWP 12171)]
[New Thread 0x7fffcd009700 (LWP 12172)]
[New Thread 0x7fffcc808700 (LWP 12173)]
HGcontroller0: onInitialize()
[New Thread 0x7fffc5f3c700 (LWP 12205)]
[Thread 0x7fffc5f3c700 (LWP 12205) exited]
[New Thread 0x7fffc573b700 (LWP 12206)]
[Thread 0x7fffc573b700 (LWP 12206) exited]
[New Thread 0x7fffb8253700 (LWP 12216)]
Warning : Three or more triangles is defined for a edge.
cloneMap.setNonNodeCloning(true);
[New Thread 0x7fffb4b04700 (LWP 12581)]
cloneMap.setNonNodeCloning(true);
[New Thread 0x7fff9f5c8700 (LWP 12582)]
cloneMap.setNonNodeCloning(true);
[New Thread 0x7fffb7572700 (LWP 12586)]
cloneMap.setNonNodeCloning(true);
[New Thread 0x7fff9edc7700 (LWP 12587)]
cloneMap.setNonNodeCloning(true);
[New Thread 0x7fff9e5c6700 (LWP 12591)]
Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt. You must
reimplement QApplication::notify() and catch all exceptions there.

terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc

Program received signal SIGABRT, Aborted.
0x00007ffff5ec20d5 in raise () from /lib/x86_64-linux-gnu/libc.so.6
(gdb) where
#0  0x00007ffff5ec20d5 in raise () from /lib/x86_64-linux-gnu/libc.so.6
#1  0x00007ffff5ec583b in abort () from /lib/x86_64-linux-gnu/libc.so.6
#2  0x00007ffff64f0d4d in __gnu_cxx::__verbose_terminate_handler() ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff64eed96 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff64eede1 in std::terminate() ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff64ef046 in __cxa_rethrow ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff695816c in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#7  0x00007ffff695cf67 in QCoreApplication::exec() ()
   from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#8  0x00007ffff7a11295 in ?? () from /usr/bin/../lib/libCnoidBase.so.1.5
#9  0x0000000000400f99 in main ()

```
これはどのようなエラーでしょうか？

k-okada, 2015-08-24:
original discussion -> https://github.com/start-jsk/rtmros_choreonoid/issues/14

s-nakaoka, 2015-09-09:
これなんですが、プログラムのどこかで例外が投げられて、それが補足されなかった場合、バックトレースで例外発生箇所が分かるので、どこが問題か分かるのですが、これについてQtの仕様がやっかいで、Qtのメインループが全ての例外を補足しちゃうんですよね。バックトレースで

#6  0x00007ffff695816c in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () 

とあることから、そうなっていることが分かります。

こうなると、どこから例外が投げられたかが分からなくなってしまいます。
しかも補足したところで、プログラムの中断を避けられるわけでもないし、はっきり言って意味不明で迷惑な仕様です。まあ何か意図はあるのでしょうが…。

で補足するかしないかを変えるオプションくらいあるだろうと思ったのですが、それも無いようです。

結論としては、Qtのソースからコンパイルする際に、

 ./configure -no-exceptions

としてコンパイルすると、この機能を無効化できます。

私は最近はこうして自前でコンパイルしたQtのバイナリを使うようにしています。これだと例外で落ちた時に例外発生箇所が分かります。

あと例外ということで言うと、omniORBがやたらと例外を投げるので、この例もそうかもしれません。ただ、omniORBの例外もよく意味が分からなかったりします…。

そのような次第で、お手数をおかけしますが、上記の手法でコンパイルしたQtで試してみてもらえますでしょうか。よろしくお願いします。

s-nakaoka, 2015-09-09:
> あと例外ということで言うと、omniORBがやたらと例外を投げるので、この例もそうかもしれません。ただ、omniORBの例外もよく意味が分からなかったりします…。

この言い方は間違いでした。

gdbでデバッグするときに、例外発生時に呼ばれる関数があって、そこにブレークポイントを設定することで例外発生箇所で止めることができます。これを使ってQtのメインループに落ちてくる前に例外の発生を検知すればよいかと思うわけですが、通常のエラー処理を例外で行うこともよくあって、これが多いと落ちる前にそこらじゅうでbreakすることになってしまうので、デバッグになりません。それでomniORBはエラー処理で例外を使いまくるので、omniORBが関わっているとこのデバッグ方法が使えません。なので Qt のコンパイルオプションを変えるしかないという話でした。まあ悪いのはやはりQtの方ですね。

YuOhara, 2015-09-10:
改めて見返してみましたが、こちら
https://github.com/start-jsk/rtmros_choreonoid/issues/17
と同じ問題のようで、
omniNamesプロセスが無いときに起きていたエラーなようです。

どこで落ちているか調べるために、qtとchoreonoidをソースから入れてコンパイルしてみましたが、
今度はエラーでプログラムが落ちずにロボットが脱力しました。
今まではchoreonoidをubuntu12.04にaptで入れてやっていたので、そのせいで少々振る舞いが違うようです。

YuOhara, 2015-09-11:
こちらはっきりとした原因はわかっていませんが、
解決方法がわかったのと、再発しないので一旦closeしたいと思います。

qtでエラーが出ているように見える理由が分かってよかったです。
丁寧なご説明ありがとうございました。今後エラーが出たら教えていただいたとおりに対応します。

何かこちらの環境で試した方がいいことがあれば、またお知らせください。

closed 2015-09-11 by YuOhara