s-nakaoka/choreonoid #37: Choreonoid内でopenhrp-model-loaderを起動できない

snozawa, 2015-09-27:
Choreonoid内でopenhrp-model-loaderを起動できないとの
ご質問がhttps://github.com/fkanehiro/hrpsys-base/issues/796#issuecomment-143515479
でございましたので、こちらにご案内させていただきます。
よろしくお願いいたします。

YoheiKakiuchi, 2015-09-28:
タイミングの問題のような気がします。
model-loader自体はチュートリアルのやり方で起動できておりますが、起動のタイミングがbodyRTCのモジュールより遅いのではないかと思います。
http://fkanehiro.github.io/hrpsys-tutorials/html-ja/choreonoid/toChoreonoid.html

可能なら、一番先に起動して、コマンド後にN秒のスリープが入るというようなことができるとありがたいです。

s-nakaoka, 2015-09-28:
ExtCommandItemに"Waiting time after started" というプロパティを追加しました。ここに時間（秒）をセットすると、コマンド起動後にその時間分待ちます。
Pythonスクリプトからは、setWaitingTimeAfterStartedという関数で値を設定できます。

y-masutani, 2015-09-28:
@YoheiKakiuchi 様， @s-nakaoka 様，
ありがとうございます．openhrp-model-loaderを一番上に持ってきた場合はOKでした．
Waiting time after startedはまだ試していません．

fkanehiro, 2015-09-28:
* BodyRTCアイテムで設定するコントローラモジュールはBodyRTCアイテムが生成されるときに、ロード生成される
* 外部コマンドアイテムで指定されたコマンドは外部コマンドアイテムが生成されるときに実行される
* アイテムはアイテムツリーの先頭から順番に生成される

という理解で良いでしょうか。であれば、今回の件はコマンド起動後の待ち時間とは関係なく、アイテムの位置関係によっては必ず起きる問題で、アナウンスしておく必要があると思います。

fkanehiro, 2015-09-29:
@s-nakaoka さん、
回答をお待ちしています。（アイテムの生成順序については幅優先か、深さ優先かも教えて下さい）

s-nakaoka, 2015-09-30:
> BodyRTCアイテムで設定するコントローラモジュールはBodyRTCアイテムが生成されるときに、ロード生成される

正確には、BodyRTCアイテムがアイテムツリー上でBodyアイテムの下に配置された時点で生成されます。

> 外部コマンドアイテムで指定されたコマンドは外部コマンドアイテムが生成されるときに実行される

正確には、外部コマンドアイテムを生成して保存したプロジェクトファイルを再度読み込んだ時に、外部コマンドアイテムが読み込まれた時点で実行されます。

> アイテムはアイテムツリーの先頭から順番に生成される

そうなります。
ツリー上の順序は深さ優先です。

s-nakaoka, 2015-12-22:
多分解決したと思うのでこちらは閉じます。

closed 2015-12-22 by s-nakaoka