s-nakaoka/choreonoid #47: CONSTRAINT_FORCEの使い方

YoheiKakiuchi, 2015-10-16:
BodyRTCのconfファイルに、以下のような記述を入れると、
body.rtcに const_force のポートができることが確認できました。
~~~
out-port = const_force:LLEG_JOINT5:CONSTRAINT_FORCE
~~~

デフォルトの設定では、このポートからは接触データが出力されていないようで、
ソースでは以下のところで、isConstraintForceOutputModeがtrueになる必要があるようです。
https://github.com/s-nakaoka/choreonoid/blob/master/src/Body/ConstraintForceSolver.cpp#L1735-L1737

設定ファイルの書き方などでCONSTRAINT_FORCEを出すようにできますでしょうか？
よろしくお願いします。

YoheiKakiuchi, 2016-04-12:
こちら、ソースで
isConstraintForceOutputModeをtrue
にすると、データが出てくるようです。

ソースを変更する以外の設定方法はございますでしょうか？

s-nakaoka, 2016-04-13:
本件放置状態になっており申し訳ありませんでした。

> ソースを変更する以外の設定方法はございますでしょうか？

とりあえずPythonから設定可能としてみました。
https://github.com/s-nakaoka/choreonoid/commit/2006213de19f4ace7c66c8158005793e78636a0d

AISTSimulatorItemに"setConstraintForceOutputEnabled"という関数を追加しましたので、これを使ってtrueにしてみてもらえますでしょうか。
やり方は、 https://github.com/s-nakaoka/choreonoid/issues/64 と同様になります。

YoheiKakiuchi, 2016-04-14:
確認できました。
ありがとうございます。

closed 2016-04-14 by YoheiKakiuchi