s-nakaoka/choreonoid #54: Request for new version release

YuOhara, 2015-11-20:
https://github.com/start-jsk/rtmros_choreonoid/issues/134
で議論されていることなのですが、

* choreonoidのOpenRTMPlugin正しく上がらない
* 原因は、"openrtmのソースでtrunkのバージョンが1.1.1 から 1.1.2へ最近変更された"こと

ということが起こっています。
choreonoidのdeb新しいversionをリリースしていただけないでしょうか？

yosuke, 2015-11-23:
```
ldd /usr/lib/choreonoid-1.5/libCnoidOpenRTMPlugin.so 
	libRTC-1.1.1.so => not found
	libcoil-1.1.1.so => not found
```
となってしまっていますね。

choreonoidパッケージの再ビルドで治ると思うので、少々お待ちください（現在ビルド中です）。

本来は、libRTC-1.1.1.soではなくlibRTC.soの方にリンクを張って欲しいところですが、そうなっていないところが本質的な問題ですね。

yosuke, 2015-11-23:
この２行の流れを見る感じ、リンクに関してはChoreonoid側は悪くなさそうな感じがしています。OpenRTM側が特殊なことしているのかな、、、。

https://github.com/s-nakaoka/choreonoid/blob/master/src/OpenRTMPlugin/CMakeLists.txt#L93

https://github.com/s-nakaoka/choreonoid/blob/master/src/OpenRTMPlugin/CMakeLists.txt#L203

yosuke, 2015-11-23:
以下の出力が
```
objdump -p /usr/lib/x86_64-linux-gnu/libRTC.so | grep SONAME
```
こうなってしまう
```
  SONAME               libRTC-1.1.2.so
```
ので、cmakeはlibRTC-1.1.2.soの方をリンクしてしまうみたいですね。

なので問題はやはりOpenRTMの方で、ldに-sonameオプションがついていないのが良くないですね。

yosuke, 2015-11-23:
ビルドが完了しました。

apt-get update && apt-get upgradeで治ると思うのですが、どうでしょうか？

yosuke, 2015-11-24:
以下メモです。

OpenRTMはlibtoolを使って共有ライブラリを作っているが、libtool標準のバージョン規則を使っていない。

OpenRTMのバージョン規則は、majorではなくminorでABIの変更がかかってしまう。なので、libtoolに-version-infoオプションを指定して、SONAMEの設定やシンボリックリンクの作成をしてもらうことができず、-releaseオプションでシンボリックリンクの作成のみをしている。

現状で、SONAMEの設定は明示的にしておらず、リンクした時のpatchも含んだバージョンにリンクされてしまう。

YuOhara, 2015-11-24:
apt-get update ; apt-get upgradeで、治ることを確認しました。
ありがとうございます。

s-nakaoka, 2015-12-22:
松坂さん調査していただきありがとうございました。
とりあえずこのissueは閉じておきます。

closed 2015-12-22 by s-nakaoka