s-nakaoka/choreonoid #55: choreonoidのdebとソースの違い？

YuOhara, 2015-11-20:
今までdebのchoreonoidを使っていたのですが、
この度git clone してコンパイルして使ってみたところ、errorが出てロボットがヘタる（力が全く入ってない）ようになってしまいました。
やったことは、
```
git clone https://github.com/s-nakaoka/choreonoid.git
cd choreonoid
mkdir build
cd build
cmake ..
sudo make install
```
だけだったと思います。

choreonoid側の出力は、
```
Detecting plugin file "/usr/local/lib/choreonoid-1.5/libCnoidPythonPlugin.so"
Detecting plugin file "/usr/local/lib/choreonoid-1.5/libCnoidPythonSimScriptPlugin.so"
Detecting plugin file "/usr/local/lib/choreonoid-1.5/libCnoidBodyPlugin.so"
Detecting plugin file "/usr/local/lib/choreonoid-1.5/libCnoidSimpleControllerPlugin.so"
Detecting plugin file "/usr/local/lib/choreonoid-1.5/libCnoidPoseSeqPlugin.so"
Detecting plugin file "/usr/local/lib/choreonoid-1.5/libCnoidBalancerPlugin.so"
Body-plugin has been activated.
PoseSeq-plugin has been activated.
Python-plugin has been activated.
PythonSimScript-plugin has been activated.
SimpleController-plugin has been activated.
Balancer-plugin has been activated.

Loading project file "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/config/JVRC_VALVE.cnoid" ...
Task sequencer '' has been deactivated.
Restoring WorldItem "World"
Restoring BodyItem "JAXON_JVRC"
Loading OpenHRP Model File "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/jvrc_models/JAXON_JVRC/JAXON_JVRCmain_hrpsys_bush.wrl"
 -> ok!
Warning: OpenRTMPlugin is not loaded.
Warning: BodyRTCItem of OpenRTMPlugin is not a registered item type.
Warning: "BodyRTC" cannot be restored.
Restoring BodyItem "wall"
Loading OpenHRP Model File "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/jvrc_models/model/tasks/O1/wall.wrl"
 -> ok!
Restoring BodyItem "surface"
Loading OpenHRP Model File "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/jvrc_models/model/tasks/O1/surface.wrl"
 -> ok!
Restoring BodyItem "valve"
Loading OpenHRP Model File "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/config/model/tasks/VALVE/valve.wrl"
 -> ok!
Restoring BodyItem "drcbox"
Loading OpenHRP Model File "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/config/model/tasks/VALVE/drcbox.wrl"
 -> ok!
Restoring BodyItem "valve2"
Loading OpenHRP Model File "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/config/model/tasks/VALVE/valve.wrl"
 -> ok!
Restoring BodyItem "drcbox2"
Loading OpenHRP Model File "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/config/model/tasks/VALVE/drcbox.wrl"
 -> ok!
Restoring AISTSimulatorItem "AISTSimulator"
Warning: JVRCPlugin is not loaded.
Warning: JVRCManagerItem of JVRCPlugin is not a registered item type.
Warning: "JVRCManager" cannot be restored.
Restoring GLVisionSimulatorItem "GLVisionSimulator"
Restoring GLVisionSimulatorItem "GLVisionSimulator"
Restoring GLVisionSimulatorItem "GLVisionSimulator"
Restoring GLVisionSimulatorItem "GLVisionSimulator"
Restoring GLVisionSimulatorItem "GLVisionSimulator"
Restoring GLVisionSimulatorItem "GLVisionSimulator"
Restoring GLVisionSimulatorItem "GLVisionSimulator"
16 / 18 item(s) are loaded.
Warning: 2 item(s) are not correctly loaded.
Project "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/config/JVRC_VALVE.cnoid" has been loaded.
SubSimulatorItem "GLVisionSimulator" has been detected.
GLVisionSimulator detected vision sensor "HEAD_LEFT_CAMERA" of JAXON_JVRC as a target.
SubSimulatorItem "GLVisionSimulator" has been detected.
GLVisionSimulator detected vision sensor "CHEST_CAMERA" of JAXON_JVRC as a target.
SubSimulatorItem "GLVisionSimulator" has been detected.
GLVisionSimulator detected vision sensor "HEAD_RANGE" of JAXON_JVRC as a target.
SubSimulatorItem "GLVisionSimulator" has been detected.
GLVisionSimulator detected vision sensor "LARM_CAMERA" of JAXON_JVRC as a target.
SubSimulatorItem "GLVisionSimulator" has been detected.
GLVisionSimulator detected vision sensor "RARM_CAMERA" of JAXON_JVRC as a target.
SubSimulatorItem "GLVisionSimulator" has been detected.
GLVisionSimulator detected vision sensor "LARM_CAMERA_N" of JAXON_JVRC as a target.
SubSimulatorItem "GLVisionSimulator" has been detected.
GLVisionSimulator detected vision sensor "RARM_CAMERA_N" of JAXON_JVRC as a target.
Simulation by AISTSimulator has started.
```
と出ています。terminalでの出力は
https://gist.github.com/YuOhara/b5f63f4417b44e956354
のようになっています。

YoheiKakiuchi, 2015-11-20:
@kindsenior さんが詳しいですが、いくつか必要なPluginを有効にする必要があります。
buildディレクトリで、以下をするとコンパイルフラグを変更できます。

```
ccmake ..
```

corbaplugin, openrtmplugin が必須だったと思います。
以下のプラグインを読んでいるところで、debと同じだけ読み込んでいるといいかと思います。
```
Detecting plugin file "/usr/local/lib/choreonoid-1.5/libCnoidPythonPlugin.so"
Detecting plugin file "/usr/local/lib/choreonoid-1.5/libCnoidPythonSimScriptPlugin.so"
Detecting plugin file "/usr/local/lib/choreonoid-1.5/libCnoidBodyPlugin.so"
Detecting plugin file "/usr/local/lib/choreonoid-1.5/libCnoidSimpleControllerPlugin.so"
Detecting plugin file "/usr/local/lib/choreonoid-1.5/libCnoidPoseSeqPlugin.so"
Detecting plugin file "/usr/local/lib/choreonoid-1.5/libCnoidBalancerPlugin.so"
```

YuOhara, 2015-11-20:
なるほど、確かにそれで動きました。
http://choreonoid.org/ja/install/build-ubuntu.html#id3
に書いてあるように見えますが、なかなか難しいですね。

ありがとうございます。

YuOhara, 2015-11-20:
解決したので閉じます

closed 2015-11-20 by YuOhara