s-nakaoka/choreonoid #57: Choreonoid 1.4におけるRIC30モデルの不具合？

y-masutani, 2015-11-25:
開発版の問題ではないのですが， choreonoid-users-jaでは応答がないので，こちらにも書きます．

Choreonoid 1.4 Windows版で添付のRIC30のモデルを使っていてよくわからな
い現象に遭遇しましたので，助けてください．

不具合が再現するできるだけ簡単な事例を以下に記します．

------------------------------------------------------------------------
Choreonoid 1.4を起動

ファイル→新規→ワールド

ワールドを選択し，ファイル→読み込み→OpenHRPモデルファイル
  C:\Program Files (x86)\Choreonoid 1.4\share\model\RIC30\RIC30.yaml

RIC30を選択し，ファイル→新規→ポーズ列

PoseSeqを選択

関節スライダ
  全関節を表示
  ラジアンを選択

ポーズロール
  0秒において，全関節0を挿入

関節スライダ
  R_KNEE_Pだけを-0.4[rad]，他は全て0

ポーズロール
  1秒において「挿入」

関節スライダに表示される関節角度は
  R_HIP_P   -0.144
  R_KNEE_P  -0.112
  R_ANKLE_P  0.144
になってしまう．

シーンに表示されるロボットを見ても，股関節と足関節のピッチは0ではない．

ポーズロール
  時刻同期にチェックを入れて，0～1秒で時刻を動かしてみると，
  0.645秒において
  R_HIP_P    0.000
  R_KNEE_P  -0.282
  R_ANKLE_P  0.000
  だったものが，0.646秒において
  R_HIP_P   -0.127
  R_KNEE_P  -0.033
  R_ANKLE_P  0.127
  に急変する．

以上のポーズ列を保存したpseqファイルの中では，
    time: 1
    （中略）
      q: [ 
        0, 0, 0, -0.4, 0, 0, 0, 0, 0, 
        0, 0, 0, 0, 0, 0, 0, 0, 0 ]
となっている．

Choreonoid 1.4を起動し直し，ワールドとRIC30を設定し，上記で保存した
pseqファイルを読み込むと，1秒における関節角度は，
  R_HIP_P   -0.144
  R_KNEE_P  -0.112
  R_ANKLE_P  0.144
になっている．

この問題は，上記の「-0.4」を「-0.3」にすると起こらない．

上記の「-0.4」を「-2.0」のような大きな値にすると，R_HIP_PとR_ANKLE_Pだ
けでなく，他の3関節R_HIP_Y, R_HIP_R, R_ANKLE_Rも連動してしまう．

同様のことを似たモデルであるGR001で試すと，関節の連動は起きているが，
ポーズを与えた時刻で異なるポーズになってしまう問題は起きない．

------------------------------------------------------------------------

以上です．cnoidファイルとpseqファイルを添付します（拡張子．ｔｘｔを追加しています）．

この問題は，皆様の環境でも再現するでしょうか？

関節が連動するのは仕様でしょうか？意味のある動きのように見えません．
また，RIC30では，ポーズを与えた時刻にポーズが再現しないのは困ります．

解決方法をご教示いただけると幸いです．

よろしくお願いいたします．

[PoseSeq-ric30.pseq.txt](https://github.com/s-nakaoka/choreonoid/files/43449/PoseSeq-ric30.pseq.txt)
[ric30-test.cnoid.txt](https://github.com/s-nakaoka/choreonoid/files/43450/ric30-test.cnoid.txt)

y-masutani, 2015-11-25:
以下もchoreonoid-users-ja MLに出した内容です．

同じことをChoreonoid 1.5 on Ubuntu 14.04で試してみました．

R_KNEE_Pだけを変化させたポーズを与えたところ，ポーズとポーズの間で他の
関節が動くという現象は同様でしたが，ポーズを与えた時刻には，設定した通
りの角度になります．

・関節の連動は仕様でしょうか？
  何の狙いがありますか？

・この部分でバージョン1.4と1.5の違いは何でしょうか？
  1.4で1.5と同じ様な動作をさせることができないでしょうか？

・RIC30とGR001の違いは何でしょうか？
  モデルファイルの内容はよく似ているようなのですが．

学生にWindowsで使わせたいので，1.4で動作してくれると大変助かります．

s-nakaoka, 2015-11-25:
本件ですが、ML(choreonoid-users-ja)の方でコメントしておきましたので、よろしくお願いします。

closed 2015-12-28 by s-nakaoka