s-nakaoka/choreonoid #61: ポーズ列アイテムとシミュレーションの併用

fkanehiro, 2015-12-05:
github.com/jvrc/samplesのsample3.cnoidで一つのcnoidファイルに関節にPD制御を適用するシミュレーション用の設定とポーズ列アイテムが入っているのですが、そのままシミュレーションを開始するとロボットが弾かれたように転倒します。ポーズ列アイテムを削除すると正常に動作します。以前はこのままで動作していたように思いますが、何か変わりましたでしょうか。また何が起きているのでしょうか？

s-nakaoka, 2015-12-06:
BodyMotionItemについては、それがセットされている（今回の場合はPoseSeqItemが格納するBodyMotionItemが孫アイテムとしてある)状況でシミュレーションを行うと、シミュレータの内部でこれ用のハイゲインコントローラを自動で生成して、シミュレータをハイゲインモードにしてシミュレーションするようになっています。これにより、パターンデータをちょっとシミュレーションしてみることが、簡単にできます。

今回の不具合は、 https://github.com/s-nakaoka/choreonoid/issues/20 で行った修正後に出てしまったようです。この修正で複数のコントローラがある場合、それらを同時に動かせるようにしたのですが、今回は先頭のコントローラのみを用いたいのに、動作パターンのコントローラも動いてしまって、それでおかしくなっています。

とりあえずの解決方法としては、BodyRTCにチェックを入れて、SampleMotionにはチェックを入れないようにしてください。するとBodyRTCのみがコントローラとして使われます。

ただし、BodyMotionItemの簡易シミュレーション機能を、他のコントローラと混ぜて使うことはないと思うので、通常のコントローラがあるときは、簡易シミュレーション機能は無効になるよう改良したいと思います。

fkanehiro, 2015-12-07:
チェックの設定でとりあえず動くことを確認しました。

s-nakaoka, 2015-12-28:
> ただし、BodyMotionItemの簡易シミュレーション機能を、他のコントローラと混ぜて使うことはないと思うので、通常のコントローラがあるときは、簡易シミュレーション機能は無効になるよう改良したいと思います。

混ぜて使うことはあまりないとしても、BodyMotionItemの簡易シミュレーションと、他のコントローラによるシミュレーションを、使い分けたいということはあるかもしれませんね。そもそもあるロボットモデルに対して動作パターンを編集しながら、そのパターンとは別のコントローラもセットしているというのが、少し特殊な状況で、この場合どちらを使うこともあり得ると思います。なので、このような状況についてはチェックによる対応を行なっていただくことにしたいと思います。なので上記改良は当面見送りたいと思います。

fkanehiro, 2015-12-28:
はい、チェックを入れるのでよいと思いますが、問題はなにが起きているかわからない、ということだと思います。
動作編集用のアイテムがシミュレーション時のコントローラとしての機能も持っているというのは、なかなか気づかないと思います。

s-nakaoka, 2017-08-16:
報告が遅くなりましたが、この件現在の開発版では"BodyMotionController"というアイテムを明示的に導入することでBodyMotionItemのシミュレーションをできるようにしています。サンプルshare/project/GR001Sample.cnoidにもこれを導入しています。

closed 2017-08-16 by s-nakaoka