s-nakaoka/choreonoid #64: シミュレーション時のモデル間の摩擦係数の設定

kindsenior, 2015-12-10:
シミュレータアイテムのプロパティビューでStatic frictionとSlip frictionを設定するとシミュレータ内にある全ての物体の摩擦係数を設定できると思っています．

一方，シミュレータ内に３つ以上の物体があるとき，３つの物体をA,B,Cとすると
A・B間,B・C間,C・A間で摩擦係数が異なるという状況があると思います．

そのように異なる物体間で異なる摩擦係数を設定する方法はありますでしょうか？
大変お忙しいとは思いますが，宜しくお願いたします．

s-nakaoka, 2015-12-11:
ごめんなさい、この機能も要望が多いのですが、まだできるようになっていません。
エンジンの内部では対応していて、後はAPIとGUIの問題となっています。
ご存知かもしれませんが、実はOpenHRP3ではこれを設定するGUIがありました。
http://fkanehiro.github.io/openhrp3-doc/jp/um_collision_view.html
この点ではOpenHRP3から退化しておりまして、Choreonoidでも同様のインタフェースを用意したいとは思っています。
あとはこれを設定するための関数をSimulatorItemにも用意して、Pythonスクリプトで設定できるようにしてもよさそうですね。慣れればGUIでやるよりも簡単に設定できそうですし。そちらの方が簡単ですので（というかGUIを作る場合も結局関数は必要なので）、まずそれを実現するのがよいかもしれません。

s-nakaoka, 2015-12-24:
> あとはこれを設定するための関数をSimulatorItemにも用意して、Pythonスクリプトで設定できるようにしてもよさそうですね。

これに関して、とりあえずプロトタイプ的な実装を行いました。AISTSimulatorItemに

 void setFriction(Link* link1, Link* link2, double staticFriction, double slipFriction);

という関数を追加しています。この関数により、特定のリンクペアに対して、個別に摩擦パラメータを設定できます。

これのPythonラッパも追加しましたので、Pythonスクリプトから設定できます。
この例として、sample/python/TuneSR1Frictions.py を用意しました。
例えばSR1Walkのサンプルに関して、

 choreonoid --python sample/python/TuneSR1Frictions.py share/project/SR1Walk.cnoid

とかすると、右足と床に関しては摩擦係数を０にして滑るような設定でシミュレーションを行います。
スクリプトに関してここではコマンドラインから指定していますが、PythonScriptItemやPythonSimScriptItemでプロジェクトに組み込んでおくこともできます。

ただ、この関数についてはプロトタイプで、汎用性と拡張性を考えて改良の予定です。なのでこれを用いたスクリプトは今後修正の必要が出てくるかと思いますが、ご了承下さい。

kindsenior, 2015-12-24:
対応ありがとうございます．
早速使ってみたいと思います．
修正の可能性についても了解いたしました．

closed 2015-12-24 by kindsenior