s-nakaoka/choreonoid #68: pcファイルで指定するライブラリが不足？

fkanehiro, 2015-12-22:
pkg-configを使ってChoreonoidのライブラリをリンクすると、boost-systemとboost-filesystemを更に追加でリンクする必要があります。
pcファイルに指定するライブラリの不足でしょうか。

s-nakaoka, 2015-12-28:
どのような状況かもう少し詳しく書いていただけますでしょうか。
作成しているものは、Choreonoidのプラグインではなくて、独立したライブラリや実行ファイルということでしょうか？
リンクエラーがでるのでしょうか？その場合はどのようなエラーメッセージですか？
boost-filesystemのリンクが必要ということで、ちょっと思い当たったのはsrc/Utilで定義されているFileUtilのヘッダですが、これをインクルードしていたりしますか？

fkanehiro, 2015-12-28:
github.com/jvrc/tutorialsのtutorials/rtc/RobotControllerRTC3をコンパイルするときの話でしたが、boost-filesystemはsrc/RobotControllerRTC.cpp内でfilesystem::pathを使用しているためでした。
失礼しました。
boost-systemはcnoid/FileUtilをインクルードすると必要になるようです。
以下のようなエラーとなります。
```
CMakeFiles/RobotControllerRTCComp.dir/RobotControllerRTC.cpp.o: 関数 `_GLOBAL__sub_I_RobotControllerRTC.cpp' 内:
RobotControllerRTC.cpp:(.text.startup+0x74): `boost::system::generic_category()' に対する定義されていない参照です
RobotControllerRTC.cpp:(.text.startup+0x80): `boost::system::generic_category()' に対する定義されていない参照です
RobotControllerRTC.cpp:(.text.startup+0x8c): `boost::system::system_category()' に対する定義されていない参照です
collect2: error: ld returned 1 exit status
make[2]: *** [src/RobotControllerRTCComp] エラー 1
make[1]: *** [src/CMakeFiles/RobotControllerRTCComp.dir/all] エラー 2
make: *** [all] エラー 2
```

closed 2017-12-30 by s-nakaoka