s-nakaoka/choreonoid #69: 32bit OSでコンパイルエラー

fkanehiro, 2015-12-23:
ubuntu 12.04, 14.04, 15.04の32bit版で以下のようなコンパイルエラーとなります。
```
/«BUILDDIR»/choreonoid-1.5.0+dfsg+20151223+508+17~ubuntu14.04.1/src/PythonPlugin/PythonConsoleView.cpp:357:65: error: no matching function for call to 'min(std::basic_string<char>::size_type, long unsigned int&)'
             maxLength = std::min(memberNames[i].size(),maxLength);
                                                                 ^
/«BUILDDIR»/choreonoid-1.5.0+dfsg+20151223+508+17~ubuntu14.04.1/src/PythonPlugin/PythonConsoleView.cpp:357:65: note: candidates are:
In file included from /usr/include/c++/4.8/algorithm:61:0,
                 from /usr/include/boost/function/detail/prologue.hpp:13,
                 from /usr/include/boost/function/function_template.hpp:13,
                 from /usr/include/boost/function/detail/maybe_include.hpp:13,
                 from /usr/include/boost/function/function0.hpp:11,
                 from /«BUILDDIR»/choreonoid-1.5.0+dfsg+20151223+508+17~ubuntu14.04.1/src/Util/SignalTemplate.h:8,
                 from /«BUILDDIR»/choreonoid-1.5.0+dfsg+20151223+508+17~ubuntu14.04.1/src/Util/Signal.h:186,
                 from /«BUILDDIR»/choreonoid-1.5.0+dfsg+20151223+508+17~ubuntu14.04.1/include/cnoid/Signal:1,
                 from /«BUILDDIR»/choreonoid-1.5.0+dfsg+20151223+508+17~ubuntu14.04.1/src/Base/View.h:8,
                 from /«BUILDDIR»/choreonoid-1.5.0+dfsg+20151223+508+17~ubuntu14.04.1/include/cnoid/View:1,
                 from /«BUILDDIR»/choreonoid-1.5.0+dfsg+20151223+508+17~ubuntu14.04.1/src/PythonPlugin/PythonConsoleView.h:8,
                 from /«BUILDDIR»/choreonoid-1.5.0+dfsg+20151223+508+17~ubuntu14.04.1/src/PythonPlugin/PythonConsoleView.cpp:5:
/usr/include/c++/4.8/bits/stl_algobase.h:193:5: note: template<class _Tp> const _Tp& std::min(const _Tp&, const _Tp&)
     min(const _Tp& __a, const _Tp& __b)
     ^
/usr/include/c++/4.8/bits/stl_algobase.h:193:5: note:   template argument deduction/substitution failed:
/«BUILDDIR»/choreonoid-1.5.0+dfsg+20151223+508+17~ubuntu14.04.1/src/PythonPlugin/PythonConsoleView.cpp:357:65: note:   deduced conflicting types for parameter 'const _Tp' ('unsigned int' and 'long unsigned int')
             maxLength = std::min(memberNames[i].size(),maxLength);
                                                                 ^
In file included from /usr/include/c++/4.8/algorithm:61:0,
                 from /usr/include/boost/function/detail/prologue.hpp:13,
                 from /usr/include/boost/function/function_template.hpp:13,
                 from /usr/include/boost/function/detail/maybe_include.hpp:13,
                 from /usr/include/boost/function/function0.hpp:11,
                 from /«BUILDDIR»/choreonoid-1.5.0+dfsg+20151223+508+17~ubuntu14.04.1/src/Util/SignalTemplate.h:8,
                 from /«BUILDDIR»/choreonoid-1.5.0+dfsg+20151223+508+17~ubuntu14.04.1/src/Util/Signal.h:186,
                 from /«BUILDDIR»/choreonoid-1.5.0+dfsg+20151223+508+17~ubuntu14.04.1/include/cnoid/Signal:1,
                 from /«BUILDDIR»/choreonoid-1.5.0+dfsg+20151223+508+17~ubuntu14.04.1/src/Base/View.h:8,
                 from /«BUILDDIR»/choreonoid-1.5.0+dfsg+20151223+508+17~ubuntu14.04.1/include/cnoid/View:1,
                 from /«BUILDDIR»/choreonoid-1.5.0+dfsg+20151223+508+17~ubuntu14.04.1/src/PythonPlugin/PythonConsoleView.h:8,
                 from /«BUILDDIR»/choreonoid-1.5.0+dfsg+20151223+508+17~ubuntu14.04.1/src/PythonPlugin/PythonConsoleView.cpp:5:
/usr/include/c++/4.8/bits/stl_algobase.h:239:5: note: template<class _Tp, class _Compare> const _Tp& std::min(const _Tp&, const _Tp&, _Compare)
     min(const _Tp& __a, const _Tp& __b, _Compare __comp)
     ^
/usr/include/c++/4.8/bits/stl_algobase.h:239:5: note:   template argument deduction/substitution failed:
/«BUILDDIR»/choreonoid-1.5.0+dfsg+20151223+508+17~ubuntu14.04.1/src/PythonPlugin/PythonConsoleView.cpp:357:65: note:   deduced conflicting types for parameter 'const _Tp' ('unsigned int' and 'long unsigned int')
             maxLength = std::min(memberNames[i].size(),maxLength);
```

s-nakaoka, 2015-12-24:
修正しました。

fkanehiro, 2015-12-25:
ありがとうございます。確認しました。

closed 2015-12-25 by fkanehiro