s-nakaoka/choreonoid #80: Invalid OpenGL call in GLSceneRendererImpl

bchretien, 2016-01-24:
After seeing the following SIGFPE in `GLSceneRenderer`:

```txt
0x00007ffff78b94cb in cnoid::GLSceneRendererImpl::pick (this=0x555555a7adb0, x=0x41, y=0x143) at choreonoid/src/Base/GLSceneRenderer.cpp:1340
1340        unsigned int id = (color[0] * 255) + ((int)(color[1] * 255) << 8) + ((int)(color[2] * 255) << 16) - 1;
gdb$ p/f color
$3 = {1.8806588e+13, 3.0611365e-41, -9.92069426e+33, 4.59163468e-41}
```

I noticed that there was no OpenGL error checking in choreonoid. After enabling that, the error appears:

```txt
OpenGL error 0x0502 (GL_INVALID_OPERATION) at choreonoid/src/Base/GLSceneRenderer.cpp:1315 - for glPushAttrib(GL_ENABLE_BIT)
```

The error should probably be solved in `SceneWidgetImpl::updateLatestEventPath()` were the call to `pick()` is made. I don't know if that's the actual issue, but the [OpenGL doc](https://www.opengl.org/sdk/docs/man2/xhtml/glPushAttrib.xml) states:

> GL_INVALID_OPERATION is generated if glPushAttrib or glPopAttrib is executed between the execution of glBegin and the corresponding execution of glEnd.

s-nakaoka, 2016-02-02:
Could you tell me what operation causes this error? Does it crashe a Choreonoid process?
Perhaps the function for the picking is executed when the OpenGL context is inactive, but I'm not sure why that occurs.

bchretien, 2016-02-02:
As I said, this happened while looking for something else, and I thought you might want to know about my findings.

Basically, I enabled SIGFPE checking with `feenableexcept(FE_ALL_EXCEPT & ~FE_INEXACT)` on Linux (since it can detect overflows, underflows, divisions by zero), and I enabled OpenGL error checking (e.g. [this](https://github.com/bchretien/choreonoid/commit/52bb64c67ae60ac196f058bf750171c6eeb43587)). It's always good to have, it can detect lots of errors (I found several in my projects with that), and it's easy to disable at compile time. The error detected here is probably not that relevant, but I guess there are more lying around, so this GitHub issue is more of a list of debugging tips that you may find useful.

s-nakaoka, 2016-04-13:
I'm sorry for this late replay, and thank you for your advise.
I understood the usefulness of inserting the debug macro for OpenGL function calls.
However, I want to keep codes simple as much as possible, and it's better to avoid inserting such extra macros if possible.
I have tested the apitrace tool you mentioned, and it seems very useful for debugging OpenGL codes. In my test, I couldn't find out how to get OpenGL error messages from the tool, but if it is possible, it's better to use it instead of inserting the macros. Do you know if it is possible to do such a thing?

s-nakaoka, 2016-04-13:
In the manual of apitrace ( https://github.com/apitrace/apitrace/blob/master/docs/USAGE.markdown ), there is a description on "OpenGL annotations". It seems that this is used for getting OpenGL error messages, and I'll try this first.

norihiro-ito, 2017-04-17:
おそらく関連する事象かもと思い、ここに報告します。
1.6のコードをUbuntu 14.04でビルドして使っています。
Choreonoidを起動し、ビューの境界をドラッグすることでSceneビューの大きさを変更したあと、マウスポインタをSceneビューの上に移動させると落ちてしまうという現象に悩まされています。
Sceneをリサイズしなければ落ちないようです。
事象だけの報告で申し訳ありませんが、よろしくお願いいたします。

s-nakaoka, 2017-04-18:
ご報告ありがとうございます。
よろしければ、OS以外の動作環境についても教えていただけますでしょうか。
具体的には、32bit or 64bit、GPU（nVidiaチップ等の場合はプロプライエタリドライバを使っているかどうか）、仮想マシン上で動かしているか、その場合仮想マシンのソフトウェアとそのバージョン、ホストOS、ゲストOS等について。
お手数をおかけしますが、よろしくお願いします。

norihiro-ito, 2017-04-18:
お手数おかけいたします。

使用している仮想マシンソフトウェアは、VMware Workstation 12 Player 12.5.5 build-5234757
PCにはnVidia GeForce GTX760が載っていますが、仮想マシン上では有効ではないと認識しています。

ホストOSは、Windows 7 Professional, 64-bit 6.1.7601, Service Pack 1
Winodws上にはnVidiaのドライバーを導入しています。

また、Ubuntuは仮想マシン上で動作させています。バージョンはUbuntu 14.04.5 LTS
Linux ubuntu 4.4.0-72-generic #93~14.04.1-Ubuntu SMP Fri Mar 31 15:05:15 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
なお、Sceneビュー上に各種サンプルのオブジェクトは表示、操作できていますので、GPUレベルではないと考えています。
簡単ですが、よろしくお願いいたします。

norihiro-ito, 2017-04-20:
デバッグ指定で再ビルドして、スタックトレースをとってみました。

> Thread #1 [choreonoid] 21667 [core: 0] (Suspended : Signal : SIGSEGV:Segmentation fault)	
> 	0x7ffff4a77034	
> 	0x7ffff4a770d4	
> 	0x7fffd71cc86f	
> 	0x7fffd71cce63	
> 	0x7fffd71cb2f5	
> 	0x7fffd6f79390	
> 	0x7fffd70f7043	
> 	0x7fffd71cb11f	
> 	0x7fffd71cabc6	
> 	glXMakeCurrentReadSGI() at 0x7ffff4a5913d	
> 	QGLContext::doneCurrent() at 0x7ffff52236e2	
> 	QGLContext::reset() at 0x7ffff522354b	
> 	QGLContext::~QGLContext() at 0x7ffff51b3cc8	
> 	QGLContext::~QGLContext() at 0x7ffff51b3d19	
> 	QGLPixelBuffer::~QGLPixelBuffer() at 0x7ffff51c2b72	
> 	QGLPixelBuffer::~QGLPixelBuffer() at 0x7ffff51c2c49	
> 	cnoid::SceneWidgetImpl::updateLatestEventPath() at SceneWidget.cpp:1,004 0x7ffff7915280	
> 	cnoid::SceneWidgetImpl::updatePointerPosition() at SceneWidget.cpp:1,387 0x7ffff79167ef	
> 	cnoid::SceneWidgetImpl::mouseMoveEvent() at SceneWidget.cpp:1,375 0x7ffff79167b7	
> 以下略

Sceneをリサイズ後に、マウスポインタを入れて落ちたときのコード位置は、cnoid::SceneWidgetImpl::updateLatestEventPath() の以下の行でした。

>     if(pixelBufferForPicking){
>         if(!usePixelBufferForPicking || pixelBufferForPicking->size() != size()){
>             pixelBufferForPicking->makeCurrent();
>             delete pixelBufferForPicking;     // この行です。
>             pixelBufferForPicking = 0;
>         }
>     }

調査に役立つとよいのですが、、、。