s-nakaoka/choreonoid #84: ForceSensor の挙動について

hmatsushita, 2016-02-18:
ForceSensor の rotation 設定で角度を指定し、センサ可視化で確認すると矢印マーカが指定した角度の向きに伸びますが、これは想定通りの挙動でしょうか。
Choreonoid 付属の SR1 モデルを以下のように修正し、AIST シミュレータのハイゲインモードでシミュレーションを実行し確認しましたが、モデルは直立して静止した状態なので、角度指定に関わらず矢印マーカが直上に伸びるのではないかと考え質問する次第となります。

```
$ diff -u SR1.wrl.orig SR1.wrl
--- SR1.wrl.orig	2016-02-18 11:20:23.988746501 +0900
+++ SR1.wrl	2016-02-18 17:39:51.597644977 +0900
@@ -785,7 +785,7 @@
                                      children [
 				       DEF LeftAnkleForceSensor ForceSensor {
 					 translation 0 0 0
-					 rotation 1 0 0 0
+					 rotation 1 0 0 0.785
 					 sensorId 0
 				       }
 				       DEF LEG_SHAPE6 Transform {
```

s-nakaoka, 2016-03-03:
#86 でfkanehiroさんが直してくれたようなので、確認してみてください。

hmatsushita, 2016-03-03:
ご連絡ありがとうございます。
問題が解決された事を確認しました。

closed 2016-03-03 by hmatsushita