s-nakaoka/choreonoid #87: yamlのモデルファイルでパス変数を使いたい

kindsenior, 2016-02-28:
yamlのモデルファイルでwrlを指定するときにパス変数を使いたいです

やり方としては
1. pythonスクリプトからならパス変数を使用してモデルファイルを読み込める
2.ファイルメニュー->プロジェクトファイルオプション->パス変数の設定でできそう?
があると思っています

以下,質問です
1. に関してはパス変数を使用したモデルファイルの読み込みはできているのですが，pythonからのプロジェクトファイルの読み込み方法が分からず，プロジェクトファイルによる様々な表示設定ができないので不便です．
pythonからプロジェクトファイルを読み込む方法はありますでしょうか？

2. に関して
a)そもそもユーザがパス変数を設定して，これをプロジェクトファイルやモデルファイル(yamlファイル)で参照できるようになるというものでしょうか？
b) a)の場合，適当に設定してみたところ,設定ファイルのpathVariablesというところに設定はできたようなのですが，プロジェクトファイルやyamlからはどのように参照すればよいでしょうか?
(ソースをぱっと見た感じだとプロジェクトの読み込み時に変数を参照する部分はまだ実装されてない感じでしょうか??)

3. 他にyamlのモデルファイルやプロジェクトファイルでパス変数をしようする方法はありますでしょうか？

お忙しいところ大変申し訳ございませんが，宜しくお願い致します

s-nakaoka, 2016-03-03:
確かにモデルファイル以外でもパス変数を使えると便利ですね。パス変数というか、何にでも使える変数という感じで。
現状では、パス変数が有効なのはプロジェクトファイルのみとなっています。
今後の改良事項に加えたいと思いますが、今立て込んでいるためこちらで少し時間がかかるかもしれません・・・。

kindsenior, 2016-03-07:
ありがとうございます

よくよく考えてみると1.でpythonでパス変数を使用してモデルファイルを読み込むスクリプトをプロジェクトにpythonスクリプトのアイテムとして登録すると，パス変数によるモデルファイルの読み込みとプロジェクトファイルへの表示設定の保存の両立ができることに気づきました．
当面はこの方針で行きたいと思います

closed 2016-03-07 by kindsenior