s-nakaoka/choreonoid #92: Windowsの場合，シーンビューでリンクを選択できない．

y-masutani, 2016-03-30:
WindowsでソースからビルドしたChoreonoidのシーンビューの編集モードにおいて，リンクを選択できません．シーンビューの中のどこでマウスを左クリックしても，ウィンドウの右下の表示はずっと
グローバル座標 ＝ (0.000 0.000 0.000)
のままです．本来はリンクの名前などが表示されるはずです．

y-masutani, 2016-04-01:
本件，大変不便しています．他の方は，再現していないでしょうか？
インタラクティブな操作ができるChoreonoidの良さが半減してしまいます．

s-nakaoka, 2016-04-01:
今確認してみたのですが、私の環境では正常に動作しています。
（ソースについてはWindows版を動くようにしてMLでアナウンスした際のバージョンと、現時点での最新版にてチェックしました。）
コンパイラや他のライブラリはマニュアルの記述に従っていて、Windowsのバージョンは8.1の64ビット版です。

ひょっとすると、ビデオドライバというか、OpenGLのドライバに問題があるのかもしれません。この機能が正常に動作するためには、OpenGLのZバッファの内容を読み出す必要があるのですが、依然仮想マシンで試したらこの関数が正しく動作していなくて、ご指摘の症状と同じ症状がでました。仮想マシンでなくても、ビデオボードやドライバによっては同様の問題があるのかもしれません。

確認までに、閲覧モードの場合はどうでしょうか？閲覧モードでは、マウスの指している座標が右下に表示されます。これは正しく出ていそうですか？あるいは、マウスの左ボタンドラッグによる視点の回転は、ドラッグを開始したときにマウスが指していた座標を中心に回転するのですが、これは正しく動作していますか？上記の問題があると、マウスの指している点とは関係なく原点まわりで回転したりします。

y-masutani, 2016-04-01:
環境は以下の通りです．

    Windows 7 Professional SP1 64bit
    Visual Studio 2012

利用しているツールやライブラリは以下の通りです．

    CMake 3.5.0
    Boost 1.60.0 boost_1_60_0-msvc-11.0-64.exe
    Qt 5.2.1 msvc2012 64-bit OpenGL
    Eigen 3.2.8
    OpenRTM-aist-1.1.1-RELEASE_x86_64_vc11.msi
    OpenRTM-aist-Python_1.1.0-RELEASE_x86_64.msi
    python-2.7.10.amd64.msi
    numpy-1.10.4+mkl-cp27-cp27m-win_amd64.whl

Qtの新しいバージョンには，VS2012の64bitのバイナリがなく，5.2.1を使いましたが，ビルドは通りました．QtのライブラリにOpenGL有のものと無のものがあるのですが，何か関係あるでしょうか？

閲覧モードでもウィンドウの右下の表示は，

    グローバル座標 ＝ (0.000 0.000 0.000)

のままで変化しません．マウスの左ボタンドラッグによる視点の回転は，マウスの指している点とは関係なく原点まわりになっています．

解決の糸口はないものでしょうか？

s-nakaoka, 2016-04-01:
なるほど、やはりOpenGLでZバッファを読む関数が正しく動作しないものとなっている可能性が高そうです。

> QtのライブラリにOpenGL有のものと無のものがあるのですが，何か関係あるでしょうか？

そう言えば以前はそのようにバイナリが分かれていましたね。
これは恐らく以下のページの説明が該当しているのではないかと思います。
http://wiki.qt.io/Qt_5_on_Windows_ANGLE_and_OpenGL_Japanese

つまり、バージョンによってはOpenGLをDirectXでエミュレートしているものがあるようで、その場合今回の症状が出るのかもと思ったのですが、ただ"OpenGL"がつくほうのバイナリを使われているということなので、その方が問題が出ないような…。それとも"OpenGL"がつくほうが、DirectXによるエミュレートのバージョンなのでしょうか？あといずれにしてもエミュレートできるのはOpenGL ESとあって、Choreonoidで使っているのはESではないOpenGLなので、エミュレートだとそもそも動かない気が・・・。うーむ、よく分かりませんね。

というわけでよくわからないのですが、"OpenGL"がつかない方のバイナリがひょっとしてエミュレートではない方だとすると、それでうまく動く可能性はあるかもしれません。

あとはやはりOpenGLのドライバに問題があるかもしれません。比較的古いPCでチップセット内蔵型のGPUだったりすると、WindowsではOpenGLのサポートがよくない場合もあります。ビデオボードやドライバを他のもので試すことは難しいでしょうか？おすすめはnVidiaのGPUです。最近のものなら大体動くと思います。

それか、Qtも最近のバージョンではDirectXのエミュレートとかあまり言わなくなったようなので、VC++とQtを最新のものにすれば問題が解決するかもしれません。

上記は全て私の憶測で恐縮なのですが、今考えられるのは以上のようなことかな、ということで、よろしくお願いします。

y-masutani, 2016-04-01:
回答ありがとうございます．自宅にnVidiaの入ったPCがあったのでビルド済みのバイナリをコピーして試してみましたが，症状は変わりませんでした．研究室の環境をVisual Studio 2012に統一しているので，新しい環境を試しにくい状況です．まずは，Qt 5.2.1のOpenGL無の方で試してみます．

y-masutani, 2016-04-02:
ところで，同じ環境において，バイナリをインストールしたChoreonoid 1.4では問題は出ていません．これはどう考えればいいですか？

s-nakaoka, 2016-04-02:
> ところで，同じ環境において，バイナリをインストールしたChoreonoid 1.4では問題は出ていません．これはどう考えればいいですか？

GPUやOpenGLドライバの問題ではなさそう、ということでしょうか。

やはりお使いのバージョンのQtのバイナリがOpenGLに関して２種類あるというところが、何か関係しているような気はします。保証はできませんが・・・。

y-masutani, 2016-04-03:
Ｑｔ 5.2.1のmsvc2012 64-bit （OpenGLなし）をインストールし，Choreonoidをビルドしてみましたが，SceneWidget.cpp のコンパイルでOpenGL関係の識別子（例えば，glBegin）が定義されていないというエラーが大量に出ます．

tsugeharu-akai, 2016-04-03:
ご存じの通り
Qt5.2.1のmsvc2012の64bitバイナリには以下の2種類があります。
(1)qt-opensource-windows-x86-msvc2012_64_opengl-5.2.1.exe
(2)qt-opensource-windows-x86-msvc2012_64-5.2.1.exe

(1)は3D描画にOpenGLのネイティブのドライバーを利用するもので、
(2)はANGLEというGoogleが開発したDirectXでOpenGL ESを仮想的にシュミレーションして使うものです。
(2)ができた経緯は古いノートPC（特にインテルGMA系のビデオチップが乗っているもの）がDirectXはサポートしているが、
OpenGLはサポートしていないものがかなりあり、GoogleがWebGL普及のためにANGLEというライブラリを開発しました。
Qtは5.x系からこのANGLEを利用しています※。

ここで重要なのはOpenGLとOpenGL ESはサポートしているOpenGLの命令が若干異なることです。
OpenGL ESはOpenGLより後発のもので、OpenGLが互換性のためにサポートしている命令（例えばglBegin）がサポートされていません。
（なのでglBeginでエラーが出るはずです）
Choreonoidの3D描画ではOpenGL ESがサポートしていない命令を使用しているので、
ChoreonoidをWindowsでビルドするには、Qt5.xを使うのであればopenglのものである必要があると思います。

※ちなみにQt5.5(?)以降のWindowsのQtにはopenglとangleの区別はなくなっています。
これはQt5.5（？）以降はQt自身がダイナミックにOpenGLとANGLEを切り替えられるようになったためです。
つなりPCにOpenGLのドライバーがあれば、OpenGLを利用し、なければANGLEでOpenGL ESをシュミレーションするようなことを最新のQtは自動で行ってくれます。

y-masutani, 2016-04-03:
情報ありがとうございます．ただいま，VS2012 64bitでQt 5.6をビルドしています．ビルドし始めてからもう3時間ぐらい経ちますが… ビルドが通ったとして，果たしてそれでうまくいくのやら…

y-masutani, 2016-04-04:
VS2012 64bitでソースからビルドしたQt 5.6のライブラリを使って，Chorenoidのcmakeからやり直して，ビルドしましたが，Ｑｔ 5.2.1のmsvc2012 64-bit （OpenGLなし）の場合と同様に，SceneWidget.cpp のコンパイルでOpenGL関係の識別子（例えば，glBegin）が定義されていないというエラーが大量に出ます．OpenGLを使うための設定が必要なのでしょうか？お手上げ状態です．

s-nakaoka, 2016-04-04:
Qtのコンパイル時の設定で、OpenGLモジュールをコンパイルするかどうかのチェックみたいなのはありませんでしたでしょうか？

tsugeharu-akai, 2016-04-04:
QtをソースからビルドしOpenGLを使用するには
configure時にOpenGLのフラグを指定する必要があります。
http://doc.qt.io/qt-5/configure-options.html
にあるように
-opengl desktop
か、もしくはANGLEとダイナミックに切り替えたいのであれば
-opengl dynamic
を指定する必要があります。

y-masutani, 2016-04-04:
ありがとうございます．やってみます．これだけ時間のかかるビルドは久しぶりです．

y-masutani, 2016-04-05:
Qt 5.6を「-opengl desktop」付きでconfigure, nmakeし（8時間かかりました！），それを使ってChoreonoidをビルドしたところ，シーンビューの問題は解消しました．ありがとうございました．お騒がせしました．

s-nakaoka, 2016-04-07:
うまく動いたようでよかったです。やはりQtのOpenGLの問題だったのですね。

y-masutani, 2016-04-07:
この問題，まだ解決していませんでした．グラフィックボードを積んでいないノートPC（Panasonic CF-J10）で実行したところ，シーンビューの状態は前のままです．同じPCでもChoreonoid 1.4では問題ありません．「-opengl dynamic」でQtをビルドすれば解決するのでしょうか？？？

tsugeharu-akai, 2016-04-08:
私自身以前にUbuntuでChoreonoid1.5上でPluginを作成した経験があるだけで、
WindowsでのChoreonoidビルド経験があるわけではありませんが、
おそらく-opengl dynamicでビルドしても結果は変わらないと思います。
-opengl dynamicでビルドするとOpenGL NativeとANGLEをダイナミックに切り替えられるようになるだけなので、ANGLEではChoreonoidは動かないと思われるためです。

Panasonic CF-J10で1.5で動かないということなので、
まずはPanasonic CF-J10のグラフィックボードのドライバーを調べて最新のものにアップデートしてみるのはどうでしょう？
ただよく分からないのはChoreonoid1.4では正常に動くということです…。

s-nakaoka, 2016-04-08:
うーん、なかなかうまく動かないのですね。
こちらで同じ状況を再現できないのがつらいところです。
再現できれば、デバッグして原因を探れるのですが・・・。
CPU内臓のGPUということであれば、私のLet's note CF-LX3の環境でも動いています。
CF-J10もHD Graphics 3000 とのことで、GPUは似たようなものなので、GPUが原因とも考えにくいですね。
シーンビューの表示自体はできているのですよね。ざっとみたかんじ、うまく動いている環境と比べてシーンビューの表示で何か異なる部分はありますでしょうか？

s-nakaoka, 2016-04-08:
参考までに、今回の症状と関連がありそうなソースコードの箇所について説明しておきます。
src/Base/GLSceneRender.cpp に、

 bool GLSceneRendererImpl::pick(int x, int y)

という関数があります。

これはシーンビューでマウスが指している箇所にある物体とその位置を取得するための関数です。これがうまく動かないと、リンクの選択ができなかったり、視点変更がドラッグ開始点を中心に行われなかったりします。

ここでglReadPixelsというOpenGLの関数が使われています。私が以前経験したのは、仮想マシンで動かしたらこの関数が実際の値を正しく返していませんでした。結果として、同じ症状が出ていました。

デバッグの際には、この関数が正常に動いていそうかをまずチェックすることになるかと思います。

y-masutani, 2016-04-08:
@tsugeharu-akai 様，

>Panasonic CF-J10で1.5で動かないということなので、
まずはPanasonic CF-J10のグラフィックボードのドライバーを調べて最新のものにアップデートしてみるのはどうでしょう？

コントロールパネル → デバイスマネージャ でディスプレアダプターの「Intel(R) HD Graphics Family」に対してドライバーソフトウェアの更新をやってみましたが，

> デバイス用のドライバーソフトウェアが最新であることが確認されました。

と表示されました．他にできることがあるでしょうか？

> ただよく分からないのはChoreonoid1.4では正常に動くということです…。

そうなんです．ですので，スッキリしないというか，諦めきれません．

y-masutani, 2016-04-08:
「-opengl dynamic」でQtをビルドしました，しかし，それを使うとChoreonoidのビルドができませんでした．