s-nakaoka/choreonoid #94: WindowsでBodyRTCのモジュールを読み込む際に異常終了

y-masutani, 2016-04-02:
WindowsのChoreonoi 1.5において，自作のRTCのdllファイルを，インストールしたディレクトリのlib\choreonoid-1.5\rtc に置き，BodyRTCのプロパティのコントローラのモジュール名を入力した直後にDebug Assertion Failed!になります．リンク先の画像をご覧ください．
https://goo.gl/photos/P6zZBExsjimLFSTw7
何が悪いのでしょうか？

なお，同じ環境で，Choreonoid 1.5 に付属のOpenRTM-SR1Walk.cnoidは正常に動作しています．

y-masutani, 2016-04-02:
dllファイルをDebugではなくReleaseでビルドしたところ，症状は出なくなりました．問題は解決していないのかもしれませんが．

s-nakaoka, 2016-04-02:
確かWindowsというかVC++では、デバッグ版のバイナリとリリース版のバイナリを混ぜて使うことはできなかったような気がします。
実際メッセージのほうもvector iteration incomatibleと出ていますね。恐らくstd::vectorの展開内容が、デバッグ版のときとリリース版の時で違うのだと思います。

ちなみにVC++のこのあたりの仕様は正直理解に苦しみます。リリース版とデバッグ版で根本的にコードが異なってくる部分があるようで、リリース版のバグをデバッグ版でデバッグしようとしたら、そちらではバグがでなかった、ということもこれまで何回もありました。これではデバッグ版の意味がありませんよね・・・。

closed 2017-08-16 by s-nakaoka