s-nakaoka/choreonoid #95: RT System Editorからシミュレーションを開始できますか？

y-masutani, 2016-04-02:
BodyRTCを使ってRTミドルウェアと一緒に使う場合，RT System EditorでActivateすることによって，Choreonoidのシミュレーションを開始する方法はありますか？

実ロボットとChoreonoidによるシミュレータを切り替えて使う学生実験の教材を作っているのですが，実ロボットの場合とシミュレータの場合の操作ができるだけ共通になるようにしたいと考えています．

実ロボットの場合は，RT System Editorでコンポーネントを接続し，Activateします．シミュレーションの場合は，Chreonoidを起動しプロジェクトを読み込み，その後はRT System Editorで操作できないかと．

s-nakaoka, 2016-04-04:
現状ではそのような機能はありませんが、プラグインを書けば可能だと思います。
Choreonoid側でシミュレータの操作を行うためのRTCを生成するようにして、それがactivateされたときにシミュレーションを開始するような実装にしておけばよろしいかと。
プラグインでRTCを生成するサンプルとして、sample/RTCVisionSensorSamplePlugin があります。cmakeでBUILD_RTC_VISION_SENSOR_SAMPLEをONにしておくとコンパイルされます。プロジェクトファイルはOpenRTM-TankJoystickVisionSensor.cnoidです。このサンプルでは、ロボットのカメライメージをロボットのRTCから取得して、プラグインで用意した表示用のRTCへと入力することで、画面に画像が表示されます。この表示用のRTCと同様にして、シミュレータアイテムのstartSimulation()とかの関数を呼ぶRTCを用意すればOKかと思います。

y-masutani, 2016-04-05:
回答ありがとうございます．ちょっと荷が重いので，今後検討します．
今は次のような方法でやって入ます．

1. Choreonoidでプロジェクトを読み込む．BodtRTCのコントローラモジュールは指定せず．
2. RT System Editorで，BodyRTCと接続するコンポーネントを含むXMLを読み込む．
3. Choreonoidでシミュレーション開始
4. RT System EditorでAll Activate

y-masutani, 2017-06-16:
中岡様，1年以上に教えていただいた本件を実際に試してみようとしています．対象は，RTミドルウェアとChoreonoidを使ったロボットプログラミングの教材です．やりたいことは，Choreonoidの外部からactivate, deactivateを送って「初期位置からのシミュレーションjン開始」「シミュレーションの停止」をすることです．

release-1.5.0を基にWindowsでVS2012で作業しています．Cmakeで「BUILD_VISION_SENSOR_RTM_SAMPLE」「BUILD_VISION_SENSOR_SAMPLE」にチェックを入れ，Choreonoidをビルドし直し，プロジェクト「OpenRTM-TankJoystickVisionSensors.cnoid」を動作させることはできました．RT System Editorで4個のコンポーネントが存在することを確認しました．この中の「VisionSensorSample」が肝心だと思うのですが，これがどこで設定されているかわかりません．アイテムビューにはそれらしきものは見当たらず，TankVisionSensors.confでもポート間の接続が定義されているだけです．

もう少しヒントをいただけないでしょうか．よろしくお願いいたします．

y-masutani, 2017-06-26:
古いissueにスレッドをつなげるのはマズいでしょうか？感知されていないでしょうか？

「VisionSensorSample」がどこで設定されているのか疑問でしたが，ソースコードをわからないなりに読んでみたところ，無条件でコンポーネントが起動されているように読めました．試しに上記でビルドしたChoreonoidを何もプロジェクトを読み込まずに起動した状態で，RT System Editorで見てみたところ，「VisionSensorSample」が存在しました．ということで，これを真似して，シミュレーション制御用のコンポーネントが作れそうな気がしてきました．

s-nakaoka, 2017-06-27:
> 古いissueにスレッドをつなげるのはマズいでしょうか？感知されていないでしょうか？

大丈夫です。書き込みがあったことも、メール通知がきております。
返信が遅くなってしまいすみません。

お察しのように、そのサンプルでは、プラグインのソースコードの方で（ハードコーディングで）RTCを生成しております。これはサンプルなのでコードを短くするためにそうしてしまったのですが、「シミュレーション制御用のコンポーネント」であれば、（目的や使い方はほぼ決まっているので）同じやり方（ハードコーディングで生成する）でよいのではないかと思います。

ちなみにVisionSensorSample的なことを汎用的に行うための機能（ポイントクラウドとかを出力するコンポーネントが別にあって、それのポートをつなくだけで可視化ができる）を今開発中です。

closed 2017-08-16 by s-nakaoka