s-nakaoka/choreonoid #96: Windowsでプラグインが読み込めない

y-masutani, 2016-04-05:
#92 で書いたようにWindows 7でChoreonoid 1.5をソースからビルドしました．インストールしたディレクトリを別のWindows 7のPCにコピーしたところ（platforms/qwindows.dllの追加は必要でしたが），問題なく動作しました．同じものを，WIndows 8.1のPCにコピーしたところ，起動はできるのですが，メッセージビューにプラグインを読み込めないというメッセージが大量に出ます．

````
プラグインファイル"C:\xxxxxxxx\lib\choreonoid-1.5\CnoidBalancerPlugin.dll"を検出しました。
プラグインの読み込みができませんでした。
Cannot load library C:\xxxxxxxx\lib\choreonoid-.5\CnoidBalancerPlugin.dll: 指定されたモジュールが見つかりません。
````
xxxxxxxx はインストールしたディレクトリです．さまざまなdllファイルに対して，上記の3行のパターンのメッセージが繰り返されています．ファイルとしては，C:\xxxxxxxx\lib\choreonoid-1.5\CnoidBalancerPlugin.dll は存在しています．「指定されたモジュール」とは何を指しているのでしょうか？

この問題はどのように解決していけばいいでしょうか？

y-masutani, 2016-04-05:
Dependency Walkerというツールで，CnoidBalancerPlugin.dllを調べてみたところ，多数のDLLファイルが，「指定されたファイルが見つかれません」となります．例えば，「BOOST_FILESYSTEM-VC110-MT-1_60.DLL」です．しかし，そのファイルはchoreonoid.exeと同じディレクトリに存在しています．

y-masutani, 2016-04-05:
同じ，Windows 8.1のPCにChoreonoid-1.4.0-win32.exeをインストールしてみました．
CnoidBalancerPlugin.dllについては同様のエラーが出ました．しかし，その他のｄｌｌについては，エラーは出ていません．

```
プラグインファイル"C:\xxxxxxxx\lib\choreonoid-1.4\yyyyyyyy.dll"を検出しました。
```
というメッセージだけがずらっと並びます．

1.5でもWindowsのビルド済みのバイナリを公開していただきたいです…

y-masutani, 2016-04-05:
CnoidBalancerPlugin.dllの「指定されたモジュールが見つかりません」のメッセージは，ビルドしたWindows 7の環境でも出ています．

s-nakaoka, 2016-04-07:
最終的に「Balancerプラグインが読み込まれました。」というメッセージが出ていませんでしょうか？
これが出ていれば、プラグインは読みこまれています。

最初にでるメッセージは、依存しているDLLがまだ読み込まれていない状態でBalancerプラグインを読み込もうとしていて出るようです。

Balancerプラグインについては以前ソースを非公開としていて、コンパイル済みのバイナリのみを添付していました。このバイナリをいろんな環境でコンパイルしたChoreooid本体や他のプラグインと組み合わせて読み込めるようにするため、DLLの作り方に少し細工をしました。その影響で、このようなメッセージがでているようです。

今はBalancerプラグインのソースもChoreonoid本体に含めていますので、今後他のプラグインと同様の作り方にしたいと思います。

s-nakaoka, 2016-04-07:
上記回答は

> CnoidBalancerPlugin.dllの「指定されたモジュールが見つかりません」のメッセージは，ビルドしたWindows 7の環境でも出ています．

に対するものでした。

Windows 8.1 にコピーしてうまくいかないのは、Windowsの違いというより、Visual C++か何かのライブラリがそのPCに入っていないのが理由かもしれません。

Choreonoidのビルドの設定でINSTALL_RUNTIME_DEPENDENCIESをONにしておくと、実行に必要なDLLをコピーするようにしています。ただ、VC++のランタイム等で必要なものが全てコピーされていないという不備があるのかもしれません。

Windows版のバイナリインストーラも用意したいと思っており、その際にこの問題も解決したいと思います。

今のところは、そのWindows8.1のPCに同じVC++をインストールするとか、どこかでダウンロードできるランタイムをインストールすれば、動くようになるかもしれません。あるいは、他のライブラリで足りないものがある可能性もあります。boostとかQtとか・・・。どれが足りないのか突き止めるのはなかなか大変なのですけれど・・・。

y-masutani, 2016-04-07:
コピー先のWindows 8.1のPCにも同じバージョンのVisual Studioが入っています．
メッセージビューに「指定されたモジュールが見つかりません。」の内訳を知ることはできないでしょうか？

y-masutani, 2016-04-07:
Windows 8.1のPCにおいて，Qt以外のライブラリは新たにインストールし，QtはWindows 7でビルドしたものをコピーして，Choreonoidをビルドしてみました．platforms/qwindows.dllの追加は必要でしたが，起動はできます．しかし，同じように多数の「指定されたモジュールが見つかりません。」が表示されました．

仕方がないので，QtもWindows 8.1のPCでビルドしてみます．

s-nakaoka, 2016-04-07:
> メッセージビューに表示された「指定されたモジュールが見つかりません。」の内訳を知ることはできないでしょうか？

私もこれで苦しんだことがあるので、内訳をメッセージで出せればと思ったのですが、Windowsが返してくるエラーメッセージにはそのような情報は含まれていないようで断念しました。なので、Dependency Walkerというツールで地道に調べたりしてます。（これ使ってもよく分からないことがありますが・・・。）

s-nakaoka, 2016-04-07:
> 仕方がないので，QtもWindows 8.1のPCでビルドしてみます．

その前に、Qtのディレクトリに入っているdllを全てChoreooidのbin以下にコピーして、症状が改善されるか試されてもよいかもしれません。それで動くようでしたら、後はどれが足りないDLLかを突き止めるだけになり、それもinstallでコピーされるようにすればよいので。
こちらでも調べてみて、何か分かりましたら報告＆改善をいたします。

y-masutani, 2016-04-07:
Dependency Walkerを使って，CnoidBodyPlugin.dllを対象として，Windows 7とWindows 8.1で結果を比較してみました．以下の三つのファイルが怪しそうです．
```
API-MS-WIN-CORE-KERNEL32-PRIVATE-L1-1-1.DLL
API-MS-WIN-CORE-PRIVATEPROFILE-L1-1-1.DLL
API-MS-WIN-SERVICE-PRIVATE-L1-1-1.DLL
```
これらのファイル名をググってみると，Windows8.1で問題あり！というような内容が色々ヒットしますが，そこから先どうすればいいかがわかりません．

y-masutani, 2016-04-07:
Windows 8.1のPCでQt 5.6をソースからビルドし，それを使ってChoreonoidをビルドしましたが，問題は解消しませんでした．開発チームでは，Windows 8.1での動作実績はありますか？

y-masutani, 2016-04-08:
Dependency Walkerで見た時に問題ではないかと思われる以下のファイルをC:\Windows\SysWOW64\downlevel で見つけたので，インストールディレクトリのbinにコピーしてみましたが，状況は変わりませんでした．
```
API-MS-WIN-CORE-KERNEL32-PRIVATE-L1-1-1.DLL
API-MS-WIN-CORE-PRIVATEPROFILE-L1-1-1.DLL
API-MS-WIN-SERVICE-PRIVATE-L1-1-1.DLL
```

y-masutani, 2016-04-08:
> その前に、Qtのディレクトリに入っているdllを全てChoreooidのbin以下にコピーして、症状が改善されるか試されてもよいかもしれません。それで動くようでしたら、後はどれが足りないDLLかを突き止めるだけになり、それもinstallでコピーされるようにすればよいので。

これもやってみました．Qtのバイナリがインストールされたディレクトリ（configureの-pregixで指定したディレクトリ）のbinディレクトリの下にあった86個のdllファイルを，Choreonoidのインストールディレクトリのbinの下にコピーしてみましたが，状況は変わりませんでした．

y-masutani, 2016-04-08:
問題を切り分けるために知りたいのですが，Windows 8.1においてChoreonoid 1.5の動作実績はありますか？そのビルドの条件を教えていただきたいです．

s-nakaoka, 2016-04-08:
> 問題を切り分けるために知りたいのですが，Windows 8.1においてChoreonoid 1.5の動作実績はありますか？そのビルドの条件を教えていただきたいです．

はい、動作実績はあります。私の以下のPCで正常に動作しています。

その１：
ノートPC Panasonic Let's note CF-LX3
CPU: Core i7-4600U
GPU: CPUに内蔵のもの (HD Graphics ナントカ）

その２：
自作デスクトップPC
GPU: nVidia Geforce GTX750Ti

その他の条件（コンパイラ、ライブラリ等）は全て以下にマニュアルに従っています。
http://choreonoid.org/ja/manuals/1.5/install/build-windows.html

y-masutani, 2016-04-08:
Win 8.1で動いているのですね！
こちらで試しているビルドの条件は，

- Visual C++ 2012 64bit
- CMake 3.5
- Boost 1.60 （VC++2012 64bit用のバイナリ）
- Qt 5.6（ソースからビルド）
- Eigen 3.2.8

です．一番違いのありそうなのは，VC++でしょうか？これを変更するのは厄介なので，やりたくないのですが… VC++の設定は，CMakeの標準のままでしょうか？

@s-nakaoka 様が，Win8.1で作られたバイナリ一式をいただいて，試させていただくことはできないでしょうか？

hattorishizuko, 2016-04-08:
ｄｌｌがロードできない問題で、お手数なのですが、簡単なプログラムを作成していただけないでしょうか。

LoadLibraryするだけの単純なものですが、うまく動けば、エラー原因が表示されると思います。
以下がコードです。

#include "stdafx.h"
#include <iostream>
#include <Windows.h>

int _tmain(int argc, _TCHAR* argv[])
{

  // DLLのロード
  HMODULE hModule = LoadLibrary(argv[1]);
  if (hModule == NULL)
  {
    std::cout << "DLLのロードに失敗しました。" <<  GetLastError() << std::endl;
    return 0;
  }else{
    std::cout << "DLLのロードに成功しました。" << std::endl;
  }

  // DLLの解放
  FreeLibrary(hModule);

  return 0;
}

VC++でコンソールアプリケーションを作成して、プリコンパイル済みヘッダーのオプションを選択してください。

できたプログラムを　cnoreonoid\binの中におき、
コマンドプロンプトでそのディレクトリに移動して、sample.exe ..\lib\choreonoid-1.5\CnoiｄBodyPlugin.dll
として実行してください。

コンパイル済みのものを次に添付します。

s-nakaoka, 2016-04-08:
[checkdll.zip](https://github.com/s-nakaoka/choreonoid/files/209656/checkdll.zip)

なお、実行ファイルの名前をcheckdll.exeに変えておきました。

y-masutani, 2016-04-08:
```
checkdll.exe ..\lib\choreonoid-1.5\CnoiｄBodyPlugin.dll
```
を実行すると，「checkdll.exe - システム エラー」というパネルが現れ，「MSVCR120.dll」がないと言われます．これは，VC++2013のランタイムですよね？このPCに入っているVC++2012でビルドしたcheckdll.exeでも同じ結果でした．ちなみに，binディレクトリには，mcvcr110.dllとmsvcp110.dllが入っています．

y-masutani, 2016-04-08:
どこかにVC++2013を前提としたコードが埋め込まれているのでしょうか？

y-masutani, 2016-04-09:
その後，分かったことを報告します．

1. VC++2013の再配布パッケージ https://www.microsoft.com/ja-jp/download/details.aspx?id=40784 をインストールすると，プログインが正常に読み込めるようになりました．Win7，Win8.1のどちらのVC+2012でビルドしたものでもOKでした．
2. VC++2013のランタイムDLLと思われるmsvcr120.dllとmsvcp120.dllをC:\Windows\SysWOW64からChoreonoidのインストールディレクトリのbinにコピーし，その後でVC++2013の再配布パッケージをアンインストールしましたが，この方法ではプラグインを読み込めませんでした．binにコピーするファイルが足りないのかもしれません．それ以上は追及していません．
3. ChoreonoidでDLLのランタイムを使わないようにすればいいのではないかと思い，CMakeでCMAKE＿CXX_FLAGS_RELEASEの/MDを/MTに変えて，Configure, Generateし，ビルドを試みましたが，大量のエラーが出てお手上げ状態です．

Choreonoidは/MDでビルドしないといけないのでしょうか？とすると，今のところ，VC++2013の再配布パッケージをインストールするというのが解決策です．ビルドに使ったバージョン以外のランタイムを要求するとはどういうことでしょう？？？ビルド済みのバイナリを配布する場合にややこしいことになりそうですね．

s-nakaoka, 2016-04-09:
報告ありがとうございます。
これを見て確認したところ、この症状の原因らしきものがみえてきました。
Choreonoidのソースに src\thirdparty\windows64\bin というディレクトリがあり、ここに外部ライブラリのビルド済みのDLLを入れていて、Choreonoidのビルド・インストール時にコピーするようになっています。どうもここからコピーしているlibz2.dllというファイルが、VC++2013の環境でコンパイルされたもので、そのランタイム(MSVCR120.DLL等）ともリンクされていたようです。このlibz2.dllがChoreonoidのDLLからもリンクされていて、それらを実行するときにVC++2013のランタイムが必要になっていたのではないかと。

ビルド済みのものをコピーするようにしている以上、そのバイナリは環境に依存しないものとしなければなりません。詳細を確認の上、上記が正しければ、それはChoreonoidの不備であり、問題を回避できるよう修正したいと思います。作業は月曜日以降になりますので、申し訳ありませんがもうしばらくお待ちいただけますでしょうか。よろしくお願いします。

s-nakaoka, 2016-04-11:
libbz2.dll について、これを必要とする機能をWindowsでは使えないようにし、このdllへの依存をなくしました。対応する機能はHRPSYSという制御ソフトのログファイルがbzip2で圧縮していてもそのまま読めるというもので、なくなっても一般のユーザには影響ないかと思います。
この改良をgithubの方にアップしておきましたので、お試しいただけますでしょうか。

y-masutani, 2016-04-12:
最新の内容をgithubからcloneし，Windows 7のVC++2012でビルドしたバイナリが，Windows 8.1とWindows
10で正常に動作することを確認しました．プラグインを読み込めないという問題は解消されました．ありがとうございます．

ただし，以前にも報告したように以下の二つのことが必要でした．それにも対応してくださるようお願いいたします．

'inttypes.h' がないというエラーを避けるために，PyOpenRTMPlugin.cppの
```
#undef HAVE_UNISTD_H
```
の下に
```
#undef HAVE_INTTYPES_H
```
を追加．

実行時のエラー
```
This application failed to start because it coould not find or load
 the Qt platform plugin "windows".
```
を避けるために，インストール先のbinディレクトリにサブディレクトリ「platforms」を作り，その中に「qwindows.dll」をコピー．

y-masutani, 2016-04-12:
まだ問題は解消していませんでした．

PCによっては，CnoidPythonPlugin.dll と CnoidPythonSimScriptPlugin.dll に対して「指定されたモジュールが見つかりません。」と表示されます．PythonがインストールされているPCでも出ることがあります．

checkdllの結果は以下の通りです．
```
checkdll.exe ..\lib\choreonoid-1.5\CnoidPythonPlugin.dll
DLLのロードに失敗しました。126

checkdll.exe ..\lib\choreonoid-1.5\CnoidPythonSimScriptPlugin.dll
DLLのロードに失敗しました。126
```

y-masutani, 2016-04-12:
自己フォローです．

>PCによっては，CnoidPythonPlugin.dll と CnoidPythonSimScriptPlugin.dll に対して「指定されたモジュールが見つかりません。」と表示されます．PythonがインストールされているPCでも出ることがあります．

問題が起こるPCに入っていたPythonが32bit版だったのが原因のようです．64bit版に入れ替えると解消しました．

Pythonのプラグインを使わない場合は，そのままにしておいても（64bit版に入れ替えなくても）大丈夫ですか？

s-nakaoka, 2016-04-13:
> ただし，以前にも報告したように以下の二つのことが必要でした．それにも対応してくださるようお願いいたします．

了解しました。こちらでも確認の上、対応したいと思います。

> Pythonのプラグインを使わない場合は，そのままにしておいても（64bit版に入れ替えなくても）大丈夫ですか？

恐らく大丈夫だと思いますが、必要なければCMakeの設定でPythonプラグインをオフにしておくと、より良いかと思います。

y-masutani, 2016-04-14:
>恐らく大丈夫だと思いますが、必要なければCMakeの設定でPythonプラグインをオフにしておくと、より良いかと思います。

大丈夫とのこと了解しました．ビルド済みのバイナリを他のPCで実行する場合を想定した質問でした．

closed 2016-07-23 by s-nakaoka