s-nakaoka/choreonoid #97: cnoidファイルの設定で相対パスを保つ方法

y-masutani, 2016-04-07:
cnoidファイルの中でBodyRTCのプロパティの moduleName や confFileName の値を相対パスで指定できないでしょうか？cnoidファイル基準でも，Choreonoidのインストールディレクトリ基準でも構いません．BodyItemでは，${SHARE}を使って後者を実現しています．それを真似してみたところ，読み込む時はOKなのですが，次に書き込むときに${SHARE}のままではなく，パスの値が展開されてしまいます．

y-masutani, 2016-04-12:
moduleNameやconfFileNameの値となるパス名ですが，
- 絶対パスならばそのまま
- 相対パスならば，cnoidファイルの位置を基準
- ${SHARE}と書けば，インストールディレクトリのshareを基準
というような仕様ではいけませんか？

コントロールモジュールや設定ファイルは，cnoidファイルの近くに置くことや，ユーザのディレクトリに置くことを想定していないのでしょうか？JVRCの頃からずっと疑問に思っています．

s-nakaoka, 2016-04-13:
確かにご指摘のように、このあたりまだ十分に整備できていない部分がありますね。
基本的には上記のようにしたいのですが、コントローラモジュールについては複数のプロジェクトに共通で使うことも多く、実行可能バイナリでもあるので、システムの所定の場所に置くと便利なことも多く、ファイル名のみのときはそこから読むようになっています。ただ、cnoidファイルの位置を基準にしたいということも分かります。
このあたり、ユースケースがいろいろと出てきましたので、ここいらでいったん整理する必要があるとは考えています。これにあたって、もう少し時間をいただけますでしょうか。

closed 2017-08-16 by s-nakaoka