s-nakaoka/choreonoid #98: cnoidファイルの仕様

y-masutani, 2016-04-08:
- cnoidファイルの中にパス名を書くとき，「\」（バックスラッシュ）を使うと問題が起きます．「"」「"」で括ってあってもダメです．「C:\Users\...」とやってハマりました．「did not find expected hexdecimal number at line XX, colum XX」と表示され，プロジェクトの読み込みが完全に中断されます．「C:\\Users\\...」とするか「C:/Users/...」とすれば回避できることを確認しました．

- この時，エラーメッセージの行数が1行少なく表示されます（最初の行が第0行）．このため余計に混乱しました．

- cnoidファイルの中で使える変数を教えてください．「${SHARE}」以外にもありますか？

- 変数${SHARE}はcnoidファイルを保存するときに，実際のパス名に置き換えられる場合とそうでない場合があるようです．置き換えられる場合に，Windowsでは区切り文字が「\」（バックスラッシュ）になるので，上述の問題が起きます．実際のパス名に置き換える必要はないと思いますが，いかがでしょうか？

s-nakaoka, 2016-04-13:
> cnoidファイルの中にパス名を書くとき，「\」（バックスラッシュ）を使うと問題が起きます．

これ（バックスラッシュをエスケープ文字として使う）はYAMLの仕様のようです。
http://www.yaml.org/spec/1.2/spec.html#id2776092

> この時，エラーメッセージの行数が1行少なく表示されます（最初の行が第0行）

そのようですね。修正できないか試してみます。

> cnoidファイルの中で使える変数を教えてください．「${SHARE}」以外にもありますか？

マニュアルの以下のページで説明していますのでご参照ください。
http://choreonoid.org/ja/manuals/1.5/basics/config.html
ここにあるように、変数はユーザが追加設定できます。

> 変数${SHARE}はcnoidファイルを保存するときに，実際のパス名に置き換えられる場合とそうでない場合があるようです．

プロジェクトファイルを読み込んで保存しなおすと、もともと${SHARE}となっていた部分がフルパスに展開されて保存されるということでしょうか。置き換えられる場合とそうでない場合の例としては、どのようなものがありますでしょうか？

s-nakaoka, 2016-04-13:
> 置き換えられる場合に，Windowsでは区切り文字が「\」（バックスラッシュ）になるので，上述の問題が起きます．

これは直さなければいけませんね。
プロジェクトファイルへの保存時には常にスラッシュを使う形式に変換するようにしたいと思います。

y-masutani, 2016-04-14:
>プロジェクトファイルを読み込んで保存しなおすと、もともと${SHARE}となっていた部分がフルパスに展開されて保存されるということでしょうか。置き換えられる場合とそうでない場合の例としては、どのようなものがありますでしょうか？

cnoidファイルをエディタで編集することを想定しています．

${SHARE}のままの例
```
modelFile: "${SHARE}/model/misc/floor.wrl"
```
置き換えられる例
```
confFileName: "${SHARE}/../../ChoreonoidAddition/turtlebot.conf"
```
は，Choreonoidで読み込んで保存すると，
```
confFileName: "C:\XXX\XXX\XXX\XXX\Choreonoid\share/../../ChoreonoidAddition/turtlebot.conf"
```
に置き換わりました．XXXは伏字です．