s-nakaoka/choreonoid #100: Windowが真っ白

y-masutani, 2016-04-14:
WindowsでビルドしたChoreonoid-1.5で起動すると，ウィンドウの中が真っ白になる現象が頻発しています．
メニューバーは出ており，表示→ビューの表示では何もチェックされていない状態です．また，その中の項目を選んでもチェックが入りません．また，ファイル→プロジェクトの読み込みを選ぶと，ファイルを選択することはでき，最下行には「読み込みに成功しました」と表示されます．

調べたところ，
```
C:\Users\ユーザ名\AppData\Roaming\Choreonoid\Choreonoid.conf
```
を削除すれば，直ることを見つけました．

しかし，かなり頻繁に起こるので不便です．対処していただけると助かります．

s-nakaoka, 2016-04-14:
可能であれば、この症状が起きた時のChoreonoid.confファイルをいただけますでしょうか。

Ubuntu上でも似たような症状がたまに出るのですが、どうもウィンドウを最大化しているときに出る可能性があるようで、その場合最大化の解除（タイトルバーの四角いボタン）をすると直るようでした。その後は最大化しても大丈夫なので、四角いボタンを２回連打することで対処していました。Windowsでも同様の操作で回復できるか試してみていただければと思います。

s-nakaoka, 2016-04-14:
最大化してないときにも出ることがあるようです。その場合も、いったん最大化とかをすると直ります。

本件できれば直したいと思うのですが、以前調査した時は原因が分からず断念しました。ただ、頻繁に起こるとなると問題ですので、再度調査してみようと思います。

s-nakaoka, 2016-07-12:
遅くなりましたが、ここしばらく本件の解消に取り組んでいました。起動時の画面レイアウトは、ウィンドウのリサイズや状態変化（最大化等）のイベントを頼りに行っているのですが、それらのイベントの発生パターンが環境やタイミングによってまちまちになっていて、それに対応しきれていないのが原因でした。ひととおり対応したところ、こちらの環境ではこの不具合が出なくなりましたので、恐らく改善できたのではないかと思います。

closed 2016-07-23 by s-nakaoka