s-nakaoka/choreonoid #106: シミュレーションのログの書き出し方法

YoheiKakiuchi, 2016-06-06:
シミュレーションでのログのファイル書き出しの方法について、マニュアルには以下の記述があります。
「ログファイルアイテム」の具体的な使い方を見つけられないないのですが、使い方をお教えいただけないでしょうか。

http://choreonoid.org/ja/manuals/1.5/simulation/execution-and-playback.html
> 注釈 長時間のシミュレーションとその記録が必要な場合は、他に「ログファイルアイテム」を用いて結果を随時ファイルに書き出して記録することも可能です。この場合、メモリ容量が十分でなくても、ファイルシステムに十分な空き領域があればOKです。

s-nakaoka, 2016-06-07:
すみません、まだマニュアルに書けていませんでした。
使い方は以下のとおりです。

1. 結果を保存したいシミュレーションのシミュレータアイテムの小アイテムとして「ワールドログファイル」アイテムを生成する。
2. このアイテムのプロパティで「ログファイル名」を設定する。
3. 「記録フレームレート」については0を指定すると、シミュレーションと同じフレームレート（分解能）で記録する。ここに値を設定すると、そのフレームレートで記録する。シミュレーションと同じ程の分解能が必要ない場合は、少なめのフレームレートを設定しておくとファイルのサイズや記録のオーバーヘッドをおさえることができる。
4. 以上の状態でシミュレーションを開始すると、シミュレーション中に随時指定したファイルに結果が追記されていく。Choreonoidが途中で落ちても、それまで記録したファイルは有効。
5. 記録の再生については、このアイテムを選択状態にして再生機能を使えば、記録ファイルが存在すればそれが再生される。なので、このアイテムを含む状態でプロジェクトファイルを作成すると、それを読み込んだ時に前回の結果のファイルがあればそれを再生することができる。
6. 「タイムスタンプ接尾辞」をtrueにしておくと、ファイル名にシミュレーション開始時のタイムスタンプがつくようになる。シミュレーションを何回もする場合に、前回の結果を上書きしないで、ログをどんどん追加していきたい場合は、この機能が便利。ただしこの場合、プロジェクトを読み込みなおした時は、前回のログを再生できない。再生するには、そのログファイルを「ログファイル」の読み込みで読み込んでから、それを選択して再生する。

以上のように利用できます。

ただし、この機能はJVRCのときの結果記録用に急遽作ったため、ファイルのフォーマットが十分練られておらず、今後変更する可能性があります。その場合、過去のログデータはみれなくなるかもしれません。（しばらくその予定はありませんが･･･。なにせ急遽作ったものでして。しかもJVRCでは結局使われませんでしたが。）

YoheiKakiuchi, 2016-06-08:
ありがとうございます。
ログは取ることが出きました。

このログに含まれているものは、何になりますでしょうか？
画像データや、CONSTRAINT_FORCEやEXTERNAL_FORCEについても含まれておりますでしょうか？

s-nakaoka, 2016-06-10:
> 画像データや、CONSTRAINT_FORCEやEXTERNAL_FORCEについても含まれておりますでしょうか？

含まれるのは、リンクの位置姿勢、関節変位と、デバイスの状態です。
従いまして、CONSTRAINT_FORCEやEXTERNAL_FORCEのデータは含まれません。
カメラやレンジセンサもデバイスなのですが、それらの状態として保存されるのは解像度や画角といったパラメータ的なものとなっていて、画像データや計測データは現状では保存されません。

YoheiKakiuchi, 2016-06-15:
ありがとうございます。
ログの取り方が分かりましたので閉じます。

closed 2016-06-15 by YoheiKakiuchi