s-nakaoka/choreonoid #107: VirtualRobotRTCに同期したRTCの作り方

YoheiKakiuchi, 2016-06-06:
BodyRTCに設定したコントローラは、VirtualRobot (robotname.rtcとしてできる) に同期して
起動されると思うのですが、同じように同期するRTCは作れるでしょうか？
目的は、CONSTRAINT_FORCEなどを取りこぼし無く取り込みたいということがあります。
periodic_rateを上げてもまれに取りこぼすことがあります。

s-nakaoka, 2016-06-07:
シミュレーション開始時にonActivateが呼ばれて、その後シミュレーションの0フレーム目からonExecuteが呼ばれるようにしたい、ということでしょうか？

ちょっと確認できていない部分もあってうまく動くか自信がないのですが、以下のようにすれば実現できそうな気がします。

1. 同期させたいRTCを、Choreonoid上で"RTCItem"として生成するようにします。RTCItemを生成して、そこの「RTCモジュール名」でRTCのファイルを指定しておくと、そのRTCのインスタンスが生成されます。
2. 上記RTCの「実行タイプ」をChoreonoidExecutionContextにする
3. ポートの接続設定をする。これは現在のところ、BodyRTCItem用のconfigファイルでconnectionという記述をすることでできます。

以上のようにすると、設計上はRTCItemで生成したRTCのonActivate, onExecuteがシミュレーションに同期して呼ばれるはずです。

ただこの機能を現状では試せていないので、うまく動かないかもしれません・・・。その場合はこちらでも同様のサンプルを作って検証・修正をしたいと思いますので、お知らせください。

なお、同期はしませんが、それ以外はOpenRTM-TankJoystick.cnoid というサンプルが上記とほぼおなじことをしていますので、参考になるかもしれません。

hattorishizuko, 2016-06-10:
上の方法ですが、バグがあって
「実行タイプ」をChoreonoidExecutionContextに設定して、プロジェクトファイルに保存し、
次回、そのプロジェクトファイルを呼び出してもChoreonoidExecutionContextに設定されません。

プロパティビューで一度、別の実行タイプを選択し、リターンキーを押すか他の場所をクリックするなどしてセルからフォーカスを外すと、RTCが作り直されます。メッセージビューにその旨が表示されます。
再度、実行タイプにChoreonoidExecutionContextを選択し上と同様にしてRTCを作りなおしてください。

バグは早急に修正しておきます。

closed 2017-08-16 by s-nakaoka