s-nakaoka/choreonoid #108: CONSTRAINT_FORCEの使い方等に関して

snozawa, 2016-06-07:
お世話になっております。
CONSTRAINT_FORCEなどにつきまして何点か質問がございます

1. CONSTRAINT_FORCEの仕様がchoreonoidのマニュアルで探せなかったのですが、https://fkanehiro.github.io/openhrp3-doc/jp/controller_bridge.html と同様に接触点3Dと接触力3Dが連なったものでしょうか。また、これらは親リンク・子リンクから伝播する力以外の外力となりますでしょうか。
2. CONSTRAINT_FORCEには、http://choreonoid.org/ja/manuals/1.5/simulation/interaction.html の引っ張り操作のように与えた力もはいりますでしょうか。
3. ロボットに外乱を与えるため、ある時刻・あるリンクにある力積（例えば力・時間を指定）を与えるようなプログラム・プラグイン・機能はございますでしょうか。

よろしくお願いします。

s-nakaoka, 2016-06-07:
> CONSTRAINT_FORCEの仕様がchoreonoidのマニュアルで探せなかったのですが、https://fkanehiro.github.io/openhrp3-doc/jp/controller_bridge.html と同様に接触点3Dと接触力3Dが連なったものでしょうか

はい、それと同じ内容となっています。

> また、これらは親リンク・子リンクから伝播する力以外の外力となりますでしょうか。

伝播する力以外の外力として、重力や、LinkオブジェクトのF_ext()で指定した力等もあるかと思うのですが、CONSTRAINT_FORCEに入るのは、接触や拘束に伴って発生する力のみとなります。

> CONSTRAINT_FORCEには、http://choreonoid.org/ja/manuals/1.5/simulation/interaction.html の引っ張り操作のように与えた力もはいりますでしょうか。

そちらはLinkオブジェクトのF_ext()メンバで指定していて、CONSTRAINT_FORCEの出力には入りません。

> ロボットに外乱を与えるため、ある時刻・あるリンクにある力積（例えば力・時間を指定）を与えるようなプログラム・プラグイン・機能はございますでしょうか。

はい、SimulatorItemのsetExternalForceという関数を使ってこれを実現できます。

これはPythonからも使えるようになっていて、そのサンプルとして、"SR1WalkPush.cnoid"というプロジェクトがあります。これを読み込んでプロジェクトを開始するとロボットが歩きますので、その途中でツールバーのスクリプト実行ボタンを押してください。するとSR1WalkPush.py が実行されて、上記関数を使って外力が与えられ、ロボットが前方に転倒するかと思います。

snozawa, 2016-06-07:
ありがとうございます。  

> 伝播する力以外の外力として、重力や、LinkオブジェクトのF_ext()で指定した力等もあるかと思うのですが、CONSTRAINT_FORCEに入るのは、接触や拘束に伴って発生する力のみとなります。

これらのCONSTRAINT_FORCEで取得できない成分の外力を取得・保存する方法はございますでしょうか。
現在、CONSTRAINT_FORCEはhrpsys-baseのDataLoggerで取得できるところまできておりまして、
例えばF_ext()で与えた外力もロギングできればと考え手ろいます。

> はい、SimulatorItemのsetExternalForceという関数を使ってこれを実現できます。

ありがとうございます。
こちらも早速試してみます。

s-nakaoka, 2016-06-07:
> これらのCONSTRAINT_FORCEで取得できない成分の外力を取得・保存する方法はございますでしょうか。

F_ext()の値は これらのCONSTRAINT_FORCEで取得できない成分の外力を取得・保存する方法はございますでしょうか。
> これらのCONSTRAINT_FORCEで取得できない成分の外力を取得・保存する方法はございますでしょうか。

F_ext()の値はEXTERNAL_FORCEを指定したポートから取ってくることができます。

snozawa, 2016-06-08:
> F_ext()の値はEXTERNAL_FORCEを指定したポートから取ってくることができます。

すいません、こちらは具体的にどのようにプログラムを修正すればよろしいでしょうか。
confファイルに
```
out-port = F_BODY:WAIST:CONSTRAINT_FORCE
```
とかいてCONSTRAIT_FORCEのポートができてCONSTRAINT_FORCEがシミュレータから取得できてる状況で、
```
out-port = E_BODY:WAIST:EXTERNAL_FORCE
```
を追記しても追加されてないようです。

hattorishizuko, 2016-06-09:
中岡の発言と矛盾して申し訳ないのですが、
EXTERNAL_FORCEは、仕様では出力ポートとして指定できないことになっています。

ですが、ソースコードを一部書き換えることでできるかもしれません。
src/OpenRTMPlugin/VirtualRobotRTC.cpp　の　136行あたり

case CONSTRAINT_FORCE:
case EXTERNAL_FORCE:       <-この行を追加
    ret = registerOutPortHandler(new LinkDataOutPortHandler(portInfo));
    break;

としてみてください。ポートは作成されると思います。

snozawa, 2016-06-09:
ありがとうございます。
追加して値が取得できることを確認しました。

snozawa, 2016-06-11:
https://github.com/s-nakaoka/choreonoid/issues/108#issue-158827555
の質問点が解決しましたのでcloseします。
ありがとうございました。

closed 2016-06-11 by snozawa