s-nakaoka/choreonoid #109: トルク制御などのときに可動域リミットをもうける方法

snozawa, 2016-06-07:
現在たしか、VRML記載の関節の可動域のllimit ulimitを越えてシミュレーション
上のロボット関節が動いていたように思います。
可動域に達したときにシミュレーション上のロボット関節が動かないようにする方法はありますでしょうか。
シミュレーションではトルクモードを使っております。

よろしくお願いします。

s-nakaoka, 2016-06-07:
ODEやBullet等のエンジンではこの拘束をサポートしていて、ODEPluginのODESimulatorItemやBulletPluginのBulletSimulatorItemでシミュレーションするとllimit, ulimitの範囲外には動かなくなります。

産総研エンジン(AISTSimulatorItem)ではこれをサポートしていません。いちおう実装方法は分かっているのですが、まだ手がついていない状況です。

snozawa, 2016-06-08:
ありがとうございます。
トルク制御モードで、関節可動域から逸脱しないようにしたく、
> ODEやBullet等のエンジンではこの拘束をサポートしていて、ODEPluginのODESimulatorItemやBulletPluginのBulletSimulatorItemでシミュレーションするとllimit, ulimitの範囲外には動かなくなります。

のかわりに関節PDサーボ系などシミュレータへのトルク入力を計算しているところで、
可動域に近づくときにはそこから遠ざかるように（バリア関数のようなもので）トルクを発生させるので
擬似的に代用できたりするものなのでしょうか。

s-nakaoka, 2016-06-08:
> 擬似的に代用できたりするものなのでしょうか。

まだ試したことはないのですが、恐らくできると思います。
トルクが大きくなりすぎてシミュレーションが破綻しないよう、反発トルクに最大値を設定しておくとよいかと思います。

あと本件とは直接関係ないのですが、私の方でODEPluginに言及してしまったので、念の為以下の注意をここに書かせてください。

Choreooidのプロセス上でOpenHRP3.1のライブラリを用いたプログラム（hrpsys関連のRTC等）を走らせる場合、ODEPluginが読み込まれていると落ちてしまいます。これはOpenHRP, ODEのどちらでも内部でOPCODEライブラリを使っていて、さらにOpenHRPのものは独自にカスタマイズしていたりするのですが、どちらも共有ライブラリがシンボルを公開してしまっており、両者の関数が混ざって呼ばれることによるものです。

先ほど私自身がこの件でハマってしまっていたため、ここにも書いておかねばと思った次第です。

s-nakaoka, 2016-06-08:
本件と関係なくて恐縮ですが、ODEPluginの件に補足です。Choreonoidも内部の干渉チェッカでOPCODEを使っているのですが、上記の問題はありません。これは共有ライブラリ内の関数について公開するものを明示的に指定し、他は全て非公開としているためです。

snozawa, 2016-06-08:
ありがとうございます。
ODEPlugin, OPCODEの使用上の中尉も承知しました。

closed 2016-06-08 by snozawa