s-nakaoka/choreonoid #111: AccelerationSensorの出力

YoheiKakiuchi, 2016-06-15:
wrlファイルでAccelerationSensorセンサーを設定するときに、rotationを入れているモデルがあるのですが、
そのモデルを使ったVirtualRobotRTCから ACCELERATION_SENSOR2 を指定してできたポートの出力が、
回転が考慮されていないように思われます。
具体的には、センサにrotationを入れて傾けて取り付けられているのに、出力はz軸方向に重力加速度がでております。
 
ちゃんと確認できておりませんが、
センサーGyro (RATE_GYRO_SENSOR2) についても同じではないかと思います。

snozawa, 2016-06-16:
少し話がそれますが、こちらの件も含めて、choreonoidの基本機能等のテストのようなものはどちらかにあったりしますでしょうか。
また、travisかテストサーバのようなものもございますでしょうか。

s-nakaoka, 2016-06-16:
> 回転が考慮されていないように思われます。

ご指摘ありがとうございます。
本問題について、ソース上で確認し、修正しました。
rotationを入れたモデルでの検証は出来ていないのですが、まずは修正を早めに使えるようにした方がよいかと思い、この状態でgithubにアップさせていただきました。

なお、これまで加速度センサの値は速度値を微分してカルマンフィルタをかけるというトリッキーな計算が行われていました。これはこの部分についてOpenHRP2の遺産がそのままになっていたからなのですが、今回これを順動力学計算の結果として出てくる加速度を直接用いる素直な実装にあらためました。

現在デフォルトではこの新しい実装が使われるようにしてあります。ただし問題が出ないとも限らないので、念の為これまでの計算方式も使えるようにしました。AISTSimulatorItemの"Old accel sensor mode"というプロパティをtrueにするとそれが有効になります。（そちらも回転の問題は修正したつもりです）。しばらく問題が出ないか様子をみて、この機能は廃止したいと思います。

> センサーGyro (RATE_GYRO_SENSOR2) についても同じではないかと思います

こちらはソース上では回転も考慮されているようにみえましたので、とりあえずそのままにしてあります。

s-nakaoka, 2016-06-16:
> 少し話がそれますが、こちらの件も含めて、choreonoidの基本機能等のテストのようなものはどちらかにあったりしますでしょうか。

まだないのが現状です。
そのようなものの必要性は認識しておりまして、なんとかしたいとは思っています。
まずはテストやCIのフレームワークを導入する必要があると思うのですが、何かおすすめのものってありますでしょうか？Travisがよいという話も聞いたのですが、GUIやグラフィックス関連のテストには使えなさそうとも聞きました。

snozawa, 2016-06-16:
ありがとうございます。

> そのようなものの必要性は認識しておりまして、なんとかしたいとは思っています。
> まずはテストやCIのフレームワークを導入する必要があると思うのですが、何かおすすめのものってありますでしょうか？Travisがよいという話も聞いたのですが、GUIやグラフィックス関連のテストには使えなさそうとも聞きました。

そうですね。
travisやjenkinsでしょうか。
githubを使う上ではtravisがやりやすいと思います。
一方で、travisではおっしゃるとおり制限が多いので、所々jenkinsも理容しています。
具体的には、jenkinsサーバが別途ありまして、travisからjenkinsのjobをなげるようにしていたと思います。

k-okada, 2016-06-17:
travis上のguiテストは以下に情報があります．
https://docs.travis-ci.com/user/gui-and-headless-browsers/

個人的にはGUIとライブラリを分離すると，このへんの問題も少し楽になるとは思います．

githubでつかえるCIツールは以下にまとまっています．
hrpsysみたいにかなり条件分けてテストを走らすにはTravisが楽だと思います．
一方，テストの条件を分けることがない場合はCircleCIにすると，テストが終了してから30分はsshで接続できるのでデバッグが楽になります．
全部試したわけではないので，他にももっといいのがあるかもしれません．
https://github.com/integrations/feature/code

s-nakaoka, 2016-06-20:
野沢先生、岡田先生、情報ありがとうございます。
CIを行うツールもWebで利用できるものも含めていろいろあるのですね。
今はChoreonoid1.5のリリースと、産総研エンジンの高速化に向けた作業に取り組んでいて、そちらを優先させる必要があるのですが、そのあたりに目処がついたらCIの導入に取り組みたいと思います。

> 個人的にはGUIとライブラリを分離すると，このへんの問題も少し楽になるとは思います．

いちおうChoreonoidでは内部的にはライブラリとGUIが分離されています。例えば、src/Util や src/Body, src/AISTCollisionDetector 等はGUIやOpenGL等には依存しないライブラリとなっています。そしてCMakeの設定で ENABLE_GUI というのがあって、それをオフにするとライブラリ部分だけをコンパイルするようになっています。（ただ、今試したところこのオプションがうまく動かないようでしたので、それは直したいと思います。）

YoheiKakiuchi, 2016-06-22:
遅くなりました。

ジャイロ、加速度センサとも正しく出力されているように確認されました。
ありがとうございます。

closed 2016-06-22 by YoheiKakiuchi