s-nakaoka/choreonoid #113: IndexedLineSetだけのモデルを描画しようとすると落ちる

fkanehiro, 2016-07-01:
```
#VRML V2.0 utf8
#IndexedLineSet example
Shape {
	geometry IndexedLineSet {
		coord Coordinate {
                    point [
				-1.0 -1.0 0.0,	#vertex 0
				1.0 1.0 0.0,	#vertex 1
				1.0 -1.0 0.0,	#vertex 2
                    ]
                }
		color Color {
                    color [
                        1.0 0.0 0.0,	#red
				0.0 1.0 0.0,	#green
                        0.0 0.0 1.0,	#blue
                    ]
            }
		coordIndex [
			#red line
			 0, 1, -1,
			#green line
			 1, 2, -1,
			#blue line
			 2, 0, -1
			]
		colorIndex [
			0, 1, 2
			]
			
	}
}
```
というVRMLを読み込んで、Bodyアイテムにチェックを入れるとchoreonoidが落ちます。
その時のバックトレースは次の通りです。
```
(gdb) bt
#0  0x00007ffff7abdb93 in cnoid::GLSceneRendererImpl::renderPlot(cnoid::SgPlot*, cnoid::SgVectorArray<Eigen::Matrix<float, 3, 1, 0, 3, 1>, std::allocator<Eigen::Matrix<float, 3, 1, 0, 3, 1> > >&, unsigned int) ()
   from /home/kanehiro/openrtp/bin/../lib/libCnoidBase.so.1.5
#1  0x00007ffff7abe218 in cnoid::GLSceneRendererImpl::visitLineSet(cnoid::SgLineSet*) () from /home/kanehiro/openrtp/bin/../lib/libCnoidBase.so.1.5
#2  0x00007ffff5b05c90 in cnoid::SceneVisitor::visitGroup(cnoid::SgGroup*) [clone .localalias.3] () from /home/kanehiro/openrtp/bin/../lib/libCnoidUtil.so.1.5
#3  0x00007ffff7abd925 in cnoid::GLSceneRenderer::visitGroup(cnoid::SgGroup*)
    () from /home/kanehiro/openrtp/bin/../lib/libCnoidBase.so.1.5
#4  0x00007ffff5b05c90 in cnoid::SceneVisitor::visitGroup(cnoid::SgGroup*) [clone .localalias.3] () from /home/kanehiro/openrtp/bin/../lib/libCnoidUtil.so.1.5
#5  0x00007ffff7abd925 in cnoid::GLSceneRenderer::visitGroup(cnoid::SgGroup*)
    () from /home/kanehiro/openrtp/bin/../lib/libCnoidBase.so.1.5
#6  0x00007ffff7ac12fa in cnoid::GLSceneRenderer::visitTransform(cnoid::SgTransform*) () from /home/kanehiro/openrtp/bin/../lib/libCnoidBase.so.1.5
#7  0x00007ffff5b05c90 in cnoid::SceneVisitor::visitGroup(cnoid::SgGroup*) [clone .localalias.3] () from /home/kanehiro/openrtp/bin/../lib/libCnoidUtil.so.1.5
#8  0x00007ffff7abd925 in cnoid::GLSceneRenderer::visitGroup(cnoid::SgGroup*)
    () from /home/kanehiro/openrtp/bin/../lib/libCnoidBase.so.1.5
#9  0x00007ffff7ac12fa in cnoid::GLSceneRenderer::visitTransform(cnoid::SgTransform*) () from /home/kanehiro/openrtp/bin/../lib/libCnoidBase.so.1.5
#10 0x00007ffff5b05c90 in cnoid::SceneVisitor::visitGroup(cnoid::SgGroup*) [clone .localalias.3] () from /home/kanehiro/openrtp/bin/../lib/libCnoidUtil.so.1.5
#11 0x00007ffff7abd925 in cnoid::GLSceneRenderer::visitGroup(cnoid::SgGroup*)
    () from /home/kanehiro/openrtp/bin/../lib/libCnoidBase.so.1.5
```

s-nakaoka, 2016-07-05:
このモデルファイルの場合、colorPerVertexの指定がありませんのでデフォルトのTRUEになります。その場合はVRMLの仕様上、colorIndexはcoordIndexと対応させて、

 colorIndex [
   0, 0, -1,
   1, 1, -1,
   2, 2, -1
 ]

となっている必要があります。

あるいは、

 colorPerVertex FALSE

を指定しておけば、元のcolorIndexでOKとなります。
http://tecfa.unige.ch/guides/vrml/vrml97/spec/part1/nodesRef.html#IndexedLineSet

ただ落ちてしまうのはよろしくないので、対応がついてない場合でも落ちないように修正しました。

fkanehiro, 2016-07-05:
ありがとうございました。

closed 2016-07-05 by fkanehiro