s-nakaoka/choreonoid #114: カスタマイザで可動範囲チェック

fkanehiro, 2016-07-01:
現状のカスタマイザでは、可動範囲をチェックせずに解を1つ返す仕様になっていると思います。
実際には複数の解が存在するため、これまでは肘や膝が逆折れしない、というのをヒントにして解を選択していますが、これをもっと一般的にulimitやllimitにアクセスして解をふるい落とすことは現状のインタフェースで可能でしょうか？

s-nakaoka, 2016-07-05:
現状のインタフェースではできないと思います。

現状のインタフェースは、OpenHRP3の開発時に、HRP-2のIKのコードやゴムブッシュ定数は隠したいが、HRP-2のユーザがIKやブッシュのシミュレーションを使えるようにしたい、ということで設計しました。機能としてもそこに限定していて、さらにOpenHRP3がバージョンアップしてAPIやABIに変更があってもなるべく隠した部分のバイナリはそのまま使えるようにした方がよいかと思い、わざわざC言語でバイナリコンパチビリティが高くなるようなAPIにして設計しました。その分機能は限られていて、カスタマイザの実装も少し面倒なものとなっています。

特定のロボットに依存した実行コードは今後も必要ですし、そこに求められる機能要求も高まっていくでしょうから、この部分改善したいところではあります。バイナリコンパチビリティは今よりは低くなるかと思いますが、できればChoreonoidのBodyライブラリを使って普通にC++でさくさく書けるようなカスタマイザの新しい仕様を作りたいとは思っています。その際、コードやパラメータは隠さずにカスタマイザはなるべくソースで公開してもらうのが望ましいですね。HRPロボットもそのようになことが可能になるとうれしいです。

s-nakaoka, 2016-07-05:
できないと書いたのは、モデルのulimitやllimitにアクセスすることなのですが、そもそもカスタマイザは特定のモデル用に書くものなので、リミット値についても分かっているものとして、その値をカスタマイザにハードコーディングすればそれでOKのような気もします。

fkanehiro, 2016-07-05:
今そうして対処しているのですが、可動範囲が変わった時に2つのファイルを直さないといけないというのはトラブルの元だと思います。

s-nakaoka, 2016-07-05:
それはそうかと思いますが、いずれにしても現状のカスタマイザではモデルのulimit,llimitを取得することはできない仕様となっています。

対応については、上記のようなより汎用的なカスタマイザの仕様を今後設計したいと思います。

fkanehiro, 2016-07-05:
はい、よろしくお願いします。
このIssueは閉じずに残しておきます。