s-nakaoka/choreonoid #121: ubuntu12でコンパイルできない

fkanehiro, 2016-08-16:
ubuntu12でコンパイルしようとすると
```
«BUILDDIR»/choreonoid-1.6.0+dfsg+20160816+662+19~ubuntu12.04.1/src/AISTCollisionDetector/AISTCollisionDetector.cpp:87:5: error: ‘mt19937’ in namespace ‘boost::random’ does not name a type
/«BUILDDIR»/choreonoid-1.6.0+dfsg+20160816+662+19~ubuntu12.04.1/src/AISTCollisionDetector/AISTCollisionDetector.cpp:88:36: error: ‘mt19937’ is not a member of ‘boost::random’
/«BUILDDIR»/choreonoid-1.6.0+dfsg+20160816+662+19~ubuntu12.04.1/src/AISTCollisionDetector/AISTCollisionDetector.cpp:88:36: note: suggested alternative:
/usr/include/boost/random/mersenne_twister.hpp:361:47: note:   ‘boost::mt19937’
```
といったエラーになるのですが、そろそろubuntu12はサポート外でしょうか。

s-nakaoka, 2016-09-05:
すみません、Ubuntu12.04についてはチェックをしていませんでした。
マニュアルの記述が間違っていたので、そこは直しておこうと思います。

上記エラーは干渉チェッカの並列化のために入れたテスト機能で使っていて、なくても動くので、Ubuntu12.04のサポートが必要であれば対応することは可能です。ただしここを直しても今後他に不具合が出てくるかもしれません。開発版はC++11に移行しましたし、Qtも今後Qt5を前提とすることになってくるので、確かにUbuntu12ではそろそろ厳しいかもしれません。ちなみにUbuntu12の必要性はどのあたりにありますでしょうか。

fkanehiro, 2016-09-05:
特にubuntu12が必要な訳ではありません。
PPAでchoreonoidのdebパッケージを配布していますが、ubuntu12がサポート外となったのであればビルド対象から外しますし、意図せずコンパイルできなくなったのであれば、修正していただきたいと思った次第です。
https://code.launchpad.net/~hrg/+recipe/hrg-packaging-choreonoid-daily

s-nakaoka, 2016-09-06:
了解です。
C++11へ移行しようと思っているので、それがUbuntu12.04のgccでサポートできるかをチェックしてから、Ubuntu12.04をもうしばらくサポートするかどうか決めたいと思います。Ubuntu12.04のデフォルトのGCCでC++11対応が不十分なようでしたら、サポートをやめようと思います。

あと上で開発版はC++11に移行したと書いたのですが、これは私の勘違いでした。C++11対応は別ブランチで行っているのですが、まだmasterにはマージしていませんでした。これについて、これからmasterへのマージを進めたいと思います。他のプラグインにも影響が出るかもしれませんが、何卒ご了承ください。

s-nakaoka, 2016-09-07:
Ubuntu12.04で試してみたのですが、まず標準のgccが-std=c++11のオプションを受け付けないようで、c++11に対応していませんでした。

CLangでコンパイルできないかも試したところ、そちらはapt-getで入るものでc++11に対応しているようでしたが、他のboost等のライブラリで対応していない部分があるようで、やはりそのままではコンパイルができませんでした。

この結果から、Ubuntu12.04の今後のサポートはやはり見送ろうと思います。

fkanehiro, 2016-09-07:
了解しました。
ubuntu12はビルド対象から外します。

closed 2016-09-07 by fkanehiro