s-nakaoka/choreonoid #124: VisionRenderer::getRangeCameraData()の仕様

fkanehiro, 2016-08-26:
https://github.com/s-nakaoka/choreonoid/blob/master/src/BodyPlugin/GLVisionSimulatorItem.cpp#L1085
のVisionRenderer::getRangeCameraData()で取得できるポイントクラウドですが、距離が近すぎると原点、距離が遠すぎるとx,yがinf,-inf,nanの何れか、zが-infという座標が返ってくるようですが、これは何かに合わせてこういう仕様なのでしょうか？

fkanehiro, 2016-08-26:
計測できないところは全てNaN、ではだめでしょうか。

s-nakaoka, 2016-09-05:
遅くなってしまいすみません。

どうしてこうしたのかなかなか思い出せず、RTMのpointcloud.idlやPCLのpcl::PointCloudクラスを見なおしてみたのですが、無効な点の表現方法については特に記述はありませんし、これらに合わせたというわけでもなさそうです。

そこでさらに考えていたのですが、どうもこのようにしておくと、無効な点について場合分けをしなくても、そのままいろんなデータ処理が適用できるという利点があるので、そうしたような気がしてきました。

例えばアライメントとかするときにも、NaNが混じってるとヘタしたら落ちそうですが、無限遠ということならそれに対して計算してもいちおう無限大という距離が出てくるので、ICPとかもそのまま動くかもしれません。

あるいは、バウンディングボックスのようなもので範囲を指定して点群を抜き出すという時に、バウンディングボックスの不等号による判定がそのまま使えます。Choreonoidでも領域を指定して点を抜き出す機能がありますが、それの実装も考慮してこうしたのではないかと思います。

NaNとNaNでない数値を比較すると、例外とかじゃなくて、ただfalseになるのでしたっけ？だとしたら比較に関しては確かにNaNでもよいのかもしれません。でも距離を出そうとするとNaNになるでしょうから、その値が後々悪さをするということもあり得ますね。

ちなみにこれでも無効な点かどうかの判定はZ座標が-infinity(奥行き無限大）かどうかで判定できそうです。

という考えがあってこうしたのではないかという気がするのですが、他のライブラリ等も含めた標準的なやり方はNaNにするということでしたら、むしろそちらに変更したいと思います。どこかにそういったドキュメントとかないものでしょうか？

fkanehiro, 2016-09-05:
僕も計測できないところをどうするのが正しい仕様なのかちょっと探したのですが見つかりませんでした。記憶が定かではありませんが、kinectではnanだったような気がします。そうでなくてinfや-infにした方がよいのではないかとのことですが、一部nanになっています（0*inf=nanのため）。できればこれをなくして頂ければ全てnanでも全てinfでもよいのですが（こちらの事情ですがFPEを無くしたいため）。

s-nakaoka, 2016-09-06:
ちょっと調べてみたのですが、実際にはdepthに関して-inf, NaN, +infが混ぜて使われているのが現状のようです。
これに関して、こんなドキュメントがありました。
http://www.ros.org/reps/rep-0117.html

このドキュメントがどういう立ち位置なのか（仕様or要望or相談?)イマイチ分からないのですが、とりあえず現状では上記３つの値が使われていて、気をつけて使いましょうみたいな感じに読めます。

それで、一般的な使われ方としては、
-inf: 手前過ぎて計測できない場合
+info: 計測範囲を超えている場合
NaN: 何らかの不具合があった場合
となっているようです。

不具合というのはハードウェアがうまく動いていないとか、通信がうまくできなかったとか、そいういうことだと思うので、レンジセンサのシミュレーションとしては、infで対応するのがよいように思います。

> 一部nanになっています（0*inf=nanのため）

ここは0になることを意図していたので、そのように修正したいと思います。
その意図するところは、真正面はx,y座標的には0,0だが、その奥行が無限大（計測範囲を超えている）ということで。
真正面以外はそれぞれの領域に傾いているので、無限遠だとx,yもinf or -infになるのかなと。
上記ドキュメントはx,yについては言及していないので、これが標準的な表現方法かは分かりませんが、いずれにしても無効な点かどうかのチェックはz座標をみていただければと思います。

s-nakaoka, 2016-09-06:
ちなみに上記ドキュメントだと、手前すぎるのが-infで、遠すぎるのが+infなのですね。depthデータだけの格納だと、確かにこの符号が自然なのかもしれません。ただ、x,y,zとある場合は、xを右向き、yを上向き、右手系でzは手間向き、というのが（3DCGでは）一般的なようなので、符号もそのようにとっていました。ここはとりあえずこの座標系のままにしておきますが、一般的な座標のとり方と異なっているようでしたらお知らせください。

s-nakaoka, 2016-09-06:
とりあえずNaNが入らないよう修正しました。
ac4ea427b3deeec9e65316ac0858fe4440ce3518

fkanehiro, 2016-09-07:
ありがとうございました。

closed 2016-09-07 by fkanehiro