s-nakaoka/choreonoid #125: 転倒時の力の値について

snozawa, 2016-09-21:
ロボットを転倒させるときのシミュレーションをchoreonoid，関節トルク入力＋ＰＤcontrollerで行っています．
衝撃力をconstraint_forceで計測しています．

K. Fujiwara and F. Kanehiro and S. Kajita and K. Kaneko and  K. Yokoi and H. Hirukawa, "UKEMI: falling motion control to minimize damage to biped humanoid robot", Intelligent Robots and Systems, 2002. IEEE/RSJ International Conference on, 
のopenhrpのときの論文を拝見しまして，同様に衝撃力を
シミュレータで見積っておられたと思います．

こちらでchoreonoidで試したところ，計測できる力がよく１ケタ大きい値になったりしてしまう
ことがあるのですが，上記論文のときにはだいたい同じ条件で転倒させたときには
同じように再現性のあるシミュレーション値が得られましたでしょうか．

また，choreonoid上で同様に転倒シミュレーションしてConstraint_forceをとろうとするとき，
何かとくに気を付けておくべき点はございますでしょうか．

よろしくお願いいたします．

s-kajita, 2016-09-21:
IROS2002の頃のOpenHRPによる衝撃力見積もりは昔の話なので忘却の彼方ですが、
Choreonoidを使った衝撃力シミュレーションにはちょっと注意点があります。

デフォルトの摩擦係数1.0のままだと、衝突時の挙動が不安定で、書かれたような
大きめの値が出る事がありました。なので、摩擦係数0.5にしたところ大体実験結果と
合う値になりました。

　なお、僕はロボット全体の並進運動量をまず計算して、差分をとって床反力を求めました。
ただ、動力学エンジンの内部の実装を考えると直接Constraint_forceを読むのと同じかも
知れません。

何かのご参考になれば幸いです。

snozawa, 2016-09-21:
@s-kajita さん

ご回答いただきまして誠にありがとうございます。

> デフォルトの摩擦係数1.0のままだと、衝突時の挙動が不安定で、書かれたような
> 大きめの値が出る事がありました。なので、摩擦係数0.5にしたところ大体実験結果と
> 合う値になりました。

なるほど、1.0でおこなっていたかもしれませんので、確認してみます。

closed 2017-04-27 by s-nakaoka