s-nakaoka/choreonoid #127: シミュレーション時にchoreonoidが落ちる

YoheiKakiuchi, 2016-09-30:
簡単な再現パターンを作れておりませんが、こちらで行っているシミュレーションにおいて
時々choreonoidがSegmentation faultします。
まずは、報告させていただきます。

BodyRTCにhrpsys-baseを接続しており、それ以外に、画像系センサと関節角度などをBodyRTCから読んで
ROSのメッセージとして出力するRTCを接続しています。
(このパッケージ https://github.com/start-jsk/rtmros_choreonoid のJVRC時と同じような構成になっております）

現状ではすべてをつないだ場合のみ問題が起こっているように見えます。
（例えば画像系のRTCを抜くと問題が起こらない。
 画像系のRTCを入れていても、他のRTC郡を抜くと、やはり問題は起こらない。）

gdbの出力は以下のとおりとなります。

```
Program terminated with signal SIGSEGV, Segmentation fault.
#0  malloc_consolidate (av=av@entry=0x7fdf34000020) at malloc.c:4163
4163	malloc.c: そのようなファイルやディレクトリはありません.
(gdb) bt
#0  malloc_consolidate (av=av@entry=0x7fdf34000020) at malloc.c:4163
#1  0x00007fdfb932ece8 in _int_malloc (av=0x7fdf34000020, bytes=8224) at malloc.c:3423
#2  0x00007fdfb93316c0 in __GI___libc_malloc (bytes=8224) at malloc.c:2891
#3  0x00007fdfb5066341 in ?? () from /usr/lib/nvidia-352/tls/libnvidia-tls.so.352.63
#4  0x00007fdfb98e8dad in operator new (sz=8224) at ../../../../src/libstdc++-v3/libsupc++/new_op.cc:51
#5  0x00007fdfb98e8ea9 in operator new[] (sz=<optimized out>)
    at ../../../../src/libstdc++-v3/libsupc++/new_opv.cc:32
#6  0x00007fdf8e348c70 in omni::giopStream_Buffer::newBuffer(unsigned int) ()
   from /usr/lib/choreonoid-1.6/../libomniORB4.so.1
#7  0x00007fdf8e36010d in omni::giopImpl12::outputNewMessage(omni::giopStream*) ()
   from /usr/lib/choreonoid-1.6/../libomniORB4.so.1
#8  0x00007fdf8e360149 in omni::giopImpl12::outputMessageBegin(omni::giopStream*, void (*)(omni::giopStream*)) ()
   from /usr/lib/choreonoid-1.6/../libomniORB4.so.1
#9  0x00007fdf8e352b85 in omni::GIOP_S::SendReply() () from /usr/lib/choreonoid-1.6/../libomniORB4.so.1
#10 0x00007fdf8e335f92 in omniCallHandle::upcall(omniServant*, omniCallDescriptor&) ()
   from /usr/lib/choreonoid-1.6/../libomniORB4.so.1
#11 0x00007fdf8f758fd0 in RTC::_impl_PortService::_dispatch(omniCallHandle&) ()
   from /usr/lib/x86_64-linux-gnu/libRTC-1.1.2.so
#12 0x00007fdf8e32807d in omni::omniOrbPOA::dispatch(omniCallHandle&, omniLocalIdentity*) ()
   from /usr/lib/choreonoid-1.6/../libomniORB4.so.1
#13 0x00007fdf8e30cb38 in omniLocalIdentity::dispatch(omniCallHandle&) ()
   from /usr/lib/choreonoid-1.6/../libomniORB4.so.1
#14 0x00007fdf8e351109 in omni::GIOP_S::handleRequest() () from /usr/lib/choreonoid-1.6/../libomniORB4.so.1
#15 0x00007fdf8e351f48 in omni::GIOP_S::dispatcher() () from /usr/lib/choreonoid-1.6/../libomniORB4.so.1
#16 0x00007fdf8e34e925 in omni::giopWorker::real_execute() () from /usr/lib/choreonoid-1.6/../libomniORB4.so.1
#17 0x00007fdf8e34f03f in omni::giopWorker::execute() () from /usr/lib/choreonoid-1.6/../libomniORB4.so.1
#18 0x00007fdf8e2ff81d in omniAsyncWorkerInfo::run() () from /usr/lib/choreonoid-1.6/../libomniORB4.so.1
#19 0x00007fdf8e2ffe4f in omniAsyncWorker::run(void*) () from /usr/lib/choreonoid-1.6/../libomniORB4.so.1
#20 0x00007fdf8e039673 in omni_thread_wrapper () from /usr/lib/choreonoid-1.6/../libomnithread.so.3
#21 0x00007fdfb7cef184 in start_thread (arg=0x7fdf3bfff700) at pthread_create.c:312
#22 0x00007fdfb93a937d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
```
https://github.com/start-jsk/rtmros_choreonoid/issues/211

s-nakaoka, 2016-09-30:
mallocで落ちているので、どこかでメモリ破壊が起きているのかもしれませんが、途中でlibnvidia-tlsの関数を経由しているというのも気になりますね。なぜmallocするのにnvidiaのライブラリが出てくるのか・・・。

omniORBに関しては、ご存知かもしれませんが、バッファサイズの設定があります。/etc/omniORB.cfg にgiopMaxMsgSizeという項目があって、このデフォルト値が少なめなので、うちでHRP-2改のシミュレーションを行う際にはこの値を大きめ(64MBとか）にしています。これが今回の問題と関係しているかは分かりませんが、多めに設定しておいた方がよいのは確かなので、これを増やすとひょっとしたら解決するかもしれません。

mttamtam, 2016-09-30:
私も YoheiKakiuchi さんと同じ使い方をしていて、同じように時々choreonoidがSegmentation Faultをはいて落ちる現象に遭遇します。
ご指摘のようにomniORBのバッファサイズについて、
~/.bashrc に、
export ORBgiopMaxMsgSize=2097152000
としているのですが、時々落ちてしまっています。
ほかに何か疑いやすいポイントなどはありますでしょうか。

また、関係があるのかわかりませんが、choreonoidを閉じるか落ちるかしたのち、もう一度立ち上げる前には
pkill -9 choreonoid
としています。

s-nakaoka, 2016-10-03:
なるほど、omniORBのバッファサイズについては既に設定されているのですね。

あとは可能性は低いのですが、ビジョンセンサ関連の描画が悪さをしているとすると、Qtのバージョン(4か5)によって症状が変わるかもしれません。QtでのOpenGLの裏バッファの扱いに違いがあるようでしたので・・・。ただ、画像系のRTCを入れていても、他のRTC郡を抜くと、やはり問題は起こらないとのことですので、やはり可能性は低いかもしれません。

ちなみに、本症状がでるようになったのは最近でしょうか？何かOSのアップグレードとかはありましたでしょうか？Choreonoidに関しては、9月になってgithubの開発版ではC++11に移行しましたので、それが何か関係している可能性もあるかもしれません。

s-nakaoka, 2017-10-02:
すみません、本件解決したか分かりませんが、議論が止まっているので一旦閉じます。

closed 2017-10-02 by s-nakaoka