s-nakaoka/choreonoid #129: ubuntu14.04上でchoreonoid1.6.0をソースからコンパイル＋実行でコアダンプします

k-nakatani, 2016-11-08:
２０１６・１１・０８
ubuntu14.04上でchoreonoid1.6.0をソースからコンパイル＋実行でコアダンプします

JVRCのサイボウズでご相談しておりましたが、金広さんから、[githubのissue]に報告すると調査がし易いと
アドバイスをいただきましたので、こちらに書き込みさせていただきます。
＃指示を取り違えていましたら、申し訳ありません。

■起こっている問題点

ubuntu14.04上でchoreonoid1.6.0をソースからコンパイルして、立ち上げたところ、
coreを吐いて、落ちます。
何が原因なのでしょうか？

よろしくおねがいします。
＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝＝

■使用環境

●OS:
ubuntu 14.04 LTS

●SYSTEM:
メモリ　31.4GiB
プロセッサ　Intel Core i7-6700CPU 3.40GHz x 8
グラフィック　Gallium 0.4 on llvmpipe (LLVM 3.4, 256bits)
OS種別　　64ビット
ディスク         212.3GB

●インストールしているchoreonoid

s-nakaoka/choreonoid から最新をダウンロード

■どのような手順で問題が起きるのか、

１．最新をインストールする
https://github.com/s-nakaoka/choreonoid

２．ソースコードからのビルドとインストール (Ubuntu Linux編)¶
を参考にしながら、インストール作業を進める（作業はすべてrootで実行）
http://choreonoid.org/ja/install/build-ubuntu.html#id11
#　unzip choreonoid-master.zip
→特にエラー、ワーニングなし。
#　cd choreonoid-master

# misc/script/install-requisites-ubuntu-14.04.sh 
（前略）アップグレード: 0 個、新規インストール: 0 個、削除: 0 個、保留: 0 個。
# cmake .
（以下、実行プロセス出力を掲載）→　not foundという表記あり。

-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/x86_64-linux-gnu/libX11.so
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Looking for include file dlfcn.h
-- Looking for include file dlfcn.h - found
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found suitable version "2.7.6", minimum required is "2.7") 
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   program_options
--   regex
--   thread
--   iostreams
--   date_time
--   python
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   chrono
-- checking for module 'eigen3'
--   found eigen3, version 3.2.0
-- Looking for Q_WS_X11
-- Looking for Q_WS_X11 - found
-- Looking for Q_WS_WIN
-- Looking for Q_WS_WIN - not found
-- Looking for Q_WS_QWS
-- Looking for Q_WS_QWS - not found
-- Looking for Q_WS_MAC
-- Looking for Q_WS_MAC - not found
-- Found Qt4: /usr/bin/qmake (found suitable version "4.8.6", minimum required is "4.7.0") 
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.8") 
-- Found PNG: /usr/lib/x86_64-linux-gnu/libpng.so (found version "1.2.50") 
-- Found JPEG: /usr/lib/x86_64-linux-gnu/libjpeg.so  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nakatani10/Downloads/nakaoka1108/choreonoid-master
 
＃ccmake .
(ON/OFF等の設定記録は、すべて画像で保存してありますが、膨大なので、ここでは省略します。）

次のものについて、defaultのOFFからONへ変更しました。
BUILD_OPENRTM_SAMPLES　OFF→ON
BUILD_OPENRTM_PLUGIN
BUILD_CORBA_PLUGIN
ENABLE_CORBA

（作業の仕方としては、先のBUILD_OPENRTM_SAMPLESをONにして、[c]すると、
以下のようなエラーメッセージが出るので、
CMake Error at sample/OpenRTM/CMakeLists.txt:13 (message):
   OpenRTM samples need to build OpenRTMPlugin.
この指定されたものについてONして、また[c]して、、、とエラーメッセージがなくなるまで、設定変更を繰り返す。）

●エラーが出なくなった時のcmake の表示トップのところに*がついていることが気になる。
（がエラーは出ていないので、そのまま続ける）

[g]で終了。

(略)/choreonoid-master# make -j4

（make 実行時のプロセス出力は70%building CXX...あたり以降から保存していますが、膨大なので、ここでは省略します。）

気になる出力としては、
＝＝＝
[ 98%] Building CXX object src/OpenRTMPlugin/CMakeFiles/CnoidOpenRTMPlugin.dir/deprecated/BridgeConf.cpp.o
In file included from /usr/include/openrtm-1.1/rtm/RTC.h:22:0,
                 from /usr/include/openrtm-1.1/rtm/Manager.h:23,
                 from /home/nakatani10/Downloads/nakaoka1108/choreonoid-master/src/OpenRTMPlugin/deprecated/BridgeConf.h:20,
                 from /home/nakatani10/Downloads/nakaoka1108/choreonoid-master/src/OpenRTMPlugin/deprecated/BridgeConf.cpp:6:
/usr/include/openrtm-1.1/rtm/config_rtc.h:165:0: warning: "_REENTRANT" redefined [enabled by default]
 #define _REENTRANT TRUE
 ^
<command-line>:0:0: note: this is the location of the previous definition
＝＝＝
と言った、warning文が複数出ていることです。
このwarningは以下の箇所で出ています。

# grep warning make1107.log 
　
（※make1107.logはmake実行時の出力表示をコピペで保存したもの。70%build以降）

/usr/include/openrtm-1.1/rtm/config_rtc.h:165:0: warning: "_REENTRANT" redefined [enabled by default]
/usr/include/openrtm-1.1/rtm/config_rtc.h:165:0: warning: "_REENTRANT" redefined [enabled by default]
/usr/include/openrtm-1.1/rtm/config_rtc.h:165:0: warning: "_REENTRANT" redefined [enabled by default]
/usr/include/openrtm-1.1/rtm/config_rtc.h:165:0: warning: "_REENTRANT" redefined [enabled by default]
/usr/include/openrtm-1.1/rtm/config_rtc.h:165:0: warning: "_REENTRANT" redefined [enabled by default]
/usr/include/openrtm-1.1/rtm/config_rtc.h:165:0: warning: "_REENTRANT" redefined [enabled by default]
/usr/include/openrtm-1.1/rtm/config_rtc.h:165:0: warning: "_REENTRANT" redefined [enabled by default]
/usr/include/openrtm-1.1/rtm/config_rtc.h:165:0: warning: "_REENTRANT" redefined [enabled by default]
/usr/include/openrtm-1.1/rtm/config_rtc.h:165:0: warning: "_REENTRANT" redefined [enabled by default]
/usr/include/openrtm-1.1/rtm/config_rtc.h:165:0: warning: "_REENTRANT" redefined [enabled by default]
/usr/include/openrtm-1.1/rtm/config_rtc.h:165:0: warning: "_REENTRANT" redefined [enabled by default]
/usr/include/openrtm-1.1/rtm/config_rtc.h:165:0: warning: "_REENTRANT" redefined [enabled by default]
/usr/include/openrtm-1.1/rtm/config_rtc.h:165:0: warning: "_REENTRANT" redefined [enabled by default]
/usr/include/openrtm-1.1/rtm/config_rtc.h:165:0: warning: "_REENTRANT" redefined [enabled by default]
/usr/include/openrtm-1.1/rtm/config_rtc.h:165:0: warning: "_REENTRANT" redefined [enabled by default]
/usr/include/openrtm-1.1/rtm/config_rtc.h:165:0: warning: "_REENTRANT" redefined [enabled by default]
/usr/include/openrtm-1.1/rtm/config_rtc.h:165:0: warning: "_REENTRANT" redefined [enabled by default]
root@nakatani10:/home/nakatani10/Downloads/nakaoka1108/choreonoid-master# 

# ./bin/choreonoid

実行すると、一旦は立ち上がりますが、表示画面を大きくすると、落ちて、
root@nakatani10:/home/nakatani10/Downloads/nakaoka1108/choreonoid-master# ./bin/choreonoid Segmentation fault (コアダンプ)

となります。

■またコアダンプするのであればその時のバックトレースの情報等を合わせて、

gdbを実行して、以下の通り結果を得ましたが、
in 以下が不明で原因がわかりません。

# ulimit -c unlimited
#  gdb -c core
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
[New LWP 30751]
[New LWP 30754]
[New LWP 30755]
[New LWP 30756]
[New LWP 30759]
[New LWP 30753]
[New LWP 30769]
[New LWP 30757]
[New LWP 30762]
[New LWP 30763]
[New LWP 30761]
[New LWP 30760]
[New LWP 30758]
Core was generated by `./bin/choreonoid'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007f26c5b55524 in ?? ()
(gdb) bt
#0  0x00007f26c5b55524 in ?? ()
#1  0x00007f26c89d2000 in ?? ()
#2  0x00007f26c87da961 in ?? ()
#3  0x0000000000000000 in ?? ()
(gdb) 

わかりました。

fkanehiro, 2016-11-08:
表示画面を大きくすると落ちるというのはグラフィックス周りの問題のように思えます。
Gallium 0.4 on llvmpipe (LLVM 3.4, 256bits)というのはOpenGLのソフトウェアエミュレーションのようですが、グラフィックスのチップは何でしょうか。それにあったドライバはないでしょうか。

http://yoshi-1968.com/2013/05/05/gallium-0-4-on-llvmpipe/

k-nakatani, 2016-11-08:
ありがとうございます。
今から、ご指摘のボード周りについて調べてみます。

なお、何度か、表示サイズを変えていたところ、１度だけ、次のようなメッセージが出ました。
何か情報を含んでいるのかもしれないと思い、ご報告いたします。

nakatani10@nakatani10:~/Downloads/nakaoka1108/choreonoid-master$ X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 72 (X_PutImage)
  Resource id:  0x44000a3

k-nakatani, 2016-11-08:
ハードウェアの情報をご報告します。
●PCのハードウェアの情報
Core i7-6700/インテル H170 Express/16GB(8GBﾗ2) /1TBHDD/DVDマルチ_x000D_
GeForce GTX 1060 6GB

fkanehiro, 2016-11-08:
NVIDIAのGPUが載っているようですので、NVIDIAのドライバを使って下さい。
新しいGPUのようですので、NVIDIAのサイトから落としてきて手動で入れる必要があるのではないかと思います。

k-nakatani, 2016-11-08:
わかりました。ドライバのインストールのためにXserverを起動させないところで苦心しています。またご報告します。

k-nakatani, 2016-11-08:
lightdtを止めることでXを止めることができ、まずは、ドライバを、
LINUX X64 (AMD64/EM64T) DISPLAY DRIVER
に置き換えることができました。
結果、数回の画面サイズ変更ではcoreは吐きません。
が、動作についてはこれからの調査です。
またご報告します。

まずは、graphics driverは差し替えることができましたので、ご報告します。

k-nakatani, 2016-11-09:
chorenoid1.6.0を（おそらく）安定的に動かすことが出来ました。ありがとうございました。
ご報告します。

現状、JVRC2015のファイルを動かすのに、1.5.0では動きますが、1.6.0では動かない問題に直面しています。
別スレッドでご相談します。

closed 2016-11-28 by s-nakaoka