s-nakaoka/choreonoid #130: Use after free bug

jun0, 2016-11-08:
Choreonoid を AddressSanitizer をつけてコンパイルすると、何も引数を与えずに立ち上げて閉じるだけで use-after-free が検出されます。ViewManagerImpl が初期化する際、与えられた ExtensionManager が内部的に持っている std::string への参照を捉えているのに、ViewManagerより ExtensionManager が先に解放されるのが直接的な原因のようです。確認できていませんが、恐らく menuManager に関しても同じ問題があると思います。

choreonoid のコンパイルに使ったコマンドは

    cmake -DCMAKE_INSTALL_PREFIX=/home/jun/dvl/openhrp/jenkins/drcutil/openrtp -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_CXX_FLAGS_RELWITHDEBINFO="-O2 -g -DNDEBUG -fsanitize=address" -DCMAKE_C_FLAGS_RELWITHDEBINFO="-O2 -g -DNDEBUG -fsanitize=address"
    make -j2 install VERBOSE=1

choreonoid の呼び出しと ASan の出力は次の通りです。

    $ choreonoid 
    Gtk-Message: Failed to load module "overlay-scrollbar"
    =================================================================
    ==5684==ERROR: AddressSanitizer: heap-use-after-free on address 0x6030001e3be0 at pc 0x7f72efb1a720 bp 0x7ffec6fd7e60 sp 0x7ffec6fd7608
    READ of size 22 at 0x6030001e3be0 thread T0
        #0 0x7f72efb1a71f in memcmp (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x7771f)
        #1 0x7f72ef4e685f in std::char_traits<char>::compare(char const*, char const*, unsigned long) /usr/include/c++/5/bits/char_traits.h:262
        #2 0x7f72ef4e685f in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::compare(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const /usr/include/c++/5/bits/basic_string.h:2318
        #3 0x7f72ef4e685f in bool std::operator< <char, std::char_traits<char>, std::allocator<char> >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) /usr/include/c++/5/bits/basic_string.h:4989
        #4 0x7f72ef4e685f in std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >::operator()(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const /usr/include/c++/5/bits/stl_function.h:387
        #5 0x7f72ef4e685f in equal_range /usr/include/c++/5/bits/stl_tree.h:1696
        #6 0x7f72ef4e685f in erase /usr/include/c++/5/bits/stl_tree.h:2272
        #7 0x7f72ef4e685f in erase /usr/include/c++/5/bits/stl_map.h:745
        #8 0x7f72ef4e685f in cnoid::ViewManagerImpl::~ViewManagerImpl() /home/jun/dvl/openhrp/jenkins/drcutil/src/choreonoid/src/Base/ViewManager.cpp:494
        #9 0x7f72ef4e9af3 in cnoid::ViewManager::~ViewManager() /home/jun/dvl/openhrp/jenkins/drcutil/src/choreonoid/src/Base/ViewManager.cpp:482
        #10 0x7f72ef5f2092 in std::default_delete<cnoid::ViewManager>::operator()(cnoid::ViewManager*) const /usr/include/c++/5/bits/unique_ptr.h:76
        #11 0x7f72ef5f2092 in std::unique_ptr<cnoid::ViewManager, std::default_delete<cnoid::ViewManager> >::~unique_ptr() /usr/include/c++/5/bits/unique_ptr.h:236
        #12 0x7f72ef5f2092 in cnoid::ExtensionManagerImpl::~ExtensionManagerImpl() /home/jun/dvl/openhrp/jenkins/drcutil/src/choreonoid/src/Base/ExtensionManager.cpp:104
        #13 0x7f72ef5f2773 in cnoid::ExtensionManager::~ExtensionManager() /home/jun/dvl/openhrp/jenkins/drcutil/src/choreonoid/src/Base/ExtensionManager.cpp:100
        #14 0x7f72c99265c2 in ~DRCUserInterfacePlugin /home/jun/dvl/openhrp/jenkins/drcutil/src/choreonoid/ext/hrpcnoid/DRCUserInterfacePlugin/DRCUserInterfacePlugin.cpp:30
        #15 0x7f72c99265c2 in ~DRCUserInterfacePlugin /home/jun/dvl/openhrp/jenkins/drcutil/src/choreonoid/ext/hrpcnoid/DRCUserInterfacePlugin/DRCUserInterfacePlugin.cpp:30
        #16 0x7f72ef4a4245 in cnoid::PluginManagerImpl::finalizePlugin(std::shared_ptr<cnoid::PluginManagerImpl::PluginInfo>) /home/jun/dvl/openhrp/jenkins/drcutil/src/choreonoid/src/Base/PluginManager.cpp:804
        #17 0x7f72ef4a4e06 in cnoid::PluginManagerImpl::finalizePlugins() /home/jun/dvl/openhrp/jenkins/drcutil/src/choreonoid/src/Base/PluginManager.cpp:869
        #18 0x7f72ef4a6ad5 in cnoid::PluginManagerImpl::~PluginManagerImpl() /home/jun/dvl/openhrp/jenkins/drcutil/src/choreonoid/src/Base/PluginManager.cpp:224
        #19 0x7f72ef4a6d60 in cnoid::PluginManager::~PluginManager() /home/jun/dvl/openhrp/jenkins/drcutil/src/choreonoid/src/Base/PluginManager.cpp:218
        #20 0x7f72ef4a6d94 in cnoid::PluginManager::finalize() /home/jun/dvl/openhrp/jenkins/drcutil/src/choreonoid/src/Base/PluginManager.cpp:178
        #21 0x7f72ef461c99 in cnoid::AppImpl::exec() /home/jun/dvl/openhrp/jenkins/drcutil/src/choreonoid/src/Base/App.cpp:320
        #22 0x40131c in main /home/jun/dvl/openhrp/jenkins/drcutil/src/choreonoid/src/Choreonoid/main.cpp:24
        #23 0x7f72ed82082f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
        #24 0x4014f8 in _start (/home/jun/dvl/openhrp/jenkins/drcutil/openrtp/bin/choreonoid+0x4014f8)
    
    0x6030001e3be0 is located 0 bytes inside of 23-byte region [0x6030001e3be0,0x6030001e3bf7)
    freed by thread T0 here:
        #0 0x7f72efb3cb2a in operator delete(void*) (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x99b2a)
        #1 0x7f72ef5f2063 in __gnu_cxx::new_allocator<char>::deallocate(char*, unsigned long) /usr/include/c++/5/ext/new_allocator.h:110
        #2 0x7f72ef5f2063 in std::allocator_traits<std::allocator<char> >::deallocate(std::allocator<char>&, char*, unsigned long) /usr/include/c++/5/bits/alloc_traits.h:517
        #3 0x7f72ef5f2063 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_destroy(unsigned long) /usr/include/c++/5/bits/basic_string.h:185
        #4 0x7f72ef5f2063 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_dispose() /usr/include/c++/5/bits/basic_string.h:180
        #5 0x7f72ef5f2063 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string() /usr/include/c++/5/bits/basic_string.h:543
        #6 0x7f72ef5f2063 in cnoid::ExtensionManagerImpl::~ExtensionManagerImpl() /home/jun/dvl/openhrp/jenkins/drcutil/src/choreonoid/src/Base/ExtensionManager.cpp:104
    
    previously allocated by thread T0 here:
        #0 0x7f72efb3c532 in operator new(unsigned long) (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x99532)
        #1 0x7f72ef5efa29 in void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char*>(char*, char*, std::forward_iterator_tag) /usr/include/c++/5/bits/basic_string.tcc:223
    
    SUMMARY: AddressSanitizer: heap-use-after-free ??:0 memcmp
    Shadow bytes around the buggy address:
      0x0c0680034720: fa fa fd fd fd fd fa fa fd fd fd fd fa fa fd fd
      0x0c0680034730: fd fd fa fa 00 00 07 fa fa fa fd fd fd fd fa fa
      0x0c0680034740: 00 00 00 07 fa fa fd fd fd fd fa fa fd fd fd fa
      0x0c0680034750: fa fa fd fd fd fd fa fa fd fd fd fd fa fa fd fd
      0x0c0680034760: fd fd fa fa fd fd fd fd fa fa fd fd fd fd fa fa
    =>0x0c0680034770: fd fd fd fd fa fa fd fd fd fd fa fa[fd]fd fd fa
      0x0c0680034780: fa fa fd fd fd fa fa fa 00 00 00 fa fa fa 00 00
      0x0c0680034790: 00 fa fa fa 00 00 00 fa fa fa 00 00 00 fa fa fa
      0x0c06800347a0: 00 00 00 fa fa fa 00 00 00 fa fa fa 00 00 00 02
      0x0c06800347b0: fa fa 00 00 07 fa fa fa 00 00 05 fa fa fa 00 00
      0x0c06800347c0: 07 fa fa fa 00 00 00 06 fa fa 00 00 00 06 fa fa
    Shadow byte legend (one shadow byte represents 8 application bytes):
      Addressable:           00
      Partially addressable: 01 02 03 04 05 06 07 
      Heap left redzone:       fa
      Heap right redzone:      fb
      Freed heap region:       fd
      Stack left redzone:      f1
      Stack mid redzone:       f2
      Stack right redzone:     f3
      Stack partial redzone:   f4
      Stack after return:      f5
      Stack use after scope:   f8
      Global redzone:          f9
      Global init order:       f6
      Poisoned by user:        f7
      Container overflow:      fc
      Array cookie:            ac
      Intra object redzone:    bb
      ASan internal:           fe
    ==5684==ABORTING

s-nakaoka, 2016-11-12:
ご報告ありがとうございます。こちらでも確認してみます。

s-nakaoka, 2016-11-28:
どうもExtentionManagerImplにおいてViewManager等がスマートポインタによって削除されるタイミングが問題のようなので、それらのメンバの定義順序を変更しました。
8648e22f47e6f05a0c880a87bc68ad64ea2aca0

これにより、ViewManager等が削除されてから、それらが参照しているExtensionManagerの文字列が削除されることになるはずなので、これで問題が解決するのではないかと考えています。

jun0, 2016-11-28:
ご対応ありがとうございます。このリークは直ったようです。

closed 2017-04-27 by s-nakaoka