s-nakaoka/choreonoid #133: 配布されているコントローラを編集して自作のモデルを制御したい

51ozaki, 2017-01-30:
クローラのｗｒｌモデルを作りました。
このモデルをOpenRTMで制御したいと考えております．

そこでchoreonoidのソースに含まれているSamleCrawlerJoystickControllerRTC.cppを編集して使いたいのですが

http://www.jvrc.org/tutorials/html-ja/basics/rt-controller.html
http://choreonoid.org/ja/manuals/1.5/simulation/crawler-simulation.html

を見ながら作業していたのですが，うまくいきません．

配布されているコントローラのソースコードを編集して使える方法または，それについての情報を教えていただけないでしょうか．
よろしくお願いします．

OSはlinux ubuntuです

s-nakaoka, 2017-01-30:
> 配布されているコントローラのソースコードを編集して使える方法または，それについての情報を教えていただけないでしょうか．

ごめんなさい、まだこの件に直接対応するドキュメントはありません。

一般的には、ソースのext以下にsamplesの各フォルダと同様にCMakeLists.txtやソースファイルを配置して、Choreonoid本体といっしょにコンパイルしてしまうのが楽ではあります。

fkanehiro, 2017-01-30:
「うまくいきません」がどのようにうまくいかないのか、説明して頂けると答えやすいと思います。

51ozaki, 2017-01-30:
choreonoid-1.5.0/sample/OpenRTMにある
SampleCrawlerJoystickControllerRTC.cpp
SampleCrawlerJoystick.conf
SampleCrawlerJoystickControllerRTC.h
CMakeLists.txt
を入れたディレクトリを/choreonoid-1.5.0/ext/に置き
http://choreonoid.org/ja/manuals/1.5/install/build-ubuntu.html
のcmake .でエラーが出ます何か足りないでしょうか？

また、SampleCrawlerJoystickControllerRTC.cppのどこを編集すれば良いのかがわかりません
idやlinkの名前を入れるところがあるのでしょうか？

fkanehiro, 2017-01-30:
cmake .
するとどのようなエラーがでるでしょうか？

また何を変えたいのでしょうか？

s-nakaoka, 2017-01-30:
すみませんが質問の仕方について以下のページも参考にしてください。
http://www.hyuki.com/writing/techask.html

51ozaki, 2017-01-31:
「cmake .するとどのようなエラーがでるでしょうか？」

choreonoid-1.5.0/sample/OpenRTMに入っている

SampleCrawlerJoystickControllerRTC.cpp
SampleCrawlerJoystick.conf
SampleCrawlerJoystickControllerRTC.h
CMakeLists.txt

を入れたディレクトリを/choreonoid-1.5.0/ext/に置き

choreonoidのソースディレクトリ（/choreonoid-1.5.0）上でcmake .したところ

mechatropc-10@mechatropc10-Z97-S01:~$ cd choreonoid-1.5.0/
mechatropc-10@mechatropc10-Z97-S01:~/choreonoid-1.5.0$ cmake .
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   program_options
--   regex
--   thread
--   iostreams
--   date_time
--   python
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   chrono
-- OPENRTM VERSION 1.1.0
-- OPENRTM_DIR=/opt/ros/indigo
-- OpenRTMConfig.cmake found.
-- Configrued by configuration mode.
-- OpenRTM-aist configuration done
-- OPENRTM VERSION 1.1.0
CMake Error at ext/UMRS/CMakeLists.txt:27 (add_library):
  add_library cannot create target "SampleCrawlerJoystickControllerRTC"
  because another target with the same name already exists.  The existing
  target is a shared library created in source directory
  "/home/mechatropc-10/choreonoid-1.5.0/sample/OpenRTM".  See documentation
  for policy CMP0002 for more details.
Call Stack (most recent call first):
  ext/UMRS/CMakeLists.txt:47 (add_cnoid_sample_rtc)

CMake Error: File /home/mechatropc-10/choreonoid-1.5.0/ext/UMRS/OpenRTM-SampleCrawlerJoystick.cnoid does not exist.
CMake Error at ext/UMRS/CMakeLists.txt:48 (configure_file):
  configure_file Problem configuring file

-- Configuring incomplete, errors occurred!
See also "/home/mechatropc-10/choreonoid-1.5.0/CMakeFiles/CMakeOutput.log".
See also "/home/mechatropc-10/choreonoid-1.5.0/CMakeFiles/CMakeError.log".
mechatropc-10@mechatropc10-Z97-S01:~/choreonoid-1.5.0$ 

と出たのでSampleCrawlerと書かれているところを全部UMRSSampleCrawlerに変えてもう一度やると

mechatropc-10@mechatropc10-Z97-S01:~$ cd choreonoid-1.5.0/
mechatropc-10@mechatropc10-Z97-S01:~/choreonoid-1.5.0$ cmake .
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   program_options
--   regex
--   thread
--   iostreams
--   date_time
--   python
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   chrono
-- OPENRTM VERSION 1.1.0
-- OPENRTM_DIR=/opt/ros/indigo
-- OpenRTMConfig.cmake found.
-- Configrued by configuration mode.
-- OpenRTM-aist configuration done
-- OPENRTM VERSION 1.1.0
CMake Error: File /home/mechatropc-10/choreonoid-1.5.0/ext/UMRS/OpenRTM-UMRSSampleCrawlerJoystick.cnoid does not exist.
CMake Error at ext/UMRS/CMakeLists.txt:48 (configure_file):
  configure_file Problem configuring file

CMake Error: File /home/mechatropc-10/choreonoid-1.5.0/ext/UMRS/UMRSSampleCrawlerJoystick.conf does not exist.
CMake Error at ext/UMRS/CMakeLists.txt:49 (configure_file):
  configure_file Problem configuring file

-- Configuring incomplete, errors occurred!
See also "/home/mechatropc-10/choreonoid-1.5.0/CMakeFiles/CMakeOutput.log".
See also "/home/mechatropc-10/choreonoid-1.5.0/CMakeFiles/CMakeError.log".
mechatropc-10@mechatropc10-Z97-S01:~/choreonoid-1.5.0$ 

となりました。

「また何を変えたいのでしょうか？」
配布されているSampleCrawlerJoystickControllerRTC.cppでは同じく配布されているcrawler.wrlしか制御できないのではと考え、自分で作ったクローラモデルを制御するために
例えば

51ozaki, 2017-01-31:
「また何を変えたいのでしょうか？」

配布されているSampleCrawlerJoystickControllerRTC.cppでは同じく配布されているcrawler.wrlしか制御できないのではと考え、自分で作ったクローラモデルを制御するためにSampleCrawlerJoystickControllerRTC.cppのどこかを編集しなければいけないと考えました。
例えば
どこかにcrawler_L Jointまたは jointId 2と編集する必要があるのではと考えました。

以下、自作したモデルの左クローラ部

DEF crawler_body_to_crawler_L Joint {
          jointType "pseudoContinuousTrack"
          jointAxis 0.0 1.0 0.0
          jointId 2
          translation 0.0 0.175 0.06
          rotation 0 1 0 0
          ulimit 1e+16
          llimit -1e+16
          uvlimit inf
          lvlimit -inf
          children [
            DEF crawler_L Segment {
              mass 1.0
              centerOfMass 0.0 0.0 0.0
              momentsOfInertia [1.0 0.0 0.0 0.0 1.0 0.0 0.0 0.0 1.0]
              children [
                Transform {
                  scale 1.0 1.0 1.0
                  translation 0.0 0.0 0.0
                  rotation 0 1 0 0
                  children [
                    Inline {
                      url "crawler-crawler_L-craw

51ozaki, 2017-01-31:
「また何を変えたいのでしょうか？」

配布されているSampleCrawlerJoystickControllerRTC.cppでは同じく配布されているcrawler.wrlしか制御できないのではと考えました。
自分で作ったクローラモデルを制御するためにSampleCrawlerJoystickControllerRTC.cppのどこかを変えたいと思っています。

例えば
SampleCrawlerJoystickControllerRTC.cppのどこかにcrawler_L Jointまたは jointId 2と書き加えるなど

以下、自作したモデルの左クローラ部

DEF crawler_body_to_crawler_L Joint {
jointType "pseudoContinuousTrack"
jointAxis 0.0 1.0 0.0
jointId 2
translation 0.0 0.175 0.06
rotation 0 1 0 0
ulimit 1e+16
llimit -1e+16
uvlimit inf
lvlimit -inf
children [
DEF crawler_L Segment {
mass 1.0
centerOfMass 0.0 0.0 0.0
momentsOfInertia [1.0 0.0 0.0 0.0 1.0 0.0 0.0 0.0 1.0]
children [
Transform {
scale 1.0 1.0 1.0
translation 0.0 0.0 0.0
rotation 0 1 0 0
children [
Inline {
url "crawler-crawler_L-craw

fkanehiro, 2017-01-31:
エラーメッセージをよく読みましょう。
「CMake Error: File /home/mechatropc-10/choreonoid-1.5.0/ext/UMRS/OpenRTM-UMRSSampleCrawlerJoystick.cnoid does not exist.」
や
「CMake Error: File /home/mechatropc-10/choreonoid-1.5.0/ext/UMRS/UMRSSampleCrawlerJoystick.conf does not exist.」
と出ていますので、これらのディレクトリやファイルがちゃんと存在するか確認する必要があります。
そもそもこれらのファイルが必要ないのであれば、CMakeLists.txtから削除する必要があります。

サンプルでは、JointIdが0と1がクローラであることが前提になっています。
https://github.com/s-nakaoka/choreonoid/blob/master/sample/OpenRTM/SampleCrawlerJoystickControllerRTC.cpp#L70-L71
クローラのJointIdが2であれば、履帯の速度(velocities)を設定する際の添字を2にしたり、velocitiesの長さを適切に設定したりする必要があると思います。
https://github.com/s-nakaoka/choreonoid/blob/master/sample/OpenRTM/SampleCrawlerJoystickControllerRTC.cpp#L52

51ozaki, 2017-02-13:
SampleCrawlerJoystickControllerRTC.cpp
SampleCrawlerJoystick.conf
SampleCrawlerJoystickControllerRTC.h
のコンパイルには成功しました．

EclipseのSystem Diagramで，作成したBodyRTCをSampleCrawlerJoystickControllerRTCと接続したいのですが私が作成したBodyRTCには「dq」ポートがありません．
BodyRTCのポートの増やし方，又はその情報の場所を教えていただけませんでしょうか？

[image]

hattorishizuko, 2017-02-13:
SampleCrawlerJoystickControllerで説明いたします。

BodyRTCItemのプロパティで、”設定モード”を”設定ファイルを使用”にします。
設定ファイル名に、SampleCrawlerJoystick.confを指定します。
このファイルには、
in-port = dq:JOINT_VELOCITY
connection = dq:SampleCrawlerJoystickControllerRTC0:dq
connection = JoystickRTC0:axes:SampleCrawlerJoystickControllerRTC0:axes
と書かれています。
一番上の行が、dqポートの作成を指定している行です。
connectionはポート接続の指定なので、System Diagramを使用する場合は不要です。

fkanehiro, 2017-02-13:
これってマニュアルのどこに書かれていますでしょうか。
どこにあったっけな、と思って探すのが結構大変なので、マニュアルに検索機能があるとよいと思うのですが如何でしょうか。

51ozaki, 2017-02-14:
写真1に示すようにアドバイス通り行いうまくいったと思ったのですがクローラは動きませんでした．
写真2に示すようにコントローラからの入力は来ているようです。

おそらくモデルがおかしいと思うのですが，
"pseudoContinuousTrack"の場所はあっているのでしょうか？
また他の場所を編集しなければいけないのでしょうか？

写真1
[image]

写真2
[image]

crawler.wrl
#VRML V2.0 utf8

PROTO Joint [
  exposedField     SFVec3f      center              0 0 0
  exposedField     MFNode       children            []
  exposedField     MFFloat      llimit              []
  exposedField     MFFloat      lvlimit             []
  exposedField     SFRotation   limitOrientation    0 0 1 0
  exposedField     SFString     name                ""
  exposedField     SFRotation   rotation            0 0 1 0
  exposedField     SFVec3f      scale               1 1 1
  exposedField     SFRotation   scaleOrientation    0 0 1 0
  exposedField     MFFloat      stiffness           [ 0 0 0 ]
  exposedField     SFVec3f      translation         0 0 0
  exposedField     MFFloat      ulimit              []
  exposedField     MFFloat      uvlimit             []
  exposedField     SFString     jointType           ""
  exposedField     SFInt32      jointId             -1
  exposedField     SFVec3f      jointAxis           0 0 1

  exposedField     SFFloat      gearRatio           1
  exposedField     SFFloat      rotorInertia        0
  exposedField     SFFloat      rotorResistor       0
  exposedField     SFFloat      torqueConst         1
  exposedField     SFFloat      encoderPulse        1
]
{
  Transform {
    center           IS center
    children         IS children
    rotation         IS rotation
    scale            IS scale
    scaleOrientation IS scaleOrientation
    translation      IS translation
  }
}

PROTO Segment [
  field           SFVec3f     bboxCenter        0 0 0
  field           SFVec3f     bboxSize          -1 -1 -1
  exposedField    SFVec3f     centerOfMass      0 0 0
  exposedField    MFNode      children          [ ]
  exposedField    SFNode      coord             NULL
  exposedField    MFNode      displacers        [ ]
  exposedField    SFFloat     mass              0
  exposedField    MFFloat     momentsOfInertia  [ 0 0 0 0 0 0 0 0 0 ]
  exposedField    SFString    name              ""
  eventIn         MFNode      addChildren
  eventIn         MFNode      removeChildren
]
{
  Group {
    addChildren    IS addChildren
    bboxCenter     IS bboxCenter
    bboxSize       IS bboxSize
    children       IS children
[... 280 lines cut ...]
  ]
  # List up all the joints' name you use
    joints [
    USE crawler_body_Joint,
    USE crawler_Ltoflipper_L1,
    USE crawler_Ltoflipper_L2,
    USE crawler_body_to_crawler_L,
    USE crawler_Rtoflipper_R1,
    USE crawler_Rtoflipper_R2,
    USE crawler_body_to_crawler_R
  ]

# List up all the segments' name you use
  segments [
  USE crawler_body_Segment,
  USE crawler_L,
  USE flipper_L1,
  USE flipper_L2,
  USE crawler_R,
  USE flipper_R1,
  USE flipper_R2
  ] 
  name "crawler_body"
  version "1.0"
}

closed 2017-08-16 by s-nakaoka