s-nakaoka/choreonoid #138: ソースコードからのビルドとインストールができない。

HiroshiIzumitani, 2017-03-16:
はじめまして。大阪府立大学工業高等専門学校４年メカトロニクスコースの和泉谷です。

choreonoid HPに記載されている「ソースコードからのビルドとインストール (Ubuntu Linux編)」（　**http://choreonoid.org/ja/install/build-ubuntu.html#id6**　）
より、ここに記載されている通りに”CMake によるビルド設定”の”ccmake .”を実行し、 BUILD_GROBOT_PLUGIN を “ON” にしました。Press [c] to configureとあるように操作したのですが図のような画面になりこの先に進めません。
[image]

何が原因かわからないのですが、このようになります。どのようにすればよいでしょうか。

以上です。

ご指導のほどおねがいします。

s-nakaoka, 2017-03-16:
BUILD_OPENRTM_PLUGINがONになっていませんか？
このエラーメッセージからすると、OpenRTMがインストールされていないか、CMakeがそれをうまく見つけられていないように見えます。

HiroshiIzumitani, 2017-03-16:
ありがとうございます。

試して見ます。

closed 2017-04-27 by s-nakaoka