s-nakaoka/choreonoid #139: eigenのバージョンによってはコンパイルエラー

fkanehiro, 2017-03-30:
eigenのバージョンによってはコンパイルエラーとなるようです。
例えばDebian8（eigen3.2.2）では以下のようなエラーとなります。
```
[ 30%] Building CXX object src/Base/CMakeFiles/CnoidBase.dir/GL1SceneRenderer.cpp.o
In file included from /usr/include/eigen3/Eigen/Core:256:0,
                 from /home/kanehiro/src/choreonoid/src/Util/EigenTypes.h:8,
                 from /home/kanehiro/src/choreonoid/src/Util/BoundingBox.h:9,
                 from /home/kanehiro/src/choreonoid/include/cnoid/BoundingBox:1,
                 from /home/kanehiro/src/choreonoid/src/Util/SceneGraph.h:10,
                 from /home/kanehiro/src/choreonoid/src/Util/SceneRenderer.h:9,
                 from /home/kanehiro/src/choreonoid/include/cnoid/SceneRenderer:1,
                 from /home/kanehiro/src/choreonoid/src/Base/GLSceneRenderer.h:9,
                 from /home/kanehiro/src/choreonoid/include/cnoid/GLSceneRenderer:1,
                 from /home/kanehiro/src/choreonoid/src/Base/GL1SceneRenderer.h:9,
                 from /home/kanehiro/src/choreonoid/src/Base/GL1SceneRenderer.cpp:6:
/usr/include/eigen3/Eigen/src/Core/util/Memory.h: In instantiation of ‘void Eigen::aligned_allocator<T>::construct(Eigen::aligned_allocator<T>::pointer, const T&) [with T = std::_Sp_counted_ptr_inplace<{anonymous}::TransparentShapeInfo, Eigen::aligned_allocator<{anonymous}::TransparentShapeInfo>, (__gnu_cxx::_Lock_policy)2u>; Eigen::aligned_allocator<T>::pointer = std::_Sp_counted_ptr_inplace<{anonymous}::TransparentShapeInfo, Eigen::aligned_allocator<{anonymous}::TransparentShapeInfo>, (__gnu_cxx::_Lock_policy)2u>*]’:
/usr/include/c++/4.9/bits/alloc_traits.h:253:4:   required from ‘static std::_Require<typename std::allocator_traits<_Alloc>::__construct_helper<_Tp, _Args>::type> std::allocator_traits<_Alloc>::_S_construct(_Alloc&, _Tp*, _Args&& ...) [with _Tp = std::_Sp_counted_ptr_inplace<{anonymous}::TransparentShapeInfo, Eigen::aligned_allocator<{anonymous}::TransparentShapeInfo>, (__gnu_cxx::_Lock_policy)2u>; _Args = {const Eigen::aligned_allocator<{anonymous}::TransparentShapeInfo>}; _Alloc = Eigen::aligned_allocator<std::_Sp_counted_ptr_inplace<{anonymous}::TransparentShapeInfo, Eigen::aligned_allocator<{anonymous}::TransparentShapeInfo>, (__gnu_cxx::_Lock_policy)2u> >; std::_Require<typename std::allocator_traits<_Alloc>::__construct_helper<_Tp, _Args>::type> = void]’
/usr/include/c++/4.9/bits/alloc_traits.h:399:57:   required from ‘static decltype (_S_construct(__a, __p, (forward<_Args>)(std::allocator_traits::construct::__args)...)) std::allocator_traits<_Alloc>::construct(_Alloc&, _Tp*, _Args&& ...) [with _Tp = std::_Sp_counted_ptr_inplace<{anonymous}::TransparentShapeInfo, Eigen::aligned_allocator<{anonymous}::TransparentShapeInfo>, (__gnu_cxx::_Lock_policy)2u>; _Args = {const Eigen::aligned_allocator<{anonymous}::TransparentShapeInfo>}; _Alloc = Eigen::aligned_allocator<std::_Sp_counted_ptr_inplace<{anonymous}::TransparentShapeInfo, Eigen::aligned_allocator<{anonymous}::TransparentShapeInfo>, (__gnu_cxx::_Lock_policy)2u> >; decltype (_S_construct(__a, __p, (forward<_Args>)(std::allocator_traits::construct::__args)...)) = <type error>]’
/usr/include/c++/4.9/bits/shared_ptr_base.h:620:37:   required from ‘std::__shared_count<_Lp>::__shared_count(std::_Sp_make_shared_tag, _Tp*, const _Alloc&, _Args&& ...) [with _Tp = {anonymous}::TransparentShapeInfo; _Alloc = Eigen::aligned_allocator<{anonymous}::TransparentShapeInfo>; _Args = {}; __gnu_cxx::_Lock_policy _Lp = (__gnu_cxx::_Lock_policy)2u]’
/usr/include/c++/4.9/bits/shared_ptr_base.h:1090:35:   required from ‘std::__shared_ptr<_Tp, _Lp>::__shared_ptr(std::_Sp_make_shared_tag, const _Alloc&, _Args&& ...) [with _Alloc = Eigen::aligned_allocator<{anonymous}::TransparentShapeInfo>; _Args = {}; _Tp = {anonymous}::TransparentShapeInfo; __gnu_cxx::_Lock_policy _Lp = (__gnu_cxx::_Lock_policy)2u]’
/usr/include/c++/4.9/bits/shared_ptr.h:316:64:   required from ‘std::shared_ptr<_Tp>::shared_ptr(std::_Sp_make_shared_tag, const _Alloc&, _Args&& ...) [with _Alloc = Eigen::aligned_allocator<{anonymous}::TransparentShapeInfo>; _Args = {}; _Tp = {anonymous}::TransparentShapeInfo]’
/usr/include/c++/4.9/bits/shared_ptr.h:588:39:   required from ‘std::shared_ptr<_Tp1> std::allocate_shared(const _Alloc&, _Args&& ...) [with _Tp = {anonymous}::TransparentShapeInfo; _Alloc = Eigen::aligned_allocator<{anonymous}::TransparentShapeInfo>; _Args = {}]’
/home/kanehiro/src/choreonoid/src/Util/EigenUtil.h:78:92:   required from ‘std::shared_ptr<_Tp1> cnoid::make_shared_aligned(Args&& ...) [with T = {anonymous}::TransparentShapeInfo; Args = {}]’
/home/kanehiro/src/choreonoid/src/Base/GL1SceneRenderer.cpp:992:98:   required from here
/usr/include/eigen3/Eigen/src/Core/util/Memory.h:731:9: error: use of deleted function ‘std::_Sp_counted_ptr_inplace<{anonymous}::TransparentShapeInfo, Eigen::aligned_allocator<{anonymous}::TransparentShapeInfo>, (__gnu_cxx::_Lock_policy)2u>::_Sp_counted_ptr_inplace(const std::_Sp_counted_ptr_inplace<{anonymous}::TransparentShapeInfo, Eigen::aligned_allocator<{anonymous}::TransparentShapeInfo>, (__gnu_cxx::_Lock_policy)2u>&)’
         ::new( p ) T( value );
         ^
In file included from /usr/include/c++/4.9/bits/shared_ptr.h:52:0,
                 from /usr/include/c++/4.9/memory:82,
                 from /usr/include/boost/move/algorithm.hpp:24,
                 from /usr/include/boost/move/move.hpp:24,
                 from /usr/include/boost/variant/detail/move.hpp:27,
                 from /usr/include/boost/variant/detail/initializer.hpp:23,
                 from /usr/include/boost/variant/variant.hpp:33,
                 from /usr/include/boost/variant.hpp:17,
                 from /home/kanehiro/src/choreonoid/src/Util/SceneDrawables.h:11,
                 from /home/kanehiro/src/choreonoid/include/cnoid/SceneDrawables:1,
                 from /home/kanehiro/src/choreonoid/src/Base/GL1SceneRenderer.cpp:7:
/usr/include/c++/4.9/bits/shared_ptr_base.h:494:11: note: ‘std::_Sp_counted_ptr_inplace<{anonymous}::TransparentShapeInfo, Eigen::aligned_allocator<{anonymous}::TransparentShapeInfo>, (__gnu_cxx::_Lock_policy)2u>::_Sp_counted_ptr_inplace(const std::_Sp_counted_ptr_inplace<{anonymous}::TransparentShapeInfo, Eigen::aligned_allocator<{anonymous}::TransparentShapeInfo>, (__gnu_cxx::_Lock_policy)2u>&)’ is implicitly deleted because the default definition would be ill-formed:
     class _Sp_counted_ptr_inplace final : public _Sp_counted_base<_Lp>
           ^
/usr/include/c++/4.9/bits/shared_ptr_base.h:203:7: error: ‘std::_Sp_counted_base<_Lp>::_Sp_counted_base(const std::_Sp_counted_base<_Lp>&) [with __gnu_cxx::_Lock_policy _Lp = (__gnu_cxx::_Lock_policy)2u]’ is private
       _Sp_counted_base(_Sp_counted_base const&) = delete;
       ^
/usr/include/c++/4.9/bits/shared_ptr_base.h:494:11: error: within this context
     class _Sp_counted_ptr_inplace final : public _Sp_counted_base<_Lp>
           ^
/usr/include/c++/4.9/bits/shared_ptr_base.h:494:11: error: use of deleted function ‘std::_Sp_counted_base<_Lp>::_Sp_counted_base(const std::_Sp_counted_base<_Lp>&) [with __gnu_cxx::_Lock_policy _Lp = (__gnu_cxx::_Lock_policy)2u]’
/usr/include/c++/4.9/bits/shared_ptr_base.h:203:7: note: declared here
       _Sp_counted_base(_Sp_counted_base const&) = delete;
       ^
src/Base/CMakeFiles/CnoidBase.dir/build.make:1904: recipe for target 'src/Base/CMakeFiles/CnoidBase.dir/GL1SceneRenderer.cpp.o' failed
make[2]: *** [src/Base/CMakeFiles/CnoidBase.dir/GL1SceneRenderer.cpp.o] Error 1
CMakeFiles/Makefile2:801: recipe for target 'src/Base/CMakeFiles/CnoidBase.dir/all' failed
make[1]: *** [src/Base/CMakeFiles/CnoidBase.dir/all] Error 2
Makefile:137: recipe for target 'all' failed
```

s-nakaoka, 2017-08-16:
別件で進めているpybind11の導入の際にも、古いEigenだとpybind11が対応しておらず、コンパイルできないという問題がありました。Ubuntu 14.04にインストールされるEigenでも古いようです。

その問題とあわせて、どうもEigenのバージョンが古いといろいろ問題が起きそうなので、今pybind11化のブランチで最新のEigenのソースをChoreonoidのソースに含める改良をしています。近々masterにマージできると思います。これが導入されれば、Eigenのバージョンが古い場合はChoreonoidに内蔵のものが自動で使われるようになります。

s-nakaoka, 2017-09-01:
Eigenソース込みのバージョンをmasterにマージしました。

closed 2017-09-01 by s-nakaoka