s-nakaoka/choreonoid #141: 関節軌道ビューを含むプロジェクトファイルが読み込めない

y-masutani, 2017-04-11:
MLの方でもお騒がせしております…本件は，単純な問題なので，こちらで投げかけてみます．

メニューの表示→ビューの生成→関節軌道
を選び，その後，プロジェクトを保存します．そのプロジェクトを新たに読み込もうとすると，失敗します，メッセージビューには以下のように表示されます．
~~~~~
プロジェクトファイル"C:\home\tmp\a3.cnoid"を読み込み中…
control characters are not allowed at line 1, column 1. at line 1, column 1
プロジェクト "C:\home\tmp\a3.cnoid" は読み込めませんでした。
~~~~~
読み込めないファイルをzipして添付します．文字コードの扱いの問題でしょうか？
[a3.zip](https://github.com/s-nakaoka/choreonoid/files/911875/a3.zip)

Windows 10上で，GitHubのrelease-1.5をVisual Studio 2012でビルドして使っています．

hattorishizuko, 2017-04-18:
ファイルへの入出力の部分を修正しました。Windowsでも入出力されるファイルの文字コードはUTF-8になります。

申し訳ありませんが、添付して頂いたファイルは、出力時に変換が間違って行われているため、修正後のプログラムを使用しても読み込むことはできません。お手数ですが、プロジェクトの作成を再度行って頂く必要があります。

また、保存したファイルをテキストエディタなどで編集する場合、Windowでは文字化けする可能性があります。（Windows10のメモ帳はUTF-8に対応したようなので、問題無いようです。）
逆に、UTF-8対応のアプリケーションで開いているにもかかわらず、文字化けする場合は、choreonoidの方のバグですので、ご指摘頂けるとありがたいです。よろしくお願いします。

y-masutani, 2017-04-18:
ありがとうございます．対応するコミットはどれでしょうか？

hattorishizuko, 2017-04-18:
githubへの反映は、中岡しか権限がないので、もう少しお待ちください。
先走ってしまってしまいました。すみません。

s-nakaoka, 2017-04-18:
遅くなってしまいすみません、githubに反映しておきました。

y-masutani, 2017-04-18:
ありがとうございます．プルしたものをビルドして，問題が解消されていることを確認しました．

closed 2017-04-27 by s-nakaoka