s-nakaoka/choreonoid #145: サンプルタンクの砲塔部分について

yukimorikane, 2017-06-13:
今年度より卒業研究にてchoreonoidを使うものです。初歩的な質問になってしまうかもしれませんがよろしくお願いします。
早速なのですがサンプルタンクのシミュレーションを行った際に砲塔が図のように本体部分を貫通してしまうのですが可動範囲を設定したり、貫通しないようにすることは可能でしょうか。

[image]

s-nakaoka, 2017-06-13:
最新の開発版では、AISTシミュレータアイテムを使う場合、その自己干渉機能を有効とすることで、砲身が車体に当たるとそこでとまるようになります。アイテムのプロパティに「自己干渉」という項目がありますので、これをtrueにしてシミュレーションしてください。

これに関して、以下のチュートリアルにも記述がありますので、ご参照ください。
http://choreonoid.org/ja/manuals/latest/simulation/tank-tutorial/step1.html

yukimorikane, 2017-06-20:
回答ありがとうございます。自己干渉をtrueにすることはチュートリアルを拝見し実践していました。しかし、問題を解決することは出来ませんでした。現在私が使用しているchoreonoidのバージョンは1.6となっているのですがHPでは最新は1.5となっています。
また、昨年別の学生がインストールしたため開発版なのかが分かりません見分ける方法があれば教えていただけませんか。リリース版と開発版の違いは常に更新されるかされないかという程度の認識しか持っていませんが間違いないでしょうか。
質問ばかりで申し訳ありません。答えていだだけけると幸いです。

s-nakaoka, 2017-06-20:
開発版は暫定的にバージョン番号を1.6.0としています。（1.6.0のリリースに向けて開発中のバージョンとお考えください。）
バージョン番号はChoreonoidのメインメニューの「ヘルプ」-「Choreonoidについて」から確認できます。

自己干渉は開発版で最近導入された機能ですので、開発版の最新版にアップデートしてください。

開発版ではChoreonoid本体もマニュアルも随時更新されます。
従って、開発版のマニュアルに記載されている内容を試す場合は開発版の最新版を使うようにしてください。

yukimorikane, 2017-06-20:
回答ありがとうございます。今使っているものは開発版であるということなので最新版にアップデートすることを試してみようと思います。普段はchoreoniodを使っているパソコンはネットに繋がっていない状態なので更新されていないのはそのことが原因となっているはずです。

closed 2017-08-16 by s-nakaoka