s-nakaoka/choreonoid #146: ロボットが消える場合がある

haraisao, 2017-06-19:
開発版、1.5ReleaseのWindows版で、G-ROBOTやRIC30のロボットモデルで、ベースを(足先などに）変更した後に、元のベース（腰部）を操作するとロボット全体が消失する（見えなくなる）という現象があります。ベース設定を元に戻せばよいのですが。これは、仕様でしょうか？

s-nakaoka, 2017-06-19:
もちろん仕様ということはないのですが、OpenGL等の環境によっては発生してしまうのかもしれません。今回私の環境(Windows10, 64bit, Intel Corei7-7600U)で1.5のインストーラでインストールして試してみたのですが、そのような症状は出ませんでした。症状が再現できれば、修正できれかもしれませんが…。WindowsのバージョンやGPUの機種は何になりますでしょうか？

haraisao, 2017-06-19:
OSは、Windows7, Windows10の両方で同じ現象が確認できました。GPUは、Core i7に内臓のものです。（Intel HDになります）Ver.1.5の場合にも、必ずではないのですが、操作中に同じ現象に出くわしたことがあります。”Move the selected origin"のボタンで再び現れるのですが、何故このような現象になるのかわかりませんでした。

haraisao, 2017-06-19:
この現象が出るときは、マウスを少し動かすと　Rの値が 0 から -160 あたりまで飛んでいるようです。IKモードやFKモードではおきずにPresetモードのときに限定的におきます。

haraisao, 2017-06-20:
現在公開されているインストーラのものであれば、問題がないようです。開発版でQt5.9、eigen-3.3.3でビルドしていますが、これだと問題が発生しています。Qt5.9では、UseDesktopOpenGLを設定していますので、その影響かもしれません。

s-nakaoka, 2017-06-23:
IKのバグでした。
修正しておきました。
https://github.com/s-nakaoka/choreonoid/commit/c5bc7ef2cdc8c975f4f35d2058abfa167aa1b600

closed 2017-06-23 by s-nakaoka