s-nakaoka/choreonoid #148: WindowsでPython 2.7.11をインストールしていると起動しない

y-masutani, 2017-06-22:
Windows 10 64bitの環境で，Python 2.7.11 64bitをインストールしていると，Choreonoidが起動しません．ウィンドウが全く表示されず，コマンドプロンプトから実行しても何もメッセージが表示されません．

同じ環境にPython 2.7.10をインストールしている場合は問題ありません．それ以外の環境やバージョンでは試していません．Choreonoidは「Windows 64ビット版 インストーラ」で試しています．

授業でChoreonoidを使っているのですが，Python 2.7.10をインストールするように指示しているのに，間違えて2.7.11をインストールした学生がおり，その学生だけがChoreonoidを使えず，メッセージが何も出ないので手がかりもなく大変困りました．後になって，2.7.11をインストールしていることがわかり，私のPCでも症状が再現しました．

y-masutani, 2017-06-22:
補足します．Python 2.7.11との問題が解消しないとしても，Pythonを何もインストールしなくてもChoreonoidは使えていますので，それと同じ状態になるようにするか，適切なエラーメッセージを表示してから終了するようにしていただきたいです．よろしくお願いいたします．

s-nakaoka, 2017-06-23:
ご報告ありがとうございます。
調査しますのでしばらくお待ちください。

なお、Pythonの組み込みにあたってはこれまでBoost.Pythonというライブラリを用いてきたのですが、これが最新のコンパイラ環境に追従しきれていないところがあるようで、現在新しいライブラリであるpybind11に移行する作業を行っています。これと同時にPythonのバージョンも3に移行する予定です。本件とは直接関係ありませんが、今後この方針を進めて、Pythonに関わるトラブルもなるべく回避できるようにしたいと思っています。

y-masutani, 2017-06-26:
報告します．
環境変数「PYTHONHOME」を「C:\Python27」に設定すると起動できるようになりました．
OpenRTM-aistでもPython 2.7.11で問題があったことを思い出し試してみました．omniidlを実行すると「ImportError: No module named site」が起こっていました．

s-nakaoka, 2017-06-27:
ご報告ありがとうございます。
では現状のPython機能についてはその方法でなんとか運用いただいて、今後はPython3やpybind11への移行とともにトラブルがなくなることを期待しようと思います。

y-masutani, 2017-06-27:
当方としては，Python 2.7.11でないといけないということはありません．
他の方が同じトラブルに遭わないように，インストールのページなどにその旨注意を書いていただくと良いと思います．また，今後の設計方針として，何もメッセージを出さずに実行できないという状況が発生しないようにしていただきたいです．それがどの程度大変なことかはわかっていませんが，素朴にユーザとしての希望です．

s-nakaoka, 2017-06-27:
了解しました。
注意書きを加える件、当方のタスクリストに入れておきます。

ちなみに技術的な話としては、Choreonoidで行っているのはPythonのインタプリタのDLL（とBoost.Pythonライブラリ）をライブラリとして組み込んで使っていて、そこはいちおうPythonやBoost.Pythonのマニュアルを頼りに行っているのですが、Pythonのリリースバージョンがひとつ変わるだけで動かなくなるというのは、なかなか事前には予測できないですし、難しいところです。Windowsでトラブルなく動かすのはなかなか難しい・・・というのが正直なところではありますが、めげずに続けていこうとは思います。

closed 2017-11-20 by s-nakaoka