s-nakaoka/choreonoid #152: importing model from Solid-Works

Saturn-96, 2017-07-31:
How should I import my humanoid robot model from solid-works to Choreonoid and define joints and links ?!
I've been told via email to ask about the new file format which Choreonoid has introduced recently.

                                                                                                         thanks

s-nakaoka, 2017-08-02:
There is a document on how to make a new format model file.
http://choreonoid.org/en/manuals/latest/handling-models/modelfile/modelfile-newformat.html

This document only describes the case where link shapes are described as primitives in a model file.
For details on how to use external mesh files exported from a CAD tool, please refer to the samples like
- share/model/tank/tank2.body
- share/model/PA10/PA10.body
- share/model/GR001/GR001.body

Saturn-96, 2017-08-09:
thanks for your help, so you suggest the YAML format over openHRP cause I've defined my model in the (.wrl) file, is it better to write it in YAML before going further?

what causes the model to pass through the floor in a simple falling simulation like http://choreonoid.org/en/manuals/latest/simulation/simulation-project.html ?   (I considered the question really basic that didn't deserve a new topic issue )

s-nakaoka, 2017-08-16:
> what causes the model to pass through the floor in a simple falling simulation like http://choreonoid.org/en/manuals/latest/simulation/simulation-project.html ? (I considered the question really basic that didn't deserve a new topic issue )

Please check if the items of the robot model and the floor model are placed under the same WorldItem.

Saturn-96, 2017-09-03:
sure, they did!  actually, the problem only appears when I run the simulation(box falling) in windows, there is no such a problem when running on Ubuntu. Anyway, my final robot model falls fine so there's no worry anymore!   thanks  :)

closed 2017-09-26 by s-nakaoka