s-nakaoka/choreonoid #153: シミュレーション時のエラー

yukimorikane, 2017-08-02:
先日アップデートを行い起動してみたところ今まで動いていたモデルが動かなくなりました。メッセージには
SimpleControllerのinitializeメソッドが失敗しました
と記述されているのですがこのことが原因と考えています。解決策があれば教えていただけませんか？
他のサンプルモデルのシミュレーションは問題なく動いていたためコンピュータに不具合があるとは思えません。

s-nakaoka, 2017-08-02:
すみません、開発版ではSimpleControllerのAPIに少し変更が加えられています。
この件に直接関わる説明は以下にあります。
http://choreonoid.org/ja/manuals/latest/simulation/howto-implement-controller.html#body
また、SimpleControllerについては以下のチュートリアルもありますので、こちらもご覧いただけますでしょうか。
http://choreonoid.org/ja/manuals/latest/simulation/tank-tutorial/index.html

yukimorikane, 2017-08-10:
返信が遅くなり申し訳ありません。ここにあるTankJoystickController.soというプログラムを書き換えるということでいいのでしょうか。
[image]

s-nakaoka, 2017-08-16:
すみませんが、ご質問の意図がよく分かりません。
何をしようとされているのか、もう少し具体的にお願いします。

s-nakaoka, 2017-08-16:
ここも参考にしてみてください。
http://www.hyuki.com/writing/techask.html

closed 2017-09-26 by s-nakaoka