s-nakaoka/choreonoid #154: シーンビュー上でロボットが一瞬で消えてしまいます。

kx-guan, 2017-08-04:
こんにちは。大変お世話になっております。Choreonoidのユザーです。
シミュレーションにChoreonoidを使わせていただいて、問題があったので、こちらで質問をさせていただきます。

World、misc/floor.wrl,　と自作のロボットモデルVRMLファイルとAISTSimulatorでプロジェクトを構成して、サンプルコントローラーに付属している位置姿勢保持の一番シンプルなコントローラーを参考して、自作のロボットの位置姿勢保持の制御を試したところ、シーンビュー上でロボットが一瞬で消えてしまいました。また、JointSliderのところに現在の角度が9999.0というような数値が出ていました。
　原因と対策を教えていただけますでしょうか。
　どうぞよろしくお願い致します。

s-nakaoka, 2017-08-04:
シミュレーションのタイムステップやPD制御のゲインの設定がよくないと、そのようなことが起きることがあります。また、自作のロボットモデルとのことですが、その質量や慣性行列の設定が適切でない（値が小さすぎる等）場合も、そのようなことが起こります。それらについてご確認いただけますでしょうか。

kx-guan, 2017-08-06:
nakaoka先生：
　お忙しいところ迅速に回答してくださいましてありがとうございました。PD制御のゲインの設定などを確認します。
　どうぞよろしくお願い致します。

kx-guan, 2017-08-09:
nakaoka先生：
　大変お世話になっております。Choreonoidのユザーです。
　PD制御のゲインをいろいろ調整して、とりあえずシーン上から消えないことになっています。ただ、一番シンプルなコントローラーを使って、位置姿勢を保持しようとしましたが、pitch方向に倒れしまいました。しかし、pitch方向のgainをこれ以上上げたら、また消える現象が発生してしまいます。Choreonoidでシミュレーションをするとき、なぜこの現象が発生するか、教えていただけますでしょうか。どうぞよろしくお願い致します。

s-nakaoka, 2017-08-09:
シミュレーションのタイムステップはどうなっていますでしょうか。
AISTSimulatorItemでトルク制御を行う場合、1ms程度まで細かくとることが望ましいです。
また、ロボットモデルの質量や慣性行列について、小さすぎる部分はないでしょうか。そのような部分があると、不安定になりやすいです。
上記を改善しても問題が生じる場合、軸慣性を設定すると安定することがあります。
これはモデルファイルのリンクの記述で"rotorInertia"というパラメータを設定することで付与できます。アクチュエータのロータ部分のロータ軸まわりの慣性に対応します。ここの適当な値を設定しておくと、軸まわりの動きに追加の慣性が加わるので、挙動が安定する傾向があります。実際のロボットでも、アクチュエータのロータにはそれなりの慣性があることが多いです。

kx-guan, 2017-08-09:
nakaoka先生：
　大変お世話になっております。Choreonoidのユザーです。

　速やかな回答に感動いたしました。心から感謝いたします。
　タイムステップ：0.001（1ms）。
　ロボットモデルの質量や慣性行列：実機に合わせた数値だと考えられます。
　rotorInertia：設定してあります。

　ほかに何の原因だと考えられますでしょうか。
　どうぞよろしくお願い致します。

s-nakaoka, 2017-08-09:
なるほど、他の設定も悪くないようですね。
すみません、これ以上は実際に作られたものを見てみないことには、回答が難しいです。
ちなみに倒れるというのは、バランスが悪くて倒れているということはないでしょうか？

kx-guan, 2017-08-09:
nakaoka先生：
　大変お世話になっております。Choreonoidのユザーです。
　お忙しいところ迅速に回答してくださいましてありがとうございました。

　人間型のロボットで、最初少しゆっくり後ろに傾いて、そして、地面に倒れます。バランスが悪くて倒れていると私は思います。
　サンプルのSR1は一番シンプルなコントローラーを使って、初期の位置姿勢を保持できることになっています。シミュレーションをしようとしているロボットは最初そのSR1と同じように、一番シンプルなコントローラーを使って、初期の位置姿勢を保持できることを期待しております。
　中岡先生の経験から、何かアドバイスをいただければありがたいでございます。

　どうぞよろしくお願い致します。

s-nakaoka, 2017-08-09:
すみませんがこれ以上はロボットの制御の話になってくると思いますので、Choreonoidに関する議論の範囲を超えると思います。
倒れないようにする制御方法については以下の教科書（梶田秀司著「ヒューマノイドロボット」）を参考にしてみてください。
https://www.amazon.co.jp/%E3%83%92%E3%83%A5%E3%83%BC%E3%83%9E%E3%83%8E%E3%82%A4%E3%83%89%E3%83%AD%E3%83%9C%E3%83%83%E3%83%88-%E6%A2%B6%E7%94%B0-%E7%A7%80%E5%8F%B8/dp/4274200582

kx-guan, 2017-08-09:
nakaoka先生：
　大変お世話になっております。Choreonoidのユザーです。
　お忙しいところ迅速に回答してくださいましてありがとうございました。
　了解致しました。
　中岡先生の速やかなご回答に本当に心から感謝申し上げます。

　どうぞよろしくお願い致します。

y-masutani, 2017-08-09:
@kx-guan こんんちは，
rotorInertiaは，モータの出力側の値に換算したものを設定してやる必要があります．減速比をnとすると，実際のモータのロータ慣性モーメントのn^2倍の値にします．

kx-guan, 2017-08-10:
大阪電気通信大学　升谷先生：
　こんにちは。
　アドバイスをしてくださいまして、ありがとうございます。
　本観点も含めて検討させていただきます。
　まずは先生にお礼を申し上げます。
　どうぞよろしくお願い致します。

closed 2017-09-19 by s-nakaoka