s-nakaoka/choreonoid #158: BodyRTCのコントローラモジュールの不適合を回避できませんか？

y-masutani, 2017-09-01:
Windowsで使っています．限られた組み合わせでしか調べていませんが，BodyRTCのコントローラモジュールのdllをビルドしたVisual StudioのバージョンとChoreonoidをビルドしたVisual Studioのバージョンが一致していないと，RTCが生成されません．以下のようなメッセージが出ます．
```
RTC "zzz" をRTCマネージャから生成できません。
 RTCファイル: "C:\xxx\yyy\zzz.dll"
 Init関数: zzzInit
 オプション: ?exec_cxt.periodic.rate=1000000&exec_cxt.periodic.type=ChoreonoidExecutionContext
RTC "zzz" を生成できませんした。
```
この問題を回避する方法はありませんでしょうか？

s-nakaoka, 2017-09-01:
VC++の場合、VC++のバージョンによって使用するC/C++ランタイムライブラリのバージョンも決まってくるようで、異なるバージョンのものをリンクして（同一プロセスにまとめて）使うことは出来ない仕様のようです。（さらに、シングルスレッド／マルチスレッド、リリース／デバッグごとにライブラリが異なっていて、それらを混ぜて使うこともできないようです。）BodyRTCから直接読み込むRTCはChoreonoidのプロセス内で生成されるので、リンクして使うのと同じことになります。

ただし、Choreonoidとは別のプロセスとして実行（生成）したRTCに関しては、上記の制約はなくなるはずです。（プロセスが分かれているので実行はできるはずですし、恐らくポート接続もできるはず。）制御ループに同期する必要のないような上位層のモジュールであれば、そのようにして使う手もあるかと思います。ただしPD制御や安定化制御等、低レベルのサーボループ等に同期して動かす必要のあるモジュールについては、同期の問題や通信速度の問題が出てきますので、VC++のバージョンをそろえてChoreonoid上から直接読み込んで使うのが無難です。

y-masutani, 2017-09-02:
解説ありがとうございます．コントローラモジュールのデバッグの際に別プロセスでコンポーネントを実行することをやっていますので，おっしゃる方法は可能です．デバッグの時でしたので，パフォーマンスの違いを気にしていませんでしたが，それも見ながら試してみます．

ロボットのVRMLのモデルとBodyRTCのコントローラモジュールの一式を公開することを予定しているのですが，それを想定して，Choreonoid-1.5.0-win64.exeでインストールしたもので検証しようとしてハマりました．

y-masutani, 2017-09-02:
同じプロセスの場合と別プロセスの場合のパフォーマンスの比較は何を指標にすればいいでしょうか？

同じプロセスで実行する場合もRTC間の通信はCORBA経由なのですか？だとすると，それがボトルネックになってパフォーマンスはあまり変わらないかもしれません．

y-masutani, 2017-09-04:
実際に試してみました．それなりに高スペックのPC（Core i7 3.6GHz, RAM 16GB, GTX1060）で試しましたが，コントローラモジュールRTCをexeファイルで別プロセス実行するとパフォーマンスが出ませんでした．と言うよりも，フィードバック制御が破綻しました．Choreonoidの内部フレームレートを500Hzにして，関節角度を受け取ってトルクを出力するRTCを試しましたが，別プロセスの場合はデータを毎回受け取ることができません．全く同じプログラムのdllを読み込んだ場合は問題ありません．こんなに明確に違いがあるのですね．Choreonoidのコントローラモジュールがこのようになっていることがよくわかりました．

y-masutani, 2017-09-04:
Choreonoidのコントローラモジュールについての理解は深まりましたが，Visual Studioを使う場合の問題は解決していません．公式サイトにおいて，代表的なVSのバージョンでビルドしたインストーラを用意していただくことはできないでしょうか．

s-nakaoka, 2017-09-14:
別プロセスで実行する場合は、パフォーマンスが出ないのも問題ですが、ご指摘のような制御の破綻を起こさないためには、そのための工夫も必要です。具体的には、シミュレーションの進行（物理計算ループ）とコントローラ（onExecute関数の呼び出し）が同期するような実行コンテキストをRTCに設定する必要があります。これがBodyRTCItemを使ってChoreonoidプロセス内で動かす場合は、そのような実行コンテキストが設定されるのですが、別プロセスとなると、実行コンテキスト自体、シミュレータと通信して同期するようなものを用意しなければなりません。デフォルトではPeriodicECといった、実時間のタイマで駆動されるものが使われますので、それだと全く同期しません。

あと、仮に同期したとしても、やはりオーバーヘッドは大きいものになると思います。参考までに、当方で以前開発していたOpenHRP3では、システムがいくつかのCORBAサーバに分かれており、力学計算(Dynamicsサーバ）とコントローラの間でも、必ず通信が必要でした。Choreonoidにおいてシステムを同一プロセスに統合し、そのようなオーバーヘッドを除去したところ、確かSR1Walk等のシンプルなシミュレーションでは20倍近い差が出ていたように記憶しています。

s-nakaoka, 2017-09-14:
ちなみに升谷先生はご存知かと思いますが、他の方も見ていると思うので、念の為上記に補足すると、ここで言っているのはシステムの中で最も短い周期で動き、直接サーボの駆動に関わるような（PD制御を行うような）低レベルのコントローラが対象の話で、物体認識や動作計画、遠隔操作等、上位層のコントローラについては当てはまりません。もちろんそれらも同一プロセスで動かしたほうが通信のオーバーヘッドは少なくなりますが、それほど短い周期で動く必要もなく、行う処理自体も重いものなので、相対的に通信のオーバーヘッドは無視できるものになります。そのような上位層では、必要に応じて別プロセス化や通信を活用することで、より複雑なシステムも作りやすく成るかと思いますし、遠隔操作のようなことも可能となります。

s-nakaoka, 2017-09-14:
> 同じプロセスで実行する場合もRTC間の通信はCORBA経由なのですか？

ここはRTMの問題で私もあまり詳しくないのですが、確か同じプロセス内でのポート間通信はプロセス内で直接関数を呼び出すかたちで実行されるようになっていると聞いたような気がします。どなたかご存知でしたらフォローをお願いします。

s-nakaoka, 2017-09-14:
> 公式サイトにおいて，代表的なVSのバージョンでビルドしたインストーラを用意していただくことはできないでしょうか．

申し訳ないのですが、それは現状では難しいと思われます。
まずこれを行うにはかなりの労力が必要ですが、現状ではそれを行うだけのマンパワー（スタッフ）が当方にありません。
また、Choreonoidの開発版ではC++11規格で書くようになっていて、C++11に対応していない古いコンパイラだとコンパイルができません。Visual C++ でバージョンごとにどこまでC++11に対応しているかは把握しておりませんが、わりと最近のバージョンでないと完全には対応していないと聞いたことがあります。

s-nakaoka, 2017-09-19:
10月にChoreonoid1.6をリリースする予定で、その際には最新版のVisual C++ (今だと2017?)に関しては、バイナリのインストーラも用意しようと思っています。
参考までにとなりますが、なぜ古いバージョンのVC++の使用が必要となるのか、また代表的なVSのバージョンとは何なのか、よろしければお聞かせください。

また、どなたかWindows版のバイナリパッケージを作れそうなよい外注先をご存知だったりしませんでしょうか？もしご存知でしたら、私宛でメールしていただけると助かります。

y-masutani, 2017-09-20:
大学の授業で決まったバージョンを使わせていたり，他のソフトウェアとの兼ね合いで，バージョンが簡単に切り替えられない状況があることもご理解ください．私の学科では，2017年度の新入生から2015を使い始めました．それまでは，数年間2012で統一していました．研究室では，来年度に2015に切り替える予定です．今，何人かの学生と私が2012と2015を両方インストールして様子を見ています．

「代表的なVSのバージョン」は何が正解かわかりませんが，うちの大学がライセンスを受けているのは，2008, 2010, 2012, 2013, 2015, 2017です．他の事例としては，boostでは，msvc 8,9,10,11,12,14 (VS 2005, 2008, 2010, 2012, 2013, 2015)のバイナリを配布しています．

s-nakaoka, 2017-09-20:
お知らせいただきありがとうございます。状況承知いたしました。
今後のChoreonoidのリリースについて、コンパイラ側でのC++11対応は最低条件になりますので、現状出ているVC++の中でどのバージョン以降がC++11対応になっているかを確認し、なるべく対応するようにしていきたいと思います。

closed 2017-10-02 by s-nakaoka