s-nakaoka/choreonoid #159: シミュレーション開始でRTCのアクティベートが伝わる仕組み

y-masutani, 2017-09-01:
ChoreonoidでBodyRTCを使う場合，シミュレーションを開始すると，BodyRTCとそれに接続されている他のRTCがアクティベートされます（間接的にでも接続されていればOKのようです）．接続がないと，アクティベートされません．これはどのような仕組みでしょうか？ポートの接続を全て追いかけてアクティベートして回っているでしょうか？それとも，RTミドルウェアにそのようなことをやってくれる機能があるのでしょうか？これをやっているのは，ソースコードのどの辺でしょうか？

外部からのアクティベートでChoreonoidのシミュレーションを開始することはまだ試せていませんが，それに関連しての疑問です．

s-nakaoka, 2017-09-01:
ご指摘のように、ポートの接続を調べてつながっているRTCをアクティベートするようなコードになっていたかと思います。
ソースコードではsrc/OpenRTMPlugin/deprecated/VirtualRobotRTC.cpp の getConnectedRtcs()という関数でつながっているコンポーネントを集めているようです。

y-masutani, 2017-09-02:
解説ありがとうございます．より具体的な事例について質問いたします．

BodyRTCとそのコントローラモジュールのRTC，そして外部のRTCの三つでシステムを構成しています．外部のRTCは，コントローラモジュールRTCとだけ接続しています．BodyRTCとそのコントローラモジュールRTCの接続は設定ファイルに記述しています．

Choreonoidと外部RTCを起動して，外部RTCとの接続を済ませた状態のRTSEのダイアグラムの状態を以下に示します．
[image]
この後にChoreonoidで「初期位置からシミュレーションを開始」をクリックした後の状態を以下に示します．
[image]
外部RTCはアクティベートされず青色のままです．

比較実験として，BodyRTCとそのコントローラモジュールRTCの接続のうち1本だけをシミュレーション開始前に外部から設定しました．
[image]
この後にChoreonoidで「初期位置からシミュレーションを開始」をクリックした後の状態を以下に示します．
[image]
この場合は外部RTCもアクティベートされ緑色になりました．

以上の結果から推測して，設定ファイルに定義された接続を実行する前に，BodyRTCからポートの接続を辿り見つかったコンポーネントをシミュレーション開始時にアクティベートするという仕様でしょうか？もしそうであれば，設定ファイルに定義された接続を実行した後にアクティベートする対象を見つけてくれる方がありがたいのですが，いかがでしょうか？

s-nakaoka, 2017-09-14:
以下、本機能の開発者からのコメントです。

升谷先生ご指摘の通り、

1. configファイルで指定されたRTCとRobotに既に接続されているRTCを検出。
2. configファイルで指定された接続を実行
３．１．で検出したRTCに対してActivateする

という処理になっています。
１．で検出したRTCはContorlのときにtickする対象にもなっています。

RTSEditorを作成する前は、外部で起動したRTCは自前でActivateしてもらったほうがいいと思っていたので、上のようにしていたのですが、RTSEditorで画面で見えていると、なんで？と思うのもわかります。

というわけなのですが、今年度ChoreonoidのRTM連携機能について改善を行う計画がありますので、その検討項目リストの中に本件入れておきます。

y-masutani, 2017-09-17:
回答ありがとうございます．
経緯や大きなコンセプトをわかっていませんが，Chorenoidと外部のRTCを組み合わせて使う場合の管理は，以下のどちらかに徹していただければと思います．
- Chorenoid外部で行う．
#95 で書いたように，外部からChoreonoidのRTCをアクティベートするとシミュレーションが開始される．
- Choreonoid内部で行う．
このissueで書いているように，Choreonoidでシミュレーションを開始すると関連するRTCがもれなくアクティベートされる．

ところで，#161 の件はいかがでしょうか？

s-nakaoka, 2017-12-30:
この件のコンセプトについて、今後は以下のようにしたいと思っています。

- BodyIoRTCアイテム、ControllerRTCItemが管理するRTCで、シミュレーション対象のWorld内に存在するBodyアイテムと関連付けられているものは、シミュレーション開始時のactivate処理などが自動で行われる
- それ以外のRTCについては、システム側で自動で管理することはしない

BodyIoRTCアイテム、ControllerRTCアイテムについては、以下のドキュメントを参照してください。
http://choreonoid.org/ja/manuals/latest/openrtm/tank-joystick-project.html

BodyIoRTCアイテムは、従来のBodyRTCアイテムを置き換えるものになります。BodyRTCアイテムもしばらくは使えるようにしますが、いずれ廃止したいと思っています。

y-masutani, 2018-01-20:
応答が遅くなり申し訳ありません．今後のコンセプトについて理解しました．

このようなコンセプトになるならば，外部からChorenoidのシミュレーションを開始する手段がほしいです．以前から書いているように（書くだけで自分ではできていないのですが）Choreonoid内のRTCが外部からアクティベートされるとシミュレーションが開始され，ディアクティベートされるとシミュレーションが終了する機能があると嬉しいです．

http://choreonoid.org/ja/manuals/latest/openrtm/tank-joystick-project.html
のように連携するRTCをChoreonoid内で管理する場合はいいのですが，実機と切り替えて運用する場合などで，連携するRTCをChoreonoid外に置き，別の手段（RT System EditorやRTShellやRTCTree）で管理することを想定しています．

s-nakaoka, 2018-06-01:
すみません、書き込みからだいぶ時間がたっておりますが…。

外部RTCがアクティベートされなかったのは、外部というよりはPythonでRTCを作成したことに原因があって、RTCの実行コンテキストがCORBAオブジェクトとしてどの型に見えるかについて、C++で実装したものとPythonで実装したものとで異なっていようです。この差異を考慮するようにした改良版のOpenRTMPluginを開発版のmasterにマージしておきました。

ただ、升谷先生のご要望としては、シミュレーション開始など、シミュレーションを操作するためのRTCが利用可能となればよさそうですね。その件は当方の開発項目に追記しておきました。