s-nakaoka/choreonoid #160: プロジェクトを読み込むとビューがクリックに反応しなくなる

m-asahi, 2017-09-05:
Ubuntu 14.04（64bit、GeForce GTX 1060搭載の実マシンでドライバにnvidia-375.82を使用）で、Choreonoid 1.6-dev(SHA: bdc264a0586dba0869f4422bedb2362dc9fa0a0e)を使っています。
引数なしでChoreonoidを起動して、ファイルメニューの「プロジェクトの読み込み」で、サンプルのConveyorSample.cnoidを読み込むと、シーンビュー以外のビューがマウスの左クリックに反応しなくなります。

タブ切り替え、スクロールバーのようなものも無反応ですが、アイテムビューでは右クリックからコンテキストメニューを呼び出すことができ、カット、チェック、アンチェックといった項目は正常に機能します。

シミュレーションバーは正常に反応しますが、タイムバーの時間や進行状況は更新されません。

画面の端をドラッグして、ウインドウサイズを変更すると正常に動作するようになります。

また、Choreonoidの引数でConveyorSample.cnoidを指定した場合はこの問題は発生しません。
その後Worldをカットして、ファイルメニューのプロジェクトの読み込みでConveyorSample.cnoidやSampleCrawler.cnoid, SR1Walk.cnoid等を読み込んでも問題は発生しませんでした。
ただし、ここでTankJoystick.cnoidを読み込むと問題が発生します。

Choreonoidの引数でTankJoystick.cnoidを読み込んだ場合は問題は起きませんでした。

AISTSimulatorをはじめ、ODESimulator, BulletSimulatorによるシミュレーションは正常に実行できています。

よろしくお願いいたします。

s-nakaoka, 2017-09-14:
Ubuntu 16.04 の環境では再現できませんでした。
Ubuntu 14.04 の環境について、今後確認したいと思います。

fkanehiro, 2017-09-14:
類似の現象は少なくとも2年前（JVRCの頃）から起きていて、反応しなくなったら最大化して戻せばよい、というのは内輪ではよく知られたTipsになっていると思います。
はっきりとはわかりませんが、nvidiaのカードを搭載したPCでよく起きていて、タイミングとしてはChoreonoidのウィンドウをバックグラウンドにまわして、再度フォアグラウンドにすると操作できなくなることが多いように思います。

s-nakaoka, 2017-09-15:
了解です。
問題は再現性なのですよね。ある手順で確実に再現できるのであれば、デバッグもできるのですが・・・。
m-asahiさんの報告はそれに該当するものだと思われますので、まずはこれで試してみます。

s-nakaoka, 2017-09-15:
どうもQt4を使っているとこの症状が起きるのですが、Qt5の時は起きないようです。ChoreonoidではどちらのQtのバージョンもサポートしています。

githubの開発版(master)について、これまでデフォルトはQt4としていたのですが、先週行った修正で、Ubuntu 16.04 の場合はQt5がデフォルトで使われるようにしました。ただしそれ以前からお使いの場合はCMakeのキャッシュが残っていますので、CMakeの設定で "USE_QT5" を "ON" に変更してビルドし直してください。

Ubuntu 14.04 については、Qt5のサポートが十分でないようでしたので、デフォルトはQt4のままとしています。ただ、install-requsities-ubuntu-14.04.sh のスクリプトを更新して、これでQt5のパッケージも入るようにしました。ですので、まずこのスクリプトを実行し、その後CMakeの設定で "USE_QT5" を "ON" にしてビルドし直してください。Ubuntu14.04については、Qt5を使うことによりこの症状は収まることを当方環境で確認したのですが、それ以外の不具合が出る可能性はあります。

s-nakaoka, 2017-09-26:
この件いががでしょう？そちらで症状変わりましたでしょうか。

m-asahi, 2017-09-27:
ご報告が遅くなりましてすみません。
Ubuntu14.04でQt5を有効にしてビルドして、以下の三つのパターンで実行しましたが、いずれも問題は発生しませんでした。

1. choreonoidを引数なしで起動して、ConveyorSample.cnoidプロジェクトを読み込んだ場合
2. choreonoidの引数にConveyorSample.cnoidを指定して起動し、Worldアイテムを削除したあとTankJoystick.cnoidを読み込んだ場合
3. choreonoidの引数にTankJoystick.cnoidを指定して起動した場合

ただ、しばしばタブやメニューが文字化けしてしまって困りました。
[image]

（何度か）起動し直すとちゃんと表示されるのですが、フォントが異なるためか画面全体がしまりのない印象になります。
[image]

今気が付きましたが、デュアルディスプレイの片方で起動すると正常で、もう片方で起動すると文字化けするようです(?)
また何かわかったら、ご報告します。

m-asahi, 2017-09-27:
いや、やはりデュアルディスプレイの片方なら良くもう片方では悪いということはないようです。（さすがに）
同じコマンドラインで起動しても文字化けするときとしないときがあって、再現条件がまったく見当つきません。

さしあたり文字化けは無視するか、２，３回起動しなおして運良く文字化けが起きなかったら使うことにします。

s-nakaoka, 2017-09-27:
ご確認＆報告いただきありがとうございます。
なるほど、当初の症状は直ったものの、新たに文字化けという問題が発生してしまうのですね。ただ起動するたびに文字化けしたりしなかったりというのは厄介な症状ですね。やはりUbuntu14.04のQt5サポートがイマイチなのかも。
当方でも調査しようと思います。何かフォント関連のパッケージを追加でインストールするとか、そんなかたちで解決できるのであればよいのですが・・・。

s-nakaoka, 2017-09-28:
当方の環境でも、文字化けが起きたり起きなかったりすることが確認できました。
どうしたら直るのかはまだ分かっていませんが、どうも日本語を表示しようとする際に起きるようなので、そもそも日本語を表示しないよう英語環境で使うという手はあるかもしれません。
Choreonoidは環境変数LANGの値を見て使用言語を決めていますので、

export LANG=C

などとしてからchoreonoidを起動すると、表示が全て英語になります。この場合文字化けは起きないようです。

まあ日本語が使えなくなってしまうのであまり解決にはなってないかもしれませんが、文字化けするくらいなら英語でもよいという場合もあるかもしれないので、その場合はこの方法をお試し下さい。

s-nakaoka, 2017-10-02:
本件いちおう暫定的な対策は示せましたので、一旦閉じます。Ubuntu14.04でQt5を使う際の文字化けについては、今後対処法が分かり次第対策をしたいと思います。

closed 2017-10-02 by s-nakaoka