s-nakaoka/choreonoid #162: SimpleControllerPlugin not found !

HN11, 2017-09-11:
1. in "simple controller class" section of the "Implementation of Controller" tutorial , the definition of the class has been addressed to “src/SimpleControllerPlugin/library/SimpleController.h” . while there isn't such a any folder named "SimpleControllerPlugin" in the "src" directory of the choreonoid.
2. this problem even showed-up during the installation while I couldn't turn one of the features on from ccmake list because it required turning the simplecontrollerplugin on , but I didn't find it in nether normal nor advanced  mode of the ccmake list !!!

s-nakaoka, 2017-09-12:
The definition of SimpleController was moved to the Body module, and the definition of SimpleControllerItem was moved to the BodyPlugin. The document has not been updated yet, I'm sorry about it.

Do you mean there are still some plugins that needs to enable SimpleControllerPlugin? Which plugins?

HN11, 2017-09-12:
This is the error I received after running cmake command :
CMake Warning at sample/SpringModel/CMakeLists.txt:14 (message):
  The spring model sample controller needs SimpleControllerPlugin.

HN11, 2017-09-12:
I also have another question from this tutorial : 
I dont understand what this part of the menual means => 
As for compile, it is described in:

`add_cnoid_simple_controller(SR1MinimumController SR1MinimumController.cpp)`

in CMakeList.txt unde the same directory. See “src/SimpleControllerPlugin/library/CMakeLists.txt” for detail of this function. Basically, it is OK to link with the library “CnoidSimplerController”. (In case of Linux, the file name of the library will be “libCnoidCimpleController.so”.

s-nakaoka, 2017-09-14:
> The spring model sample controller needs SimpleControllerPlugin.

Thank you for informing this.
I fixed this problem.
f4f1aacf08105e4fb46cc2eae99c21cd25524d4b

s-nakaoka, 2017-09-14:
> should I treat my code like a usual cpp code or there are special ways for compiling and creating shared object file ?!

Usually you can use the add_cnoid_simple_controller function described in the document. If you want to know what the function exactly does, please refer to the implementation of the function. It is implemented in src/Body/CMakeLists.txt.

> What's the best procedure of having my original controller for the model ?!

It depends on the situation.
If you need the inter-process communication or network communication, it might be better to use OpenRTM (RT-Components). Or there will be a option to use ROS. Now I'm developing ROSPlugin for this purpose.

HN11, 2017-09-23:
thanks for your consideration :))

What should be the true result of recording a video out of simulation in choreonoid? Cause I receive lots of frames in .png format when I try to record my simulation !

s-nakaoka, 2017-09-26:
That is a right result.
You can use "ffmpeg" command,  for example, to generate a movie file from the sequential image files

`ffmpeg -r 30 -i scene%08d.png -r 30 movie.mp4`

closed 2017-10-02 by s-nakaoka