s-nakaoka/choreonoid #163: A NameService doesn't exist at "corbaloc:iiop:localhost:2809/NameService" エラーが発生

kx-guan, 2017-09-17:
中岡先生
　大変お世話になっております。kx-guanです。
　Choreonoidを起動時、掲題のようなエラーが発生してしまいました。以前はこの問題がありませんでしたが知らない間エラーが発生してしましました。OmniNameServerについてどこかで手動で変更されたことがあるかもしれません。Service再起動などの方法を試してみましたが、問題を解決できません。解決方法を教えていただけますでしょうか。
　よろしくお願い致します。

s-nakaoka, 2017-09-19:
この件試しにネームサーバを停止して（CorbaPluginやOpneRTMPluginも有効にしてある）Choreonoidを起動してみたのですが、その場合でも起動時にエラーは出ませんでした。
もう少し詳細をお知らせいただけますでしょうか？
（あとすみません、このページにしてもMLにしても必ずしも私が回答すると決まっているものではありませんので、特に理由がなければ宛名はなくて結構です。）

kx-guan, 2017-09-26:
Choreonoidを起動時、「A NameService doesn't exist at "corbaloc:iiop:localhost:2809/NameService".」とうエラーがありました。
OpenHRPのサンプルプロジェクトをChoreonoidで開いてシミュレーションをする時、、下記のエラーがありました。ロボットが自由落体のように地面に倒れました。
SR1LiftupControllerFor3.1 for SR1 failed to initialize.
SR1LiftupControllerFor3.0 for SR1 failed to initialize.

ほかには特に詳細がありません。すみません。
よろしくお願い致します。

s-nakaoka, 2017-09-26:
Choreonoidのバージョンは何になりますか？
githubの開発版か、リリース版ならバージョンいくつか。
私の上記コメントは開発版でテストしていました。

s-nakaoka, 2017-09-26:
あとOSはUbuntuでよろしかったでしょうか。その場合、Ubuntuのバージョンはいくつになりますでしょうか。

kx-guan, 2017-09-26:
Choreonoidのバージョンは1.5.0です。githubではなく、http://choreonoid.org/ja/download.htmlからソースパッケージchoreonoid-1.5.0.zipをダウンロードしてコンパイルしました。
OSはUbuntu　14.04です。

s-nakaoka, 2017-09-26:
Ubuntu 14.04 でChoreonoid-1.5.0 をインストールして、

 sudo service omniorb4-nameserver stop

としてから起動することで、症状を再現できました。その後

 sudo service omniorb4-nameserver start

としてネームサーバを再度起動してからchoreonoidを起動すると、エラーはなくなりました。

従って、そちらの環境で単純にネームサーバがうまく起動していないのではないかと思います。

上記のserviceコマンドを試してみていただけますでしょうか？

また、

ps ax| grep omniNames

を実行するとどうなりますか？omniNamesのプロセスは表示されますか？

表示されていなかったら、うまく起動していません。その場合、Choreonoidの問題ではなくて、omniORBの問題になります。

kx-guan, 2017-10-01:
先生、ご回答ありがとうございます。
ps ax| grep omniNamesを実行して、omniNamesのプロセスは表示されます。
sudo service omniorb4-nameserver startを実行しても、改善されません。
ただ一つ面白い現象を発見しました。OpenHRP用のEclipseを起動してからChoreonoidを起動して、症状が出ません。Eclipseを閉じたら、症状がまた出ます。（以前評価のため、OpenHRPもインストールしてみました。）

OpenHRP用のEclipseを起動するとき、下記のような情報がconsoleに出力されます。ですから、Choreonoidは起動するとき、何か自動起動されないNameServiceを利用している可能性があります。その辺には何か対策を教えていただけますでしょうか。
/usr/local/bin/openhrp-aist-dynamics-simulator  -ORBInitRef NameService=corbaloc:iiop:127.0.0.1:2809/NameService 
start:OK(DynamicsSimulatorFactory)

よろしくお願い致します。

s-nakaoka, 2017-10-02:
ひょっとして、メインメニューの「オプション」-「CORBA」-「ネームサーバ未検出時に自前のネームサーバを起動」にチェックが入っていたりするでしょうか。
このチェックが入っていると、書かれているとおりのことをします。この自前のネームサーバについては、 #167 の不具合があるようですので、
これが悪さをしているのかもしれません。

Ubuntuではこのチェックはデフォルトでは入っておりません。もし入っているとしたら、そちらで何かのはずみで入れてしまったのかもしれません。

s-nakaoka, 2017-10-02:
あるいは、omniNamesのキャッシュが悪さをしている可能性もあります。omniNamesでは前回起動時のネットワーク環境に関するキャッシュを保存して毎回それを復帰させようとするようなのですが、これがとにかくトラブルのもとになります。以下のコマンドで消去できますので、当方ではこれをスクリプトに書いておいて、何かおかしいときはそれを実行するようにしています。

sudo /etc/init.d/omniorb4-nameserver stop
sudo rm /var/log/omniorb-nameserver.log
sudo rm /var/lib/omniorb/omninames-*.bak
sudo rm /var/lib/omniorb/omninames-*.log
sudo /etc/init.d/omniorb4-nameserver start

kx-guan, 2017-10-02:
「ネームサーバ未検出時に自前のネームサーバを起動」にチェックが入っていません。デフォルトのままです。
ただ、先生が教えてくださったomniNamesのキャッシュを削除する対策でやってみて、問題を解決しました。

先生、本当にありがとうございました。

s-nakaoka, 2017-10-02:
なるほど、そうでしたか。
すみません、もう少し早くそれをコメントしていればよかったですね。

ネームサーバのキャッシュの件は本当に厄介ですね。多分これがそこらじゅうでトラブルを起こしまくっていて、CORBA嫌いを増やしている一つの要因になっていると思います。

実はChoreonoidの自前のネームサーバと言うのも、この件を解決したくて作ったものです。そちらはあえてキャッシュの処理を行わないようにしています。ただ現状ではRTCの登録がうまくできないようですので、その点は今後改善したいと思います。

s-nakaoka, 2017-10-02:
ただ、OpenHRP用のEclipseを起動していたら不具合が出なくなるのですね。それも謎ですが・・・。

これまでさんざんomniNamesのキャッシュのせいにしてきましたが、そもそもChoreonoidでのネームサーバの参照の仕方が悪いのかなぁ。どなたかこの件詳しい方いらっしゃらないでしょうか？

s-nakaoka, 2017-10-02:
とりあえずキャッシュをクリアするスクリプトを今後Choreonoidに標準装備して、マニュアルにもこの件に関する説明を入れることにいたします。

closed 2017-10-02 by s-nakaoka